espressif/esptool #773: Failing to upload new ESP8266 firmware (Flash Firmware) using NodeMCU PyFlasher 5.0.0

kudarukuni, 2022-08-27:
### Operating System

Windows 10

### Esptool Version

4.2.1

### Python Version

3.10.3

### Chip Description

NODEMCU ESP8266 ESP-12E v3

### Hardware Configuration

USB Cable

### How is Esptool Run

NodeMCU PyFlasher 5.0.0

### Full Esptool Command Line that Was Run

esptool.py --port COM4 --baud 9600 --after no_reset write_flash --flash_size detect --flash_mode dio 0x00000 C:\Users\Proggie\Desktop\ESP8266 FLASH\nodemcu-release-7-modules-2022-08-27-11-32-08-float.bin --erase-all

### Esptool Output

```shell
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\Proggie\Desktop\ESP8266 FLASH\PYFLASHER\Main.py", line 101, in run
    esptool.main(command)
  File "C:\Users\Proggie\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\__init__.py", line 637, in main
    esp = esp or get_default_connected_device(
  File "C:\Users\Proggie\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\__init__.py", line 880, in get_default_connected_device
    _esp = detect_chip(
  File "C:\Users\Proggie\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\cmds.py", line 91, in detect_chip
    detect_port.connect(connect_mode, connect_attempts, detecting=True)
  File "C:\Users\Proggie\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 603, in connect
    last_error = self._connect_attempt(
  File "C:\Users\Proggie\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 544, in _connect_attempt
    self.sync()
  File "C:\Users\Proggie\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 407, in sync
    val, _ = self.command(
  File "C:\Users\Proggie\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 337, in command
    self.write(pkt)
  File "C:\Users\Proggie\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 289, in write
    self._port.write(buf)
  File "C:\Users\Proggie\AppData\Local\Programs\Python\Python310\lib\site-packages\serial\serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Proggie\AppData\Local\Programs\Python\Python310\lib\threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "C:\Users\Proggie\Desktop\ESP8266 FLASH\PYFLASHER\Main.py", line 108, in run
    self._parent.report_error(e.strerror)
  File "C:\Users\Proggie\Desktop\ESP8266 FLASH\PYFLASHER\Main.py", line 383, in report_error
    self.console_ctrl.SetValue(message)
TypeError: TextEntry.SetValue(): argument 1 has unexpected type 'NoneType'
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-08-29:
Hi @kudarukuni,

To be honest, this seems like a NodeMCU PyFlasher problem. Have you tried using bare esptool? 

You can [install it from PyPI](https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html) and try running the same command.

If this helps, this issue should be opened in the [NodeMCU PyFlasher repository](https://github.com/marcelstoer/nodemcu-pyflasher).

kudarukuni, 2022-08-29:
> 

Thank you for reaching out, this issue should be closed, to solve it i ended up using the esptool.py via cli and it did flash. do not know why the pyflasher was failing. otherwise i followed this procedure (https://github.com/kudarukuni/NodeMCU-PyFlasher/blob/main/README.md) for a successful flash

radimkarnis, 2022-08-29:
Thanks for the update! I am glad this got resolved.

closed 2022-08-29 by radimkarnis