espressif/esptool #776: Can't flash esp32s3, with error "struct.error: unpack requires a buffer of 16 bytes"

QuentinFarizon, 2022-09-05:
### Operating System

Windows

### Esptool Version

3.2, 4.2.1

### Python Version

3.8

### Chip Description

ESP32-S3

### Device Description

Espressif ESP32-S3-DevKitM-1

### How is Esptool Run

By python library

### Full Esptool Command Line that Was Run

See below

### Esptool Output

```shell
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Serial port COM4
Connecting....
Exception in thread Thread-1:
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "C:\Gitlab-Runner\builds\P6sDV2sr\0\boks3\boks-link\flasher\link_flasher.py", line 61, in _flash
  File "C:\Gitlab-Runner\builds\P6sDV2sr\0\boks3\boks-link\flasher\link_flasher.py", line 54, in esptool_flash
  File "esptool\__init__.py", line 646, in main
  File "esptool\__init__.py", line 895, in get_default_connected_device
  File "esptool\loader.py", line 675, in connect
  File "esptool\loader.py", line 1374, in check_chip_id
  File "esptool\loader.py", line 108, in inner
  File "esptool\loader.py", line 862, in get_chip_id
struct.error: unpack requires a buffer of 16 bytes
```

### More Information

I have found other issues pertaining to this, but they all seemed to be resolved by setting `--chip` to avoid chip detection, but I do have specified --chip esp32s3

### Other Steps to Reproduce

```
        command = ['--before', 'default_reset',
                   '--after', 'hard_reset',
                   '--chip', 'esp32s3',
                   'write_flash',
                   '--flash_mode', 'dio',
                   '--flash_freq', '80m',
                   '0x0', get_bundled_path_or_parent(build_directory + '/bootloader/bootloader.bin'),
                   '0x8000', get_bundled_path_or_parent(build_directory + '/partition_table/partition-table.bin'),
                   '0xd000', get_bundled_path_or_parent(build_directory + '/ota_data_initial.bin'),
                   '0x10000', get_bundled_path_or_parent(build_directory + app_bin)
                   ]

    esptool.main(command)
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

closed 2022-09-05 by QuentinFarizon

radimkarnis, 2022-09-05:
Hello @QuentinFarizon, 

I see you’ve closed the issue - is this not relevant anymore?

QuentinFarizon, 2022-09-05:
Yes sorry, I was connecting an esp32s2...