espressif/esptool #777: win11 'esptool.py' is not recognized as an internal or external command, operable program or batch file.

spikey1973, 2022-09-08:
### Operating System

win11 

### Esptool Version

v4.2.1

### Python Version

Python 3.10

### Full Esptool Command Line that Was Run

esptool.py -h

### Esptool Output

```shell
'esptool.py' is not recognized as an internal or external command,
operable program or batch file.
```

### What is the Expected Behaviour?

help function 

### More Information

ran: pip3 install esptool: 

# requirements already satisfied

Requirement already satisfied: esptool in c:\users\mvkin\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (4.2.1)
Requirement already satisfied: bitstring>=3.1.6 in c:\users\mvkin\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from esptool) (3.1.9)
Requirement already satisfied: cryptography>=2.1.4 in c:\users\mvkin\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from esptool) (37.0.4)
Requirement already satisfied: ecdsa>=0.16.0 in c:\users\mvkin\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from esptool) (0.18.0)
Requirement already satisfied: pyserial>=3.0 in c:\users\mvkin\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from esptool) (3.5)
Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in c:\users\mvkin\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from esptool) (1.5.4)
Requirement already satisfied: cffi>=1.12 in c:\users\mvkin\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from cryptography>=2.1.4->esptool) (1.15.1)
Requirement already satisfied: six>=1.9.0 in c:\users\mvkin\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from ecdsa>=0.16.0->esptool) (1.16.0)
Requirement already satisfied: pycparser in c:\users\mvkin\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from cffi>=1.12->cryptography>=2.1.4->esptool) (2.21)

ran:  pip show -f esptool: 

# couldn't find /bin/esptool.py

Name: esptool
Version: 4.2.1
Summary: A serial utility to communicate & flash code to Espressif chips.
Home-page: https://github.com/espressif/esptool/
Author: Fredrik Ahlberg (themadinventor) & Angus Gratton (projectgus) & Espressif Systems
Author-email:
License: GPLv2+
Location: c:\users\mvkin\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages
Requires: bitstring, cryptography, ecdsa, pyserial, reedsolo
Required-by:
Files:
  ..\Scripts\espefuse.py-script.py
  ..\Scripts\espefuse.py.exe
  ..\Scripts\espsecure.py-script.py
  ..\Scripts\espsecure.py.exe
  ..\Scripts\esptool.py-script.py
  ..\Scripts\esptool.py.exe
  espefuse\__init__.py
  espefuse\__main__.py
  espefuse\__pycache__\__init__.cpython-310.pyc
  espefuse\__pycache__\__main__.cpython-310.pyc
  espefuse\efuse\__init__.py
  espefuse\efuse\__pycache__\__init__.cpython-310.pyc
[... 134 lines cut ...]

Googled the error msg: 

# It was found, but none of the proposed answers i found worked for me. 

I looked in the local folder .../site-packages/esptool:

# did not find eptool.py

28/08/2022  21:32    <DIR>          .
28/08/2022  21:34    <DIR>          ..
09/08/2022  18:00            44.667 bin_image.py
09/08/2022  18:00            39.092 cmds.py
09/08/2022  18:00            56.604 loader.py
28/08/2022  21:32    <DIR>          targets
09/08/2022  18:00             4.318 util.py
09/08/2022  18:00            33.658 __init__.py
09/08/2022  18:00               270 __main__.py
28/08/2022  21:32    <DIR>          __pycache__
               6 File(s)        178.609 bytes
               4 Dir(s)  732.253.667.328 bytes free

### Other Steps to Reproduce

run any esptool command

radimkarnis, 2022-09-09:
Hello @spikey1973,

thanks for reporting this. Just to check - can you run the command without the `.py` extension (e.g. `esptool flash_id`)?

spikey1973, 2022-09-09:
Hello, 

just to double check i ran the command but no esptool command with or without .py works. 

Kind greats 

Matthieiu

radimkarnis, 2022-09-12:
Can you run other python modules this way (install with pip -> run on a command line)?

This seems like a $PATH problem in your shell - $PATH being the environment variable that controls how (and where) your shell is able to locate and invoke commands like `epstool.py` (the command line tool). You need to fix your $PATH. On Windows, you have to locate the `esptool.py` executable and add the enclosing folder to your $PATH, then restart all of your shells (so the updated environment variable can take effect.) In fact, pip probably told you you had to do this when you installed the tool - it's usually pretty aware of when it installs a command line tool but isn't able to put it on your path for you.

Another thing to try is to invoke esptool as: `python -m esptool` (or `python3 -m esptool`)

radimkarnis, 2022-09-21:
I am closing this issue, it has been answered and it seems like it's not relevant anymore. Feel free to reopen if necessary!

closed 2022-09-21 by radimkarnis