espressif/esptool #784: esptool fails with Failed to autodetect chip type

c0debreaker, 2022-09-29:
### Operating System

Linux Debian 10

### Esptool Version

esptool.py v2.5.1

### Python Version

Python 3.10.2

### Full Esptool Command Line that Was Run

esptool --port /dev/ttyACM8 write_flash 0x0 bootloader.bin  0x8000 partition-table.bin 0x10000 esp-gyrologger.bin

### Esptool Output

```shell
esptool.py v2.5.1
Serial port /dev/ttyACM8
Connecting...
Detecting chip type...

A fatal error occurred: Unexpected UART datecode value 0x03700000. Failed to autodetect chip type.
```

### What is the Expected Behaviour?

It should flash the board

### More Information

I ran esptool trace hoping it will tell you why I am encountering the error

```
root# esptool --trace --before no_reset --port /dev/ttyACM8 chip_id
esptool.py v2.5.1
Serial port /dev/ttyACM8
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 41 bytes: 
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0               | ... .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 70 bytes: 
    c001080400070712 2000000000c0c001 | ........ .......
    0804000707122000 000000c0c0010804 | ...... .........
    0007071220000000 00c0c00108040007 | .... ...........
    07122000000000c0 c001080400070712 | .. .............
    2000000000c0                      |  .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000

Detecting chip type...TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=78000060
TRACE +0.000 Write 14 bytes: c0000a04000000000078000060c0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a04000000700300000000c0
TRACE +0.000 Received full packet: 010a04000000700300000000

A fatal error occurred: Unexpected UART datecode value 0x03700000. Failed to autodetect chip type.                                                             exit:2
    
```

c0debreaker, 2022-09-29:
I added --chip esp32 and it helped a little but I have a new error related to flash download mode

```
esptool.py v2.5.1
Serial port /dev/ttyACM8
Connecting...
Chip is ESP32D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core
MAC: 00:00:00:00:00:00
Enabling default SPI flash mode...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 010 exit:2
```

dobairoland, 2022-09-29:
Hi @c0debreaker. You are using a 4 years old version. A lot of has changed since then. Even if we would able to reproduce this, we cannot provide a fix for it. Could you please try it with the latest v4.3 version or the master branch?

radimkarnis, 2022-10-10:
Please try to use the latest version of esptool. I will close this issue due to inactivity, feel free to reopen it if the problems persist on later versions.

closed 2022-10-10 by radimkarnis