espressif/esptool #785: ESP C3 cannot upload script, repl does not answer - flashing: OK

Erkman84, 2022-10-02:
### Operating System

Micropython

### Esptool Version

v4.3

### Python Version

1.19

### Chip Description

nodemcu esp c3-13/130-Kit

### Device Description

Factory model, no modifications

### Hardware Configuration

factory soldiered, no modification, no external connectors, just a simple device

### How is Esptool Run

no, run from basic shell promt (terminal)

### Full Esptool Command Line that Was Run

sudo ampy --port /dev/ttyUSB0 put boot.py && picocom /dev/ttyUSB0 -b115200

### Esptool Output

```shell
upload: no output, seems to loading, but nothing happens (I can cancel with ctrl+c in terminal)
picocom: you will get a promt, but you cannot write anything in it (cursor not working), ex: import os, and so on.
```

### More Information

Hi,
I have a similar issue like (https://github.com/espressif/esptool/issues/630): after flashing with micropython (Raspberry OS) i do not get any repl promt, and what is strange: i cannot upload any script (main.py, boot.py), In this case I used a ESP C3 module. But unfortunatelly with some esp8266 boards the same thing i cannot understand.
Repeating myself: I successfully flashed every board (!) (mentioned like on the man pages) BUT for some models after that i cannot upload any code, picocom repl answert, but there is input seems to be freezd. I thought it is conneted with CH340 driver but the system is up-to-date, so no kernel problems should be tried on antoher ubuntu 18 up-to-date system but it seems not to be an issue with my system.

Other Example:
Wemos D1 mini eps8266 V1 --> flash: ok, write script: ok. That's okay. (also with newest micropython bin-s.)
Wemos D1 mini esp8266 V2 (or V3 i dont know) --> maybe other chip (?): flash: ok, write script fails, no repl prompt in picocom.
Classic NodeMCU Esp8266: flash: ok, upload script: fails.
Lillygo T-Call 1.4 or 1.5

Other devices I have no issues:
ESP32-CAM (flash throug TTL RX-TX pin, then upload script with causal usb wires, but this is in the manual, so that's okay) 
Lillygo T-Call 1.3 (flash and upload script via causal USB)
--> these works like a charm.

Additinal info: tried booth ways: causal USB connection and over usb ttl serial TR-RX-5V-GRN wire.

Tried windows 7 (with updated driver) with putty to connect, also no repl promt for the selected devices.

How I upload from terminal: ampy --port /dev/ttyUSB0 put boot.py 

Anyone an idea? Micropython 1.19 bin --> this should be also not the problem, with the non-problematic devices it's working fine also older versions

### Other Steps to Reproduce

I've tried on other PC with Ubuntu 18 (up-to-date) tried with update usb driver, maybe this didn't was successfull, but I do not know.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-10-03:
Hi @Erkman84,

I am sorry, but this is not an esptool issue. You even say that you "successfully flashed every board", so esptool is working as intended.

Your problem will most likely be incompatible hardware or wrong settings. I suggest you to:

- Read the [troubleshooting installation problems](https://docs.micropython.org/en/latest/esp8266/tutorial/intro.html#troubleshooting-installation-problems) in Micropython docs.
- Search through issues in the [Micropython repository](https://github.com/micropython/micropython). If your problems persist, open an issue ticket there.
- Check the [Micropython forum](https://forum.micropython.org/) or their new [GH discussions](https://github.com/orgs/micropython/discussions)

radimkarnis, 2022-10-06:
Closing this issue. Feel free to reopen if you feel like we should discuss more!

closed 2022-10-06 by radimkarnis