espressif/esptool #787: ESP8266 Write Flash problem on Linux

BLLuis, 2022-10-04:
### Operating System

Debian Linux 11

### Esptool Version

esptool.py v4.4-dev

### Python Version

Python 3.9.2

### Chip Description

ESP8266-1 8Mbit

### Device Description

Generic board

### Hardware Configuration

No

### How is Esptool Run

Command line

### Full Esptool Command Line that Was Run

python3 esptool/esptool.py --port /dev/ttyUSB0 --before no_reset_no_sync --after no_reset --chip esp8266 write_flash 0x00000 ESP8266_IOT_PLATFORM/bin/boot_v1.2.bin

### Esptool Output

```shell
esptool.py v4.4-dev
Serial port /dev/ttyUSB0
WARNING: Pre-connection option "no_reset_no_sync" was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
Connecting...
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:0f:0b:73
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00000fff...
Compressed 1936 bytes to 1460...
Wrote 1936 bytes (1460 compressed) at 0x00000000 in 0.2 seconds (effective 75.9 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
```

### More Information

After flashing boot.bin with esptool in Linux, the chip wont boot anymore, while doing the same from Windows with ESP8266 FLASH DOWNLOAD TOOL V3.9.3, using the same address, bin file and hardware the chip boots ok after the flash.
I try to use options like --fm to "force" qio (as noted in your troubleshooting guide", just in case, but the tool don't like the parm.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-10-04:
Hi @BLLuis,

the FLASH DOWNLOAD TOOL probably sets something in the background ( e.g. flash mode) when flashing. You could try to read the first few bytes of the memory in both working and non-working cases with `esptool.py read_flash 0 8 out.bin` and check the [image header](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/firmware-image-format.html#file-header) for differences.

AFAIK, the FLASH DOWNLOAD TOOL uses esptool under the hood. So if there are no differences in the image header, you could try to use an older esptool and see if that fixes the issue. If that is the case, we could find the commit that introduced this possible regression.

BLLuis, 2022-10-04:
> Could you explain why are you using the `no_reset_no_sync` option? It is pretty uncommon and one should use it as an optimization option. But I wouldn't recommend it to use while you have problem with flashing.

Because I manually reset and put in program mode the board, not using DTR/RTS.

BLLuis, 2022-10-04:
> Hi @BLLuis,
>
> the FLASH DOWNLOAD TOOL probably sets something in the background ( e.g. flash mode) when flashing. You could try to read the first few bytes of the memory in both working and [...]

Thanks for your response, finally I detect the problem (using the method you commented) and solved using:

write_flash --flash_freq 40m --flash_mode qio

I've tried before (just to clear possible problems) but was not working, maybe because using 0 (zero) in the word qio or maybe some special char in spaces copy/pasted from some web page, but now its working.
I been working for years with a ESP8266 wroom, even with OTA updates, so, many years without using esptool, some days ago I found a ESP8266-1 in my lab and start to move one of my proyect to this board and because OTA doesn't work yet I need to use esptool after every brick.

Regards
Luis

radimkarnis, 2022-10-05:
Great! I am glad you were able to solve this!

closed 2022-10-05 by radimkarnis