espressif/esptool #788: Receive ROM_INVALID_RECV_MSG When Uploading stub...

yuruxuan, 2022-10-05:
### Operating System

MacOS Mojave 10.14.6

### Esptool Version

4.3

### Python Version

3.9.7

### Chip Description

ESP32C3

### Device Description

合宙ESP32C3 with CH343 Chip Version

### Hardware Configuration

https://cdn.openluat-luatcommunity.openluat.com/attachment/CORE-ESP32-C3设计原理图_V1.2.pdf

### How is Esptool Run

VSCode & Command Line Both

### Full Esptool Command Line that Was Run

/Users/yuruxuan/.espressif/python_env/idf5.0_py3.9_env/bin/python /Users/yuruxuan/esp/esp-idf/components/esptool_py/esptool/esptool.py -t -p /dev/cu.wchusbserial544C00096701 -b 460800 --before default_reset --after hard_reset --chip esp32c3 write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin

### Esptool Output

```shell
yuruxuandeMacBook-Pro:build yuruxuan$ /Users/yuruxuan/.espressif/python_env/idf5.0_py3.9_env/bin/python /Users/yuruxuan/esp/esp-idf/components/esptool_py/esptool/esptool.py -t -p /dev/cu.wchusbserial544C00096701 -b 460800 --before default_reset --after hard_reset --chip esp32c3 write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin 
esptool.py v4.3
Serial port /dev/cu.wchusbserial544C00096701
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Yuuuu
TRACE +0.103 No serial data received.
.TRACE +0.055 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Yuuuu
TRACE +0.011 Read 1 bytes: c0
TRACE +0.000 Read 63 bytes: 
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0c0010804000707 | ... ............
    122000000000c0c0 01080400070712   | . .............
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Yuuuu
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Yuuuu
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Yuuuu
TRACE +0.000 Received full packet: 010804000707122000000000
[... 544 lines cut ...]
Traceback (most recent call last):
  File "/Users/yuruxuan/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 22, in <module>
    esptool._main()  # type: ignore
  File "/Users/yuruxuan/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/esptool/__init__.py", line 1022, in _main
    main()
  File "/Users/yuruxuan/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/esptool/__init__.py", line 702, in main
    esp = esp.run_stub()
  File "/Users/yuruxuan/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 932, in run_stub
    self.mem_block(field[from_offs:to_offs], seq)
  File "/Users/yuruxuan/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 754, in mem_block
    return self.check_command(
  File "/Users/yuruxuan/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 408, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/Users/yuruxuan/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 391, in command
    raise UnsupportedCommandError(self, op)
esptool.util.UnsupportedCommandError: Invalid (unsupported) command 0x7
```

### More Information

We debug the CH343 Chip Driver on MacOS. Sometime we can not flash the ESP32C3 and receive 010504005560000001050000. This mean the chip send ROM_INVALID_RECV_MSG = 0x05. For compare, we looked at the output of a normal burn. It is 010704005560000000000000. Do you have any idea about this error? And what real mean we received invalid msg？

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-10-13:
Hello @yuruxuan,

We test esptool with a multitude of different combinations (changing OS, host machine, ESP chip type, USB-to-UART bridge, ...). Sadly, it is not possible to test every possible one. 

esptool is heavily dependent on a working serial connection, which includes a working USB-to-UART chip and drivers. Most likely there is a bug in the CH343 drivers on macOS. For instance, we have encountered faulty CH9102F drivers. This is why we use thoroughly tested end verify USB-to-UART chips on our devkits.

The `ROM_INVALID_RECV_MSG = 0x05` (meaning the format of the received message is invalid) return value from ROM is probably caused by receiving mangled data. This won't be visible in the esptool trace, but should be detectable when you capture the communication with a logic analyzer and compare it with a successful one. Another thing to try is to analyze the raw USB packets and see, if there is anything fishy.

I am sorry, but I don't have an immediate solution for this. Also, I believe this shouldn't be fixed in esptool, but on the side of drivers.

closed 2022-10-29 by yuruxuan

yuruxuan, 2022-10-29:
You are right. I communicate with CH343 MacOS driver engineer. He send me new version driver pkg and fixed this issue.