espressif/esptool #791: sonoff_nspanel

hervefox, 2022-10-20:
### Operating System

Ubuntu 22.04.1 LTS

### Esptool Version

v4.3

### Python Version

Python 2.7.18

### Chip Description

ESP32-D0WD-V3 (revision v3.0)

### Device Description

nspanel eur 
[image]

### Hardware Configuration

GPIO00	None
GPIO01	None
GPIO02	None
GPIO03	None
GPIO04	Output Lo
GPIO05	None
GPIO09	None
GPIO10	None
GPIO12	None
GPIO13	None
GPIO14	Button 1
GPIO15	None
GPIO16	None
GPIO17	None
GPIO18	None
GPIO19	Relay 2
GPIO20	None
GPIO21	Buzzer
GPIO22	Relay 1
GPIO23	User
GPIO24	None
GPIO25	None
GPIO26	None
GPIO27	Button 2
GPIO6	None
GPIO7	None
GPIO8	None
GPIO11	None
GPIO32	None
GPIO33	None
GPIO34	None
GPIO35	None
GPIO36	None
GPIO37	None
GPIO38	ADC Temperature
GPIO39	None

### How is Esptool Run

terminal 

### Full Esptool Command Line that Was Run

esptool read_flash 0 0x400000 nsp.bin

### Esptool Output

```shell
./esptool read_flash 0 0x400000 nsp.bin
esptool.py v4.3
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting...............
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 44:17:93:7c:15:38
Uploading stub...
Running stub...
Stub running...
```

### Other Steps to Reproduce

i can wait one hour, nothing . always running....

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-10-24:
Hello @hervefox,

please try the same approach with the `--no-stub` option (e.g. `./esptool --no-stub read_flash 0 0x400000 nsp.bin`).

hervefox, 2022-10-27:
Thanks , it's ok with no-stub , have a good day

closed 2022-10-27 by hervefox