espressif/esptool #792: A fatal error occurred: Timed out waiting for packet header...using a MaxLinear Quad UART

jwyoungster, 2022-11-02:
### Operating System

windows

### Esptool Version

esptool_py arduino add on  and direct command prompts

### Python Version

na

### Chip Description

ESP32-WROOM-32uE

### Device Description

custom board with the ESP32-wroom-32uE connected to a MaxLinear XR21B1424 Quad uart

### Hardware Configuration

have the RTS/DTS lines as is per all ESP32 dev boards (2 NPN's ..one collector to EN pin, the other collector to the GPIO 0 (pin 25 on the esp32)
tx rx from esp32 (pins 34/35) to the XR21B  
[vstsom_esptool_trace1.docx](https://github.com/espressif/esptool/files/9923775/vstsom_esptool_trace1.docx)

### How is Esptool Run

arduino interface AND command prompt direct

### Full Esptool Command Line that Was Run

see the attached doc...

### Esptool Output

```shell
see attached trace..it shows the difference between running command on ADAFRUIT feather  and running command on my cusotm board.
One thought I had was the diff between the Silcon Labs CP21x uart/usb bridge and the MaxLienar XR21B1424 registry info.....but couldnt find anything in registery ???
```

### More Information

scope shows the EN, GPIO0, TX and RX lines are almost identical from FEATHRE to custom
see pics....once that reset goes high, the ESP transmits the same burst of data.....on both feather and custom
[image]
[image]

### Other Steps to Reproduce

tried arduino, tried command prompt, changed baud rates, connected 10uF to ESP EN line, 
[esptool_info.docx](https://github.com/espressif/esptool/files/9923836/esptool_info.docx)

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2022-11-04:
You are using version 2.6 which is unsupported at the moment. Even if we could reproduce this, we cannot provide a patch for it. 

Please try the latest 4.3 version. A lot has changed since 2.6.

If you still encounter the issue then please send the logs in form of text where one can search for relevant parts. It is very inconvenient to analyze error messages sent as screenshots inside Word documents. Thanks!

radimkarnis, 2022-11-23:
@jwyoungster have you been able to try the latest version of esptool? V4.4 has been released recently.

radimkarnis, 2022-12-05:
Closing for inactivity, this is probably not relevant anymore. Feel free to reopen if you feel like further discussion is needed.

closed 2022-12-05 by radimkarnis