espressif/esptool #793: Writing at 0x0008257f... (100 %)Traceback (most recent call last):

LC1C, 2022-11-03:
### Operating System

Win10

### Esptool Version

esptool.py v4.3

### Python Version

Python311

### Chip Description

ESP8266

### Device Description

FTDI TTL

### How is Esptool Run

Windows terminal

### Full Esptool Command Line that Was Run

esptool.py --baud 115200 --port COM6 write_flash -fs 1MB -fm dio 0x0 fw-backup-1M.bin

### Esptool Output

```shell
esptool.py v4.3
Serial port COM6
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 68:c6:3a:a7:8f:44
Stub is already running. No upload is necessary.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x000fffff...
Compressed 1048576 bytes to 312715...
Writing at 0x0008257f... (100 %)Traceback (most recent call last):
  File "C:\Users\x\AppData\Local\Programs\Python\Python311\Scripts\esptool.py-script.py", line 33, in <module>
    sys.exit(load_entry_point('esptool==4.3', 'console_scripts', 'esptool.py')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\x\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\__init__.py", line 1022, in _main
    main()
  File "C:\Users\x\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\__init__.py", line 824, in main
    operation_func(esp, args)
  File "C:\Users\x\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\cmds.py", line 556, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\x\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\loader.py", line 109, in inner
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\x\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\loader.py", line 988, in flash_defl_block
    self.check_command(
  File "C:\Users\x\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\loader.py", line 407, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\x\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\loader.py", line 376, in command
    p = self.read()
        ^^^^^^^^^^^
  File "C:\Users\x\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\loader.py", line 308, in read
    return next(self._slip_reader)
           ^^^^^^^^^^^^^^^^^^^^^^^
StopIteration
```

### More Information

Also it was run with that command "esptool.py --baud 115200 --port COM6 write_flash 0x0 fw-backup-1M.bin"
Output same error as described at Esptool Output.

Python version issue? 

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2022-11-04:
> Stub is already running. No upload is necessary.

Was this intended for some reason? Does the result change if you add `--before default_reset` to the arguments? Could you please also add `--trace` and send us the logs (in TXT format)?

> Python version issue?

Could you please try it with something older than 3.11? We don't have yet experience with 3.11. Did you just update your computer? Was the same board working before?

LC1C, 2022-11-04:
>Stub is already running. No upload is necessary.

>Was this intended for some reason?

It comes by default. I not intend for it. On -read_flash it says that stub is uploaded.

>Does the result change if you add --before default_reset to the arguments?

No. Same error.

>Could you please also add --trace and send us the logs (in TXT format)?

Is there any way to save all traced dump to file? Because in terminal window not fits all traced data.

>Python version issue?
>Could you please try it with something older than 3.11? We don't have yet experience with 3.11.

Downgraded to 3.8.10
Instead of error at ver. 3.11 "Writing at 0x0008257f... (100 %)Traceback (most recent call last):
File "C:\Users\x\AppData\L........"
I got error without -trace "Writing at 0x00086901... (100 %)
A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption."

with --trace " 3fd193f8c4b0d433 b216290c5972301d | ?......3..).Yr0.85c0 | ..
Traceback (most recent call last):
File "C:\Users\x\AppData\Local\Programs\Python\Python38........."

>Did you just update your computer?

No. after a while MS updates stop working previous python version. MS suggested to install 3.10 which not work at all. i manualy install 3.11 and disable 3.10 from MS app store.

>Was the same board working before?

Yes. I did copy from working esp8266.
Today also for figure out where coming that error i upload working arduino sketch to esp8266, read out and tried to upload back to check if it can be some reading errors from another esp8266

dobairoland, 2022-11-07:
> Is there any way to save all traced dump to file? Because in terminal window not fits all traced data.

Esptool doesn't allow that. However, you can redirect the output to file. Probably appending `> file.txt` to the command should do it in your terminal. Or probably there is an option for your terminal to increase the limit.

Thanks! We really need the full log. Just a couple lines doesn't help us.

> No. after a while MS updates stop working previous python version. MS suggested to install 3.10 which not work at all. i manualy install 3.11 and disable 3.10 from MS app store.

3.11 is very new and there are a lot of issues with it (not esptool). If I can suggest then 3.8 is the most tested version on Windows (because ESP-IDF environments deploys it for a couple of years now).

radimkarnis, 2022-12-05:
@LC1C is this still an issue or can we close this? btw, this could also be a result of an insufficient power supply.

LC1C, 2022-12-05:
> @LC1C is this still an issue or can we close this? 

Still issue and for now don't have so much time to investigate more on that fault.

>btw, this could also be a result of an insufficient power supply.

Yes, i know about caused problems with insufficient power. And i tried with different power sources but result was same.

besi, 2022-12-18:
I ran into similar issues flashing micropython to my ESP-01m. I was able to flash the firmware by manually adjusting the timeouts in `loader.py`. 

See #816.

radimkarnis, 2023-03-01:
As reported by @tyeth [here](https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/100#issuecomment-1422846463), this is probably a result of a slow environment. A possible solution is to [increase the timeout with a config file](https://docs.espressif.com/projects/esptool/en/latest/esp8266/esptool/configuration-file.html).

I will close this issue because there is not much more we can do.

closed 2023-03-01 by radimkarnis