espressif/esptool #797: ImportError on import espefuse line of custom python script

belezyakov, 2022-11-17:
### Operating System

Ubuntu 20.04.1

### Esptool Version

v4.3

### Python Version

Python 3.8.10

### Full Esptool Command Line that Was Run

import espefuse

### Esptool Output

```shell
Traceback (most recent call last):
  File "/home/.../flasher_esp.py", line 4, in <module>
    import espefuse
  File "/home/.../my_env_folder/esptest/lib/python3.8/site-packages/espefuse/__init__.py", line 13, in <module>
    import espefuse.efuse.esp32 as esp32_efuse
  File "/home/.../my_env_folder/esptest/lib/python3.8/site-packages/espefuse/efuse/esp32/__init__.py", line 1, in <module>
    from . import operations
  File "/home/.../my_env_folder/esptest/lib/python3.8/site-packages/espefuse/efuse/esp32/operations.py", line 17, in <module>
    from . import fields
  File "/home/.../my_env_folder/esptest/lib/python3.8/site-packages/espefuse/efuse/esp32/fields.py", line 16, in <module>
    from .. import base_fields
  File "/home/.../my_env_folder/esptest/lib/python3.8/site-packages/espefuse/efuse/base_fields.py", line 13, in <module>
    from bitstring import BitArray, BitString, CreationError
ImportError: cannot import name 'BitString' from 'bitstring' (/home/.../my_env_folder/esptest/lib/python3.8/site-packages/bitstring.py)
```

### What is the Expected Behaviour?

No error on importing espefuse

### More Information

It seems like installing ```esptool``` in raw environment using ```pip``` collects ```bitstring``` wheel of new version 4.0.0 or 4.0.1 which satisfies requirement ```subitstring>=3.1.6```, but incompatible with ```esptool```. Preinstalling ```bitstring==3.1.9``` eliminates the issue.

dobairoland, 2022-11-18:
Thanks @belezyakov for informing us. We will investigate....

The desired solution would be to keep compatibility with old and new bitstring as well.

belezyakov, 2022-11-21:
According to [bitstring release notes](https://bitstring.readthedocs.io/en/latest/release_notes.html) for v4.0.1:
```Removed ConstBitArray and BitString class aliases. Use Bits and BitStream instead.```

commit 16fa5841 (2022-11-29): fix(espefuse): Fix compatibility with Bitstring>=4

bitstring==4 dropped class alias for BitStream. BitString was just an
alias in all supported bitstring versions:
https://github.com/scott-griffiths/bitstring/blob/bitstring-3.1.6/bitstring.py#L4247

Closes #797
closed 2022-11-23 by espressif-bot