espressif/esptool #798: A fatal error occurred: Failed to connect to Espressif device: No serial data received

xiaoye-x, 2022-11-21:
### Operating System

aarch64 GNU/Linux

### Esptool Version

esptool.py v4.3 4.3

### Python Version

Python 3.9.2

### Chip Description

esp32

### Device Description

[image]

### Hardware Configuration

[image]

### How is Esptool Run

raspberry

### Full Esptool Command Line that Was Run

esptool.py -b 115200  --port /dev/ttyACM0 erase_flash

### Esptool Output

```shell
esptool.py v4.3
Serial port /dev/ttyACM0
Connecting......................................
```

### More Information

I try to press the boot button, the programmed led lights up, but the burn always fails

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-11-21:
Hello @xiaoye-x,

please try manually switching the board into the download mode by holding down the `IO0` button and pressing the `RST` button. Then, you can release the `IO0` button.

After that, a new serial port should appear. Try to locate the right port manually and see if the chip is really connected (e.g. with `lsusb`) or try esptool autodetection (autodetection is attempted if you don't specify the port - delete the `--port /dev/ttyACM0` part).

After that, running `esptool.py erase_flash` should work.

xiaoye-x, 2022-11-21:
Hello @radimkarnis 
Thank you very much, I followed your instructions and successfully wiped the flash and got to the download mode via reset-boot, now I need to find a suitable bin file for it to burn.
Do you have a recommended version, I'm testing the connection between esp32 and (Linux) armv7 via UART.

radimkarnis, 2022-11-21:
Cool! I am glad you managed to make it work.

I don't know what you mean by `a suitable bin file`. You can now program the board using ESP-IDF, Arduino, Circuitpython, or other frameworks. I would suggest reading through the [official devkit instructions](https://python4iots2.readthedocs.io/en/latest/iots2_setup/link_mu.html) to get started.

Since the issue with esptool has been solved, I am closing this ticket.

closed 2022-11-21 by radimkarnis