espressif/esptool #800: 4.4: pytest is failing

kloczek, 2022-11-22:
### Operating System

Linux/x86_64

### Esptool Version

4.4

### Python Version

3.8.15

### Full Esptool Command Line that Was Run

N/A

### Esptool Output

```shell
N/A
```

### What is the Expected Behaviour?

pytest should not fail.

### Other Steps to Reproduce

I'm packaging your module as an rpm package so I'm using the typical PEP517 based build, install and test cycle used on building packages from non-root account.
- `python3 -sBm build -w --no-isolation`
- because I'm calling `build` with `--no-isolation` I'm using during all processes only locally installed modules
- install .whl file in </install/prefix>
- run pytest with PYTHONPATH pointing to sitearch and sitelib inside </install/prefix>

Here is pytest output:
<details>

```console
+ PYTHONPATH=/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.4-2.fc35.x86_64/usr/lib64/python3.8/site-packages:/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.4-2.fc35.x86_64/usr/lib/python3.8/site-packages
+ /usr/bin/pytest -ra
=========================================================================== test session starts ============================================================================
platform linux -- Python 3.8.15, pytest-7.2.0, pluggy-1.0.0
rootdir: /home/tkloczko/rpmbuild/BUILD/esptool-4.4
collected 53 items / 2 errors

================================================================================== ERRORS ==================================================================================
__________________________________________________________________ ERROR collecting test/test_espefuse.py __________________________________________________________________
ImportError while importing test module '/home/tkloczko/rpmbuild/BUILD/esptool-4.4/test/test_espefuse.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib64/python3.8/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
test/test_espefuse.py:33: in <module>
    from bitstring import BitString
E   ImportError: cannot import name 'BitString' from 'bitstring' (/usr/lib/python3.8/site-packages/bitstring.py)
__________________________________________________________________ ERROR collecting test/test_esptool.py ___________________________________________________________________
ImportError while importing test module '/home/tkloczko/rpmbuild/BUILD/esptool-4.4/test/test_esptool.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib64/python3.8/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
test_esptool.py:32: in <module>
    import espefuse
../../../BUILDROOT/esptool-4.4-2.fc35.x86_64/usr/lib/python3.8/site-packages/espefuse/__init__.py:11: in <module>
    import espefuse.efuse.esp32 as esp32_efuse
../../../BUILDROOT/esptool-4.4-2.fc35.x86_64/usr/lib/python3.8/site-packages/espefuse/efuse/esp32/__init__.py:1: in <module>
    from . import operations
../../../BUILDROOT/esptool-4.4-2.fc35.x86_64/usr/lib/python3.8/site-packages/espefuse/efuse/esp32/operations.py:15: in <module>
    from . import fields
../../../BUILDROOT/esptool-4.4-2.fc35.x86_64/usr/lib/python3.8/site-packages/espefuse/efuse/esp32/fields.py:14: in <module>
    from .. import base_fields
../../../BUILDROOT/esptool-4.4-2.fc35.x86_64/usr/lib/python3.8/site-packages/espefuse/efuse/base_fields.py:11: in <module>
    from bitstring import BitArray, BitString, CreationError
E   ImportError: cannot import name 'BitString' from 'bitstring' (/usr/lib/python3.8/site-packages/bitstring.py)
========================================================================= short test summary info ==========================================================================
ERROR test/test_espefuse.py
ERROR test/test_esptool.py
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 2 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
============================================================================ 2 errors in 0.49s =============================================================================
```
</details>

Here is list of installed modules in build env
<details>

```console
Package         Version
--------------- --------------
appdirs         1.4.4
attrs           22.1.0
bitstring       4.0.1
Brlapi          0.8.3
build           0.9.0
cffi            1.15.1
contourpy       1.0.6
cryptography    38.0.1
cssselect       1.1.0
cycler          0.11.0
distro          1.8.0
dnspython       2.2.1
ecdsa           0.18.0
exceptiongroup  1.0.0
extras          1.0.0
fixtures        4.0.0
fonttools       4.38.0
gpg             1.17.1-unknown
iniconfig       1.1.1
kiwisolver      1.4.4
libcomps        0.1.19
louis           3.23.0
lxml            4.9.1
matplotlib      3.6.2
numpy           1.23.1
olefile         0.46
packaging       21.3
pbr             5.9.0
pep517          0.13.0
Pillow          9.3.0
pip             22.3.1
pluggy          1.0.0
ply             3.11
pycparser       2.21
pyelftools      0.28
PyGObject       3.42.2
pyparsing       3.0.9
pyserial        3.5
pytest          7.2.0
python-dateutil 2.8.2
reedsolo        1.5.4
rpm             4.17.0
scour           0.38.2
setuptools      65.6.0
six             1.16.0
testtools       2.5.0
tomli           2.0.1
wheel           0.38.4
```
</details>

kloczek, 2022-11-22:
After add those two files in which pytest is failing to --ignore list other 4 units are failing
<details>

```console
self._lib.Cryptography_HAS_PROVIDERS
                and errors[0]._lib_reason_match(
                    self._lib.ERR_LIB_PROV,
                    self._lib.PROV_R_BAD_DECRYPT,
                )
            )
        ):
            raise ValueError("Bad decrypt. Incorrect password?")

        elif any(
            error._lib_reason_match(
                self._lib.ERR_LIB_EVP,
                self._lib.EVP_R_UNSUPPORTED_PRIVATE_KEY_ALGORITHM,
            )
            for error in errors
        ):
            raise ValueError("Unsupported public key algorithm.")

        else:
            errors_with_text = binding._errors_with_text(errors)
>           raise ValueError(
                "Could not deserialize key data. The data may be in an "
                "incorrect format, it may be encrypted with an unsupported "
                "algorithm, or it may be an unsupported key type (e.g. EC "
                "curves with explicit parameters).",
                errors_with_text,
            )
E           ValueError: ('Could not deserialize key data. The data may be in an incorrect format, it may be encrypted with an unsupported algorithm, or it may be an unsupported key type (e.g. EC curves with explicit parameters).', [_OpenSSLErrorWithText(code=75497580, lib=9, reason=108, reason_text=b'error:0480006C:PEM routines::no start line')])

/usr/lib64/python3.8/site-packages/cryptography/hazmat/backends/openssl/backend.py:1248: ValueError
--------------------------------------------------------------------------- Captured stdout call ---------------------------------------------------------------------------
Verifying 21696 bytes of data
Signature is valid
Signature block 0 is valid (RSA).
Signature block 0 verification successful with /home/tkloczko/rpmbuild/BUILD/esptool-4.4/test/secure_images/rsa_secure_boot_signing_pubkey.pem (RSA).
Signature block 1 invalid. Skipping.
Signature block 2 invalid. Skipping.
Signature block 0 is valid (ECDSA).
Signature block 0 verification successful with /home/tkloczko/rpmbuild/BUILD/esptool-4.4/test/secure_images/ecdsa_secure_boot_signing_pubkey.pem (ECDSA).
_______________________________________________________________ TestSigning.test_generate_and_extract_key_v2 _______________________________________________________________

self = <test_espsecure.TestSigning object at 0x7f4c24fefe20>

    def test_generate_and_extract_key_v2(self):
        with tempfile.TemporaryDirectory() as keydir:
            # keyfile cannot exist before generation -> tempfile.NamedTemporaryFile()
            # cannot be used for keyfile
            keyfile_name = os.path.join(keydir, "key.pem")

            # We need to manually delete the keyfile as we are iterating over
            # different schemes with the same keyfile so instead of using addCleanup,
            # we remove it using os.remove at the end of each pass
            for scheme in ["rsa3072", "ecdsa192", "ecdsa256"]:
                args = self.GenerateKeyArgs("2", scheme, keyfile_name)
                espsecure.generate_signing_key(args)

                with tempfile.NamedTemporaryFile() as pub_keyfile, open(
                    keyfile_name, "rb"
                ) as keyfile:
                    args = self.ExtractKeyArgs("2", keyfile, pub_keyfile)
>                   espsecure.extract_public_key(args)

/home/tkloczko/rpmbuild/BUILD/esptool-4.4/test/test_espsecure.py:515:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.4-2.fc35.x86_64/usr/lib/python3.8/site-packages/espsecure/__init__.py:694: in extract_public_key
    sk = _load_sbv2_signing_key(args.keyfile.read())
[... 48 lines cut ...]

        else:
            errors_with_text = binding._errors_with_text(errors)
>           raise ValueError(
                "Could not deserialize key data. The data may be in an "
                "incorrect format, it may be encrypted with an unsupported "
                "algorithm, or it may be an unsupported key type (e.g. EC "
                "curves with explicit parameters).",
                errors_with_text,
            )
E           ValueError: ('Could not deserialize key data. The data may be in an incorrect format, it may be encrypted with an unsupported algorithm, or it may be an unsupported key type (e.g. EC curves with explicit parameters).', [_OpenSSLErrorWithText(code=503841036, lib=60, reason=524556, reason_text=b'error:1E08010C:DECODER routines::unsupported'), _OpenSSLErrorWithText(code=134217857, lib=16, reason=129, reason_text=b'error:08000081:elliptic curve routines::unknown group'), _OpenSSLErrorWithText(code=134217847, lib=16, reason=119, reason_text=b'error:08000077:elliptic curve routines::ec group new by name failure'), _OpenSSLErrorWithText(code=134742032, lib=16, reason=524304, reason_text=b'error:08080010:elliptic curve routines::EC lib'), _OpenSSLErrorWithText(code=109052072, lib=13, reason=168, reason_text=b'error:068000A8:asn1 encoding routines::wrong tag'), _OpenSSLErrorWithText(code=109576458, lib=13, reason=524554, reason_text=b'error:0688010A:asn1 encoding routines::nested asn1 error'), _OpenSSLErrorWithText(code=109576458, lib=13, reason=524554, reason_text=b'error:0688010A:asn1 encoding routines::nested asn1 er [... cut ...]
FAILED test/test_espsecure.py::TestSigning::test_verify_signature_public_key - ValueError: ('Could not deserialize key data. The data may be in an incorrect format, it may be encrypted with an unsupported algorithm, or it may be an unsupported ke...
FAILED test/test_espsecure.py::TestSigning::test_generate_and_extract_key_v2 - ValueError: ('Could not deserialize key data. The data may be in an incorrect format, it may be encrypted with an unsupported algorithm, or it may be an unsupported ke...
====================================================================== 4 failed, 49 passed in 11.26s =======================================================================
```
</details>

dobairoland, 2022-11-22:
Why do you have `bitstring       4.0.1`?

See the dependency: https://github.com/espressif/esptool/blob/master/setup.py#L120

radimkarnis, 2022-11-22:
Hello @kloczek,

the version of bitstring is [currently restricted](https://github.com/espressif/esptool/blob/master/setup.py#L120) to `bitstring>=3.1.6,<4"`. You have an incompatible `bistring==4.0.1` in your dev environment.

kloczek, 2022-11-22:
OK thx to lat me know.
Do you have any plans to update for latest `bitstring`? 🤔

kloczek, 2022-11-22:
OK I'll close this ticket and subscribe to https://github.com/espressif/esptool/issues/797.

closed 2022-11-22 by kloczek

dobairoland, 2022-11-22:
Yes, you can follow https://github.com/espressif/esptool/issues/797 for that.

kloczek, 2022-11-23:
Just tested last commit and looks like many units are failing because missing images/ directorya and lack of access to /dev/ttyUSB0
<details>

```console
+ PYTHONPATH=/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.4-2.fc35.x86_64/usr/lib64/python3.8/site-packages:/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.4-2.fc35.x86_64/usr/lib/python3.8/site-packages
+ /usr/bin/pytest -ra
=========================================================================== test session starts ============================================================================
platform linux -- Python 3.8.15, pytest-7.2.0, pluggy-1.0.0
rootdir: /home/tkloczko/rpmbuild/BUILD/esptool-4.4
collected 199 items

test/test_espefuse.py .......................ssssssssssss......sss.sss..ss..ss.....ssssss.ss........ss..                                                             [ 41%]
test/test_espsecure.py ......F...FF.F.........                                                                                                                       [ 52%]
test/test_esptool.py FFxFFFFFFFFFFFF...FFFFFFFsFFFF.FFFFFsFFFFFFFFFFsFFFEEEFsFFFFFFFF                                                                                [ 84%]
test/test_image_info.py .......                                                                                                                                      [ 88%]
test/test_imagegen.py ..............                                                                                                                                 [ 95%]
test/test_merge_bin.py ........                                                                                                                                      [ 99%]
test/test_modules.py .                                                                                                                                               [100%]

================================================================================== ERRORS ==================================================================================
________________________________________________ ERROR at setup of TestKeepImageSettings.test_keep_does_not_change_settings ________________________________________________

self = <class 'test_esptool.TestKeepImageSettings'>

    @classmethod
    def setup_class(self):
        super(TestKeepImageSettings, self).setup_class()
        self.BL_IMAGE = BL_IMAGES[arg_chip]
        self.flash_offset = (
            0x1000 if arg_chip in ("esp32", "esp32s2") else 0
        )  # bootloader offset
>       with open(self.BL_IMAGE, "rb") as f:
E       FileNotFoundError: [Errno 2] No such file or directory: 'images/bootloader_esp32.bin'

/home/tkloczko/rpmbuild/BUILD/esptool-4.4/test/test_esptool.py:695: FileNotFoundError
-------------------------------------------------------------------------- Captured stdout setup ---------------------------------------------------------------------------

**************************************************
__________________________________________________ ERROR at setup of TestKeepImageSettings.test_detect_size_changes_size ___________________________________________________

self = <class 'test_esptool.TestKeepImageSettings'>

    @classmethod
    def setup_class(self):
        super(TestKeepImageSettings, self).setup_class()
        self.BL_IMAGE = BL_IMAGES[arg_chip]
        self.flash_offset = (
            0x1000 if arg_chip in ("esp32", "esp32s2") else 0
        )  # bootloader offset
>       with open(self.BL_IMAGE, "rb") as f:
E       FileNotFoundError: [Errno 2] No such file or directory: 'images/bootloader_esp32.bin'

/home/tkloczko/rpmbuild/BUILD/esptool-4.4/test/test_esptool.py:695: FileNotFoundError
_________________________________________________ ERROR at setup of TestKeepImageSettings.test_explicit_set_size_freq_mode _________________________________________________

self = <class 'test_esptool.TestKeepImageSettings'>

    @classmethod
    def setup_class(self):
        super(TestKeepImageSettings, self).setup_class()
        self.BL_IMAGE = BL_IMAGES[arg_chip]
        self.flash_offset = (
            0x1000 if arg_chip in ("esp32", "esp32s2") else 0
        )  # bootloader offset
>       with open(self.BL_IMAGE, "rb") as f:
E       FileNotFoundError: [Errno 2] No such file or directory: 'images/bootloader_esp32.bin'

/home/tkloczko/rpmbuild/BUILD/esptool-4.4/test/test_esptool.py:695: FileNotFoundError
================================================================================= FAILURES =================================================================================
______________________________________________________________________ TestSigning.test_sign_v2_data _______________________________________________________________________
[... 579 lines cut ...]
FAILED test/test_esptool.py::TestErase::test_large_region_erase - subprocess.CalledProcessError: Command '['/usr/bin/python3', '/home/tkloczko/rpmbuild/BUILD/esptool-4.4/test/../esptool/__init__.py', '--chip', 'esp32', '--port', '/de...
FAILED test/test_esptool.py::TestSectorBoundaries::test_end_sector - subprocess.CalledProcessError: Command '['/usr/bin/python3', '/home/tkloczko/rpmbuild/BUILD/esptool-4.4/test/../esptool/__init__.py', '--chip', 'esp32', '--port', '/de...
FAILED test/test_esptool.py::TestSectorBoundaries::test_end_sector_uncompressed - subprocess.CalledProcessError: Command '['/usr/bin/python3', '/home/tkloczko/rpmbuild/BUILD/esptool-4.4/test/../esptool/__ini [... cut ...]
FAILED test/test_esptool.py::TestReadWriteMemory::test_read_write_memory_rom - esptool.util.FatalError: Could not open /dev/ttyUSB0, the port doesn't exist
FAILED test/test_esptool.py::TestReadWriteMemory::test_read_write_memory_stub - esptool.util.FatalError: Could not open /dev/ttyUSB0, the port doesn't exist
========================================== 56 failed, 103 passed, 36 skipped, 1 xfailed, 1 warning, 3 errors in 89.38s (0:01:29) ===========================================
```
</details>

Do I need connected esp device over serial bus to pass test suite?