espressif/esptool #801: Trying to use esptool with a ZBBridge fails every time

AndreiGr, 2022-11-25:
### Operating System

Window 10

### Esptool Version

Esptool 4.4

### Python Version

Python 3.10.4

### Chip Description

ESP8266EX

### Device Description

I am using a Sonoff Zigbee Bridge.

### Hardware Configuration

I am using the following USB to Serial adapter: https://ardushop.ro/ro/home/55-usb-to-ttl-converter-uart-ch340g-ch340-33v-5v-switch.html

### How is Esptool Run

Windows PowerShell

### Full Esptool Command Line that Was Run

esptool.py -p COM17 -b 9600 --trace read_mac

### Esptool Output

```shell
esptool.py v4.4
Serial port COM17
Connecting...
Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type...
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 98:f4:ab:e3:63:da
Uploading stub...
Running stub...
Stub running...
Traceback (most recent call last):
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\Scripts\esptool.py-script.py", line 33, in <module>
    sys.exit(load_entry_point('esptool==4.4', 'console_scripts', 'esptool.py')())
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\__init__.py", line 1026, in _main
    main()
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\__init__.py", line 826, in main
    operation_func(esp, args)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\cmds.py", line 956, in read_mac
    mac = esp.read_mac()
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\targets\esp8266.py", line 136, in read_mac
    mac0 = self.read_reg(self.ESP_OTP_MAC0)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 690, in read_reg
    val, data = self.command(
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 376, in command
    p = self.read()
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 308, in read
    return next(self._slip_reader)
StopIteration

Output with trace: https://pastebin.com/kkB9bNaC
```

### More Information

I am trying to flash Tasmota on a Zigbee Bridge from Sonoff.
I have tried every configuration and application (Tasmotizer, ESPFlashTool, ESPTool) to flash it and it keeps failing everytime.

The errors I get in Esptool are the most complete, but I still have no idea about what to do next in order to make it work.

The error provided above is in the configuration that got me the furthest.
The configurations I have used and the errors I got are the following:

1. Jumper on VCC&5V on the adapter
    Error: Write timeout
2. Jumper on VCC&3.3V on the adapter
    Error: No serial data received
3. No jumper on the adapter
    Error on stub running. This is the error pasted above in full, with trace here: https://pastebin.com/kkB9bNaC.

A few pictures with the board and configurations here: https://imgur.com/a/AofGpop

Thank you in advance for your help.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-11-25:
Hi @AndreiGr,

Since you used Tasmotizer and ESPFlashTool and they all fail, it seems like this is not an esptool problem, but a hardware one. This might include a faulty flash chip, connection, or USB to Serial adapter.

Some lines in the logs suggest a problem with the flash chip:
* `Unable to verify flash chip connection (No serial data received.`
* `Unable to perform XMC flash chip startup sequence`

Can you try running `esptool.py flash_id` and paste the output?

Also, you can try all of the commands with the `--no-stub` option.

Please try and let me know how it went.

AndreiGr, 2022-11-25:
Hi @radimkarnis.
This is on the config with no jumper
```
PS D:\Tasmota\esptool-4.4> esptool.py -p COM17 flash_id
esptool.py v4.4
Serial port COM17
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 98:f4:ab:e3:63:da
Uploading stub...
Running stub...
Stub running...
Traceback (most recent call last):
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\Scripts\esptool.py-script.py", line 33, in <module>
    sys.exit(load_entry_point('esptool==4.4', 'console_scripts', 'esptool.py')())
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\__init__.py", line 1026, in _main
    main()
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\__init__.py", line 826, in main
    operation_func(esp, args)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\cmds.py", line 1008, in flash_id
    flash_id = esp.flash_id()
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 868, in flash_id
    return self.run_spiflash_command(SPIFLASH_RDID, b"", 24)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 1250, in run_spiflash_command
    old_spi_usr = self.read_reg(SPI_USR_REG)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 690, in read_reg
    val, data = self.command(
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 376, in command
    p = self.read()
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 308, in read
    return next(self._slip_reader)
StopIteration

PS D:\Tasmota\esptool-4.4> esptool.py -p COM17 --no-stub flash_id
esptool.py v4.4
Serial port COM17

A fatal error occurred: Could not open COM17, the port doesn't exist
PS D:\Tasmota\esptool-4.4> esptool.py -p COM17 --no-stub flash_id
esptool.py v4.4
Serial port COM17
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 98:f4:ab:e3:63:da
Enabling default SPI flash mode...

A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption.```

AndreiGr, 2022-11-25:
If I try to put the jumper on 3.3V, which should be how this should work, I just get this:
```
PS D:\Tasmota\esptool-4.4> esptool.py -p COM17 --no-stub flash_id
esptool.py v4.4
Serial port COM17
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html```

AndreiGr, 2022-11-27:
**Update:** I managed to make it somewhat work, by connecting the Z_RST pin to GND. But now I run into another issue

I managed to read&backup the existing firmware, erase the flash, but when I try to upload the new firmware, it gets stuck at 9%, for some reason (both in ESPTool and in Tasmotizer).

Do you have any ideas/solutions for this?

Here is the ESPTool command & output:
```
PS D:\Tasmota\esptool-4.4> esptool.py --port COM13 write_flash -fm dout 0x0 tasmota-zbbridge.bin
esptool.py v4.4
Serial port COM13
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 98:f4:ab:e3:63:da
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x000b2fff...
Compressed 731904 bytes to 512145...
Writing at 0x0000b400... (9 %)Traceback (most recent call last):
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\Scripts\esptool.py-script.py", line 33, in <module>
    sys.exit(load_entry_point('esptool==4.4', 'console_scripts', 'esptool.py')())
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\__init__.py", line 1026, in _main
    main()
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\__init__.py", line 826, in main
    operation_func(esp, args)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\cmds.py", line 556, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 109, in inner
    return func(*args, **kwargs)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 991, in flash_defl_block
    self.check_command(
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 407, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 376, in command
    p = self.read()
  File "C:\Users\andre\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool-4.4-py3.10.egg\esptool\loader.py", line 308, in read
    return next(self._slip_reader)
StopIteration
```

radimkarnis, 2022-11-28:
Hi @AndreiGr,
I don't know why writing to flash fails at 9%. I can recommend you to try a lower baudrate or the `--no-stub` option.
There might be other reasons for the failure - such as insufficient power supply. Please read the mentioned troubleshooting guide and try everything.

radimkarnis, 2022-12-05:
Closing this issue as resolved. Feel free to reopen if needed!

closed 2022-12-05 by radimkarnis