espressif/esptool #803: ESPEFUSE doesn't close COM port when throws an exception when called from Python

reykada, 2022-12-01:
### Operating System

Windows 10 Pro 21H2

### Esptool Version

v4.3

### Python Version

Python 3.9.6

### Chip Description

ESP32-D0WD-V3 (revision v3.0)

### Device Description

NodeMCU ESP-32 v1.1

### How is Esptool Run

No IDE

### Full Esptool Command Line that Was Run

From Python script

### Esptool Output

```shell
An exception occurred: BLOCK2 is read-protected. Burn in this case may damage an already written value.(use '--force-write-always' option to ignore it)
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

reykada, 2022-12-01:
Hi

I use Python script which includes several subsequent calls of ESPTOOL and ESPEFUSE for production purpose.
The problem is if I do the calls like:
```
command = ['--port', comport, 'burn_key', 'flash_encryption', path_to_flash_enc_key, '--do-not-confirm']
espefuse.main(command)
```
and ESPEFUSE throws an exception which leads to my Python script termination. But I need the further ESPTOOL calls in my script to be executed.

So I decided to add exception handler:

```
    command = ['--port', comport, 'burn_key', 'flash_encryption', path_to_flash_enc_key, '--do-not-confirm']
    try:
        espefuse.main(command)
    except BaseException as error:
        print(f'An exception occurred: {format(error)}')
```

I can get and print an exception in this case:

`An exception occurred: BLOCK2 is read-protected. Burn in this case may damage an already written value.(use '--force-write-always' option to ignore it)`

But when ESPEFUSE is called further in my Python script I get the following error:

```
An exception occurred: Could not open COM3, the port doesn't exist
Please make sure that you have specified the right port with the --port argument
```
So looks like ESPEFUSE doesn't release the COM port when throw an exception.

ESPTOOL version: 4.3

Are there any workaround?
Regards

radimkarnis, 2022-12-05:
Hi @reykada,
thanks for the detailed report! I will look into it and provide a fix.

radimkarnis, 2022-12-05:
Hello @reykada,
could you please verify this patch: [close_port.txt](https://github.com/espressif/esptool/files/10154821/close_port.txt) ?

To try it, please clone this repo and install it in development mode with:
```
cd esptool
pip install -e .
```
You can try your script at this point - it should still fail.

Then please download `close_port.txt` and apply the patch in the `esptool` folder with `git am close_port.txt`. When you try your script now, it should work.

Thank you!

reykada, 2022-12-05:
Hi @radimkarnis 

It does work! The port is being closed when ESPEFUSE throws an exception.
Thank you so much for the quick fix!

closed 2022-12-06 by espressif-bot

radimkarnis, 2022-12-06:
@reykada thanks for your help and the report! The fix has been merged to master.