espressif/esptool #804: Flash error with USB-built-in-Serial in wsl2

Forairaaaaa, 2022-12-03:
### Operating System

Ubuntu 20.04.5 LTS ( WSL2 WIN11-22H2 )

### Esptool Version

v4.4

### Python Version

Python 3.8.10

### Chip Description

ESP32-S3-WROOM-1-N16R8

### Device Description

DIY board with ESP32-S3-WROOM-1-N16R8, no hardware attached.

### Hardware Configuration

Minimum System, USB port connected to usb GPIO.

### How is Esptool Run

Linux terminal

### Full Esptool Command Line that Was Run

esptool esp32s3 -p /dev/ttyACMO -b 460800 --before=default_reset--after=hard_reset write flash --flash_mode dio --flash_freg 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin

### Esptool Output

```shell
esptool.py v4.4
Serial port /dev/ttyACMO
Connecting.Chip is ESP32-S3 (revision v0.1)
Features: WiFi. BLE
Crystal is 40MHz
MAC: f4:12:fa:cf:32:f4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed .
Confiquring flash size..
FLash will be erased from 0x00000000 to 0x00005fff
Flash will be erased from 0x00010000 to 0x00041fff
Flash will be erased from 0x00008000 to 0x00008fff
Compressed 20816 bvtes to 13274...
writing at 0x00000000... (100 %)
Wrote 20816 bytes (13274 compressed) at 0x00000000 in 0.4 seconds (effective 402.5 kbit/s)...
A fatal error occurred: Packet content transfer stopped (received 24 bytes)
CMake Error at run_serial tool.cmake:55 (message):
components/esptool_py/esptool/esptool.py;--chip;esp32s3failed
/home/parker/.espressif/python_env/idf5.0_py3.8_env/bin/python;;/home/parker/esp/esp-idf
ATIED. CMalefiles/flashcd /home/parker/esp/esp-idf/components/esptool_py &s /usr/bin/cmake -D IDF_PATH=/home/parker/esp/esp-idf -D "SERIAL_TOOL=/home/parker/.espressif/pvthon-env/idf5.0 py3.8 env/bin/ovthponents/esptool_py/esptool/esptool.py;--chip;esp32s3"-D "S: :/home/parker/esp/esp-idf/conRIAL_TOOL_ARGS=--before=default-reset;--after=hard-reset;write-flash;@flash-args" -D WORKIG_DIRECTORY=/home/parker/esp/hello-world/build -p /home/parker/esp/esp-idf/components/esptcol_pv/run_serial tool.cmakeninja: build stopped: subcommand failed.ninia failed with exit code 1. output of the command is in the /home/parker/esp/hello worlo/build/log/idf_py_stderr_output_5282 and /home/parker/esp/hello_world/build/log/idf_py_stdout output 5282
```

### More Information

I recently setup the **IDF v5.0** stable envirment, and try to flash the example program with the **USB built-in Serial** like I allways did, but error occurred: 
[image]
I tried flash again in **download mode**, **switch usb port**, **reboot** my wsl and computer. None of them worked.
**But it works** by flashing with **uart0** like the traditional way. It seems to be some issues comes with the **built-in usb serial controller**?

Then I tried again in **IDF 4.4.3** stable envirment, which is using **esptool** **v3.3.2**. And every thing **goes well** with the **built-in usb-serial,** no matter **downloading** or **logging**:
[image]
So I wondered if it's the **update from esptool v3.3.2 to esptool v4.4** causes this issue?
I sincerely hope to get some helps.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

Forairaaaaa, 2022-12-03:
By the way, I use **usbip** to attach my board into wsl2, they're all updated.

radimkarnis, 2022-12-03:
Hello @Forairaaaaa,
thanks for reporting this!

I have a suspicion this might be caused by [one of the new features](https://github.com/espressif/esptool/commit/dccf4dff0a678f5efccdd73f112e47d76c66e79c) implemented in esptool v4.4.

Could you please help us to verify this by [installing esptool](https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html) and trying to run the same command in v4.3 and v4.4? You can install a specific version by running `pip install esptool==4.3`.

This feature raises the CPU clock speed when flashing ESP32-S3 over USB-JTAG/Serial. This results in faster data transfers but also increased power consumption. If your power supply is not able to keep up with the increased power demands, the chip might be browning out.

If you disable the flasher stub (either in menuconfig or by using `--no-stub`), the transfer should succeed, could you please also try this?

Thank you!

Forairaaaaa, 2022-12-04:
> Hello @Forairaaaaa, thanks for reporting this!
>
> I have a suspicion this might be caused by [one of the new [...]

Hello @radimkarnis ,
Thank you for helping me out!
I tried flash with **esptool v4.3**, and **disable the flasher stub**, they both make it works!
As for power supply, I'll try to find an other board to try again then.

But I found an other strange thing, when I do the same thing in **windows** instead of wsl, it works fine with esptool v4.4, the command is like this:

```
esptool esp32s3 -p COM5 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x0 bootloader/bootloader.bin 0x10000 test.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.4
Serial port COM5
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:cf:32:f4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00041fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 21888 bytes to 13804...
Writing at 0x00000000... (100 %)
Wrote 21888 bytes (13804 compressed) at 0x00000000 in 0.4 seconds (effective 421.2 kbit/s)...
Hash of data verified.
Compressed 200896 bytes to 104968...
Writing at 0x00010000... (14 %)
Writing at 0x0001c0a0... (28 %)
Writing at 0x00021c0b... (42 %)
Writing at 0x00027b9a... (57 %)
Writing at 0x00030853... (71 %)
Writing at 0x00038854... (85 %)
Writing at 0x0003e5ab... (100 %)
Wrote 200896 bytes (104968 compressed) at 0x00010000 in 1.4 seconds (effective 1110.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 350.1 kbit/s)...
Hash of data verified.

Leaving...
```
I wonder if wsl's usb mechanism may also be related?
Thanks again for helping me solve the problem!

Forairaaaaa, 2022-12-05:
> 

Thanks a lot!

closed 2022-12-05 by Forairaaaaa