espressif/esptool #806: esp tool fails with exception on Mac M1

MishraShivendra, 2022-12-06:
### Operating System

Ventura 13.1 Mac M1

### Esptool Version

v4.2.1

### Python Version

3.10.8

### Chip Description

AI-Thinker ESP32-CAM

### How is Esptool Run

Ardrino IDE

### Esptool Output

```shell
Sketch uses 1470797 bytes (46%) of program storage space. Maximum is 3145728 bytes.
Global variables use 65568 bytes (20%) of dynamic memory, leaving 262112 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.2.1
Serial port /dev/cu.usbserial-10
Connecting.....Traceback (most recent call last):
  File "esptool.py", line 34, in <module>
  File "esptool/__init__.py", line 1004, in _main
  File "esptool/__init__.py", line 790, in main
  File "esptool/loader.py", line 1108, in flash_set_parameters
  File "esptool/loader.py", line 406, in check_command
  File "esptool/loader.py", line 375, in command
  File "esptool/loader.py", line 307, in read
StopIteration
[4514] Failed to execute script 'esptool' due to unhandled exception!

Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.82MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 0c:dc:7e:62:bc:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Failed uploading: uploading error: exit status 1
```

### Other Steps to Reproduce

Open camwebserver example, compile and upload.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-12-06:
Hi @MishraShivendra,
I am sorry, but it is impossible to tell the issue just based on this report. Please provide more information:

1) Have you tried [running esptool manually](https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html)?
2) Have you tried the `--no-stub` option?
3) Have you tried any other chip or host? Do you believe this is a Mac M1 issue? If so, why?
4) Have you tried another esptool version?

There seems to be a warning about the crystal frequency:
```
WARNING: Detected crystal freq 41.82MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
```

This might be an indication of faulty hardware.

Please try more things and post the results here.

Jason2866, 2022-12-07:
I have seen this
```
WARNING: Detected crystal freq 41.82MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
```
on my M1 when not using a USB 2 hub or Dockingstation. It is probably related to #712

cyrillkuettel, 2022-12-08:
Confirming this issue, also on a Mac, but without M1 chip.

@radimkarnis I tried running manually `esptool.py --chip esp32 erase_flash` , this works fine. 
However, when I want to upload some code from Arduino IDE, I get this same `StopIteration` error
```bash
StopIteration
[20910] Failed to execute script 'esptool' due to unhandled exception!
```

Maybe this is a Arduino (Version `2.0.3`) specific issue. I tried running the exact same thing on another ESP32, there it worked fine.

MishraShivendra, 2022-12-10:
I think this is something to do with baud rate. Following command seems to work. we may close this issue, doesn't look like esp tool issue.

```
"/esptool" --chip esp32 --port "/dev/cu.usbserial-11" --baud 115200  --before default_reset --after hard_reset write_flash  -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 "CameraWebServer.ino.bootloader.bin" 0x8000 "CameraWebServer.ino.partitions.bin" 0xe000 "boot_app0.bin" 0x10000 "CameraWebServer.ino.bin"
esptool.py v4.2.1
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.82MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 0c:dc:7e:62:bc:20
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00178fff...
Compressed 18912 bytes to 13025...
Wrote 18912 bytes (13025 compressed) at 0x00001000 in 1.6 seconds (effective 94.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 117...
Wrote 3072 bytes (117 compressed) at 0x00008000 in 0.1 seconds (effective 255.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 458.2 kbit/s)...
Hash of data verified.
Compressed 1476576 bytes to 964585...
Wrote 1476576 bytes (964585 compressed) at 0x00010000 in 89.0 seconds (effective 132.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

closed 2022-12-10 by MishraShivendra

siteswapjuggler, 2022-12-23:
I got the same problem here and the culprit was anaconda and the base environment with python 3.9. It's not the first time that python environment come in conflict with the esp-core requirement perhaps there is something to do to avoid this confusion.

santiagax99, 2023-01-16:
Faced same issue and removing --baud 921600 from command line fixes the problem on Macbook M1 apple silicon

zumatt, 2023-08-04:
Faced the same issue with this error message:
```
esptool.py v4.2.1
Serial port /dev/cu.usbserial-110
Connecting......Traceback (most recent call last):
  File "esptool.py", line 34, in <module>
  File "esptool/__init__.py", line 1004, in _main
  File "esptool/__init__.py", line 790, in main
  File "esptool/loader.py", line 1108, in flash_set_parameters
  File "esptool/loader.py", line 406, in check_command
  File "esptool/loader.py", line 375, in command
  File "esptool/loader.py", line 307, in read
StopIteration
[20528] Failed to execute script 'esptool' due to unhandled exception!

Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 7c:87:ce:f8:c4:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Failed uploading: uploading error: exit status 1
```

I solved it, on my Macbook M1 Apple Silicon using Arduino IDE 2.1.1 , changing the upload speed via `Tools > Upload Speed` from `921600` to `115200`

I hope that this could help, cheers.

Sparsh123d, 2024-10-25:
When I change upload speed from 921600 to 115200, I get the following error while uploading:

A serial exception error occurred: Write timeout
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html