espressif/esptool #808: esp32s3 fails chip_id, erase_flash, etc

bamason2, 2022-12-08:
### Operating System

macOs 13.0.1

### Esptool Version

Esptool 4.4

### Python Version

Python 3.9.15

### Chip Description

esp32-s3

### Device Description

Adafruit feather esp32-s3

### How is Esptool Run

Terminal

### Full Esptool Command Line that Was Run

esptool.py --port /dev/tty.usbmodem101 --baud 115200 --chip esp32s3 chip_id

### Esptool Output

```shell
Serial port /dev/tty.usbmodem101
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:5c:92:c4
Uploading stub...
Running stub...

and the trace;

esptool.py --trace --port /dev/tty.usbmodem101 --baud 115200 --chip esp32s3 chip_id
esptool.py v4.4
Serial port /dev/tty.usbmodem101
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=2.000 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.002 Read 1 bytes: c0
TRACE +0.000 Read 27 bytes: 
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0           | ..... .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 42 bytes: 
    c001080400070712 2000000000c0c001 | ........ .......
    0804000707122000 000000c0c0010804 | ...... .........
    0007071220000000 00c0             | .... .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 28 bytes: 
    c001080400070712 2000000000c0c001 | ........ .......
    0804000707122000 000000c0         | ...... .....
[... 728 lines cut ...]
TRACE +0.000 Write 30 bytes: 
    c0000714000a0000 0004000000000000 | ................
    0000000000000000 001000ca3fc0     | ............?.
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 0107040012f4000000000000c0
TRACE +0.000 Received full packet: 0107040012f4000000000000
Running stub...
TRACE +0.000 command op=0x06 data len=8 wait_response=1 timeout=2.000 data=000000000c8a3740
TRACE +0.000 Write 18 bytes: 
    c000060800000000 00000000000c8a37 | ...............7
    40c0                              | @.
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 0106040012f4000000000000c0
TRACE +0.000 Received full packet: 0106040012f4000000000000
```

### More Information

Esptool runs but stops on 'running stub' indefinitely. Device is in boot mode. Have tried various options: --before no_reset_no_sync, no_reset --after no reset.  Tried various larger values of MEM_END_ROM_TIMEOUT up to 2.5. I don't think it is a comms problem since the Web Serial JS tool works fine (https://github.com/makermelissa/Adafruit_WebSerial_ESPTool), can enter REPL using rshell, etc. Also running the --no_stub option works fine but unable then to erase_flash.

Tried on multiple esp32-s3 all have same problem.  Esptool works fine on same setup with ESP32-D0WD-V3 and others with usb to serial (SIL 2104, etc). 

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-12-09:
Hi @bamason2,
ESP32-S3 is tested with esptool and works fine in the stub mode. We need to investigate more.

> Tried on multiple esp32-s3 all have same problem

Are these all the same revision `v0.1` ?

Also, can you please try older versions of esptool? Maybe we can try to pinpoint the commit that introduced a regression (if there is one).

Thanks!

Emill, 2022-12-13:
We got the same issue on 1 of 70 tried custom boards. On this single one failing board the problem is consistent, i.e. happens all the time. The flash is empty and has never been programmed.
Commit https://github.com/espressif/esptool/tree/0656f77d7980fe29c7969efd9f04b23083f22ec7 works but https://github.com/espressif/esptool/tree/dccf4dff0a678f5efccdd73f112e47d76c66e79c and onwards do not. `OHAI` is not printed so I guess the stub hangs during init. "Saved pc" after reset is "0x400003c0" (as printed by a separate UART cable). Esptool itself runs over usb.

I still think it has something to do with the part that increases the clock frequency, since a few rearrangements of the code there leads to the same issue, but only when the code that is increasing cpu frequency is present. Same code but setting it to 160 MHz instead of 240 MHz does not lead to the issue. Maybe there are some precautions that need to be taken when increasing the clock speed, such as doing it in a particular order or something? Adding delays in form of a `for (volatile int i = 0; i < 10000; i++) {}` between the register writes do not seem to fix anything.

Or maybe the particular ESP32-S3 module is simply broken on this unit? In any case, it would be good if @bamason2 could test the two mentioned commits above. @radimkarnis if you want the board that's failing we could send it to you.

fennecbutt, 2022-12-14:
I'm getting the same issue with an UnexpectedMaker TinyS3. I bought a ProS3 at the same time and this works fine.

OS: MacOS Mojave: 10.14.6 (18G9216)
Python: Python 2.7.16
ESPTool: v4.4

Problem:
So the ProS3 works fine, with and without --no-stub. The TinyS3 will only work with --no-stub option, which doesn't allow for erase_flash command.

If I run the command: `esptool.py --trace -p /dev/tty.usbmodem14301 -c esp32s3 flash_id` then it will just hang after "Running stub..." and never complete.

The output is:
```
esptool.py v4.4
Serial port /dev/tty.usbmodem14301
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000

TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a04000900000000000000c0
TRACE +0.000 Received full packet: 010a04000900000000000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=4cf1ce3f
TRACE +0.000 Write 14 bytes: c0000a0400000000004cf1ce3fc0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a04000400000000000000c0
TRACE +0.000 Received full packet: 010a04000400000000000000
TRACE +0.000 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.000 Write 10 bytes: c00014000000000000c0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 33 bytes: 
    0114180004000000 0000000000000000 | ................
    0000000c09000000 0000000000000000 | ................
    c0                                | .
TRACE +0.000 Received full packet: 
    0114180004000000 0000000000000000 | ................
    0000000c09000000 0000000000000000 | ................
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=58700060
TRACE +0.000 Write 14 bytes: c0000a04000000000058700060c0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a040020dc067c00000000c0
TRACE +0.000 Received full packet: 010a040020dc067c00000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=58700060
TRACE +0.000 Write 14 bytes: c0000a04000000000058700060c0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a040020dc067c00000000c0
[... 685 lines cut ...]
TRACE +0.000 Write 26 bytes: 
    c000051000000000 0004000000010000 | ................
    0000180000fc2bcb 3fc0             | ......+.?.
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 0105040012f4000000000000c0
TRACE +0.000 Received full packet: 0105040012f4000000000000
TRACE +0.000 command op=0x07 data len=20 wait_response=1 timeout=3.000 data=
    0400000000000000 0000000000000000 | ................
    1000ca3f                          | ...?
TRACE +0.000 Write 30 bytes: 
    c0000714000a0000 0004000000000000 | ................
    0000000000000000 001000ca3fc0     | ............?.
TRACE +0.000 Read 1 bytes: c0
TRACE +0.001 Read 13 bytes: 0107040012f4000000000000c0
TRACE +0.000 Received full packet: 0107040012f4000000000000
Running stub...
TRACE +0.000 command op=0x06 data len=8 wait_response=1 timeout=0.050 data=000000000c8a3740
TRACE +0.000 Write 18 bytes: 
    c000060800000000 00000000000c8a37 | ...............7
    40c0                              | @.
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 0106040012f4000000000000c0
TRACE +0.000 Received full packet: 0106040012f4000000000000
```
I'm going to try a few things tonight and see if I can come up with anything, but I'm pretty unfamiliar with native ESP stuff.

fennecbutt, 2022-12-15:
Just an update: I've just tried esptool v4.3 and it worked flawlessly. I'm still curious as to why it doesn't work with v4.4 but also why this only affects the tinyS3 and not the proS3 module. Although it sounds like Emill above encountered the issue with 1 part out of a bunch...so perhaps it's just some variance in the hardware itself?

I might apply some of the changes between 4.3 and 4.4 and see if I can help find what might be causing it...

radimkarnis, 2022-12-16:
@Emill @fennecbutt thanks for the feedback. It seems like when the CPU frequency gets increased for faster transfer speeds, the variance in the hardware comes into play. Maybe little differences between the crystals on the devkits, I am not sure. 

@fennecbutt you could try running esptool on https://github.com/espressif/esptool/commit/b25606b95920bd06df87aff9202c7a15377d4a30, this is one commit before the CPU speed increase was introduced.

stefanounick, 2022-12-18:
This commit https://github.com/espressif/esptool/commit/dccf4dff0a678f5efccdd73f112e47d76c66e79c seems to cause the issue, previous one works fine but from this one and onwards it gets stuck at Running stub. 
 
 Using an esp32-s3 as well.

radimkarnis, 2022-12-20:
Update: I have been able to reproduce this issue on a windows 10 machine (v4.3 works, v4.4 doesn't), but if I try the same thing on a Mac, everything works. So the issue is both devkit AND host machine dependent.

radimkarnis, 2022-12-28:
Hello @stefanounick @fennecbutt @Emill @bamason2,
could you please try [this branch containing a possible fix](https://github.com/radimkarnis/esptool/tree/test/lower_cpu_freq) and verify if this helps? 

We don't want to revert this feature entirely and there are reports that slowing down the CPU helps. I would be grateful if you could help me test this. Thank you!

Emill, 2022-12-29:
Yes that branch works on my otherwise failing board, just as in my experiments as explained above, i.e. the workaround to set the frequency to 160 MHz instead of 240 MHz.

I would like to know the cause though why 240 MHz does not work. I don't see any requirements in the reference manual about a sequence or similar how to safely change the frequency. I don't want to end up in the situation where a board can be programmed at 160 MHz but running the esp-idf application later at 240 MHz fails (for the same reason as we are investigating here).

fennecbutt, 2023-01-02:
@radimkarnis sorry for the delay, holidays and all that.

I've tested your branch; it works perfect every time. I've also been rebuilding the stub in that branch locally and SYSTEM_CPUPERIOD_MAX=2 always encounters the issue, SYSTEM_CPUPERIOD_MAX=1 never does.

The weird thing I've discovered is that on one board, it works perfect every time (pro s3 as mentioned). On other board (tinys3) when cpu is set to 240Mhz, it will _sometimes_ work. Hold down boot button and connect to usb, it won't work immediately but after leaving it some time it will work once, then fail. Or it will fail and then randomly work once before failing again.

It seems I can consistently get it to work if I have left it for a certain duration.

--- 30 minutes later ---

So I've discovered that if I put the board in the freezer (or fridge) for a minute or two (not long at all), then it consistently works perfectly fine! Once it warms up again after a short while it stops working, I did this several times and on one held my thumb over the esp chip itself to warm it up faster, and it starting failing again after a shorter time. Very interesting and also frustrating. Not sure if it's bc the esp can't do 240MHz at room temp (which is low atm, it's Winter) or if it's a bad solder joint or something causing it that gets compressed when the board has been briefly chilled.

radimkarnis, 2023-01-03:
@Emill @fennecbutt thanks for your thorough tests and verification! 

Although the reason is still unknown, I think we can now confirm that lowering the CPU freq to 160MHz is the solution. Fortunately, the benefits of faster transfer speeds are still here. In my tests, switching from 80MHz -> 160MHz nearly doubles the speeds, but 160MHZ -> 240MHz does not make any more significant difference.

ESP32-S3 is the only chip in our lineup so far that supports 240MHz and USB-JTAG/Serial. I guess this is the only configuration having issues, we had no other reports.

> The weird thing I've discovered is that on one board, it works perfect every time
> So I've discovered that if I put the board in the freezer (or fridge) for a minute or two (not long at all), then it consistently works perfectly fine!

I believe this only confirms the idea that this has something to do with the hardware. Unfortunately, I don't have the time to investigate about the possible implications for IDF apps. The goal here is to get rid of the regression in esptool.

fennecbutt, 2023-01-03:
@radimkarnis and thank you for your attention on this issue, I understand you have a lot of things to do & this is just one thing on the list; such is tech!

I've found a fix! (Let me know if you want me to throw together a PR) But root cause still unknown.

In `flasher_stub/stub_flasher.c`:
```c
static void set_max_cpu_freq()
{
  if (can_use_max_cpu_freq())
  {
    /* Set CPU frequency to max. This also increases SPI speed. */
    #if ESP32C6
    pcr_sysclk_conf_reg = READ_REG(PCR_SYSCLK_CONF_REG);
    WRITE_REG(PCR_SYSCLK_CONF_REG, (pcr_sysclk_conf_reg & ~PCR_SOC_CLK_SEL_M) | (PCR_SOC_CLK_MAX << PCR_SOC_CLK_SEL_S));
    #else
    cpu_per_conf_reg = READ_REG(SYSTEM_CPU_PER_CONF_REG);
    sysclk_conf_reg = READ_REG(SYSTEM_SYSCLK_CONF_REG);
    WRITE_REG(SYSTEM_SYSCLK_CONF_REG, (sysclk_conf_reg & ~SYSTEM_SOC_CLK_SEL_M) | (SYSTEM_SOC_CLK_MAX << SYSTEM_SOC_CLK_SEL_S));
    ets_delay_us(100);
    WRITE_REG(SYSTEM_CPU_PER_CONF_REG, (cpu_per_conf_reg & ~SYSTEM_CPUPERIOD_SEL_M) | (SYSTEM_CPUPERIOD_MAX << SYSTEM_CPUPERIOD_SEL_S));
    #endif
  }
}
```

`SYSTEM_SYSCLK_CONF_REG` **must** be written before `SYSTEM_CPU_PER_CONF_REG` and there must be a small delay between these writes (a delay after both does not suffice).
I have tried delays down to 60, but `ets_delay_us(100);` seems to works flawlessly from what I can tell, with 60 it would work say 10 times in a row in quick succession (read_mac command) and then eventually fail.

With this change I was able to leave `#define SYSTEM_CPUPERIOD_MAX 2` in and have it work consistently every time (by running read_mac a dozen or so times in a row without failure).

I'm not sure if I'll spend more time looking into this either; the fix above works for me at 240MHz. My current theory is APB going from 320MHz to 480MHz on some chips take some time to settle or some other funky physics related thing, but now I should actually start the project I was intending to build with this board, ha ha.

radimkarnis, 2023-01-04:
@fennecbutt this is great! thanks for looking into it and being persistent. 

We could try and consider this solution if we get more than one reports that this helps. Unfortunately, I am not able to reproduce this issue in my environment. Maybe we could ask @Emill to help.
If this works for him, I would be happy to accept your PR, contributors are welcome here!

> but now I should actually start the project I was intending to build with this board, ha ha.

Good luck with that, hopefully no other weird issue pops up to slow you down.

stefanounick, 2023-01-04:
I can confirm that @fennecbutt 's changes fix the issue here as well. 
I've been writing to the flash at 240MHz consistently without any issues for the past two hours.

radimkarnis, 2023-01-05:
@stefanounick thanks a lot for the confirmation! I will merge @fennecbutt's solution ASAP.

closed 2023-01-05 by espressif-bot

rkagerer, 2023-01-29:
I had the same problem with my UnexpectedMaker FeatherS3 board.  Unfortunately  esptool 4.5.dev2 does not seem to fix it for me.  However,  4.3 worked correctly.  Attached traces for both if it helps.

[4.3-works.txt](https://github.com/espressif/esptool/files/10528901/4.3-works.txt)

[4.5.dev2-hangs.txt](https://github.com/espressif/esptool/files/10528902/4.5.dev2-hangs.txt)