espressif/esptool #810: [Secure ROM DL Mode] read_flash command fails

chipweinberger, 2022-12-10:
### Operating System

macOS 12.5.1

### Esptool Version

v4.4.3-215-g9f3f82f54b

### Python Version

3.9.9

### Chip Description

ESP32-S3

### Device Description

ESP32-S3 Dev Kit C

### How is Esptool Run

Mac OS Terminal

### Full Esptool Command Line that Was Run

/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf4.4_py3.9_env/bin/python /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/cu.SLAB_USBtoUART -b 460800 --before default_reset --after no_reset --chip esp32s3 --no-stub read_flash 0x0 100 /Users/charlesweinberger/Desktop/testdata.bin

### Esptool Output

```shell
esptool.py v3.3.2
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP32-S3 in Secure Download Mode
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Traceback (most recent call last):
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 5482, in <module>
    _main()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 5475, in _main
    main()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 4907, in main
    operation_func(esp, args)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 4359, in read_flash
    data = esp.read_flash(args.address, args.size, flash_progress)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 1058, in read_flash
    return self.read_flash_slow(offset, length, progress_fn)  # ROM-only routine
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 1805, in read_flash_slow
    r = self.check_command("read flash block", self.ESP_READ_FLASH_SLOW,
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 520, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 505, in command
    raise UnsupportedCommandError(self, op)
__main__.UnsupportedCommandError: This command (0xe) is not supported in Secure Download Mode
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % git describe
fatal: No names found, cannot describe anything.
```

### More Information

Accoording to the docs read_flash should work:

https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/security/flash-encryption.html#best-practices

`The UART ROM Download Mode should be disabled entirely if it is not needed, or permanently set to “Secure Download Mode” otherwise. Secure Download Mode permanently limits the available commands to basic flash read and write only. The default behaviour is to set Secure Download Mode on first boot in Release mode. To disable Download Mode entirely select select the [CONFIG_SECURE_UART_ROM_DL_MODE](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/kconfig.html#config-secure-uart-rom-dl-mode) to “Permanently disable ROM Download Mode (recommended)” or call [esp_efuse_disable_rom_download_mode()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/system/efuse.html#_CPPv435esp_efuse_disable_rom_download_modev) at runtime.`

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-12-14:
The fixes have been merged, the docs should update automatically today. Thanks again for the report!

closed 2022-12-14 by radimkarnis

chipweinberger, 2022-12-14:
Glad I could help!