espressif/esptool #811: [espsecure.py] encrypt_flash_data needs the ability to pass the key as a command line. Writing sensitive keys to disk is discouraged.

chipweinberger, 2022-12-11:
### Is your feature request related to a problem?

`encrypt_flash_data` needs to support a ' --keystring' command , ' -s'.

keyfiles are convenient, but annoying from a security practice. It's extremely cumbersome to delete files in a secure way.

### Describe the solution you'd like

`encrypt_flash_data --keystring ab374301288ffee91283750193abde00ab374301288ffee91283750193abde00`

### Additional context

```
espsecure.py encrypt_flash_data --help

usage: espsecure encrypt_flash_data [-h] [--aes_xts] --keyfile KEYFILE --output OUTPUT --address ADDRESS [--flash_crypt_conf FLASH_CRYPT_CONF] plaintext_file

positional arguments:
  plaintext_file        File with plaintext content for encrypting

optional arguments:
  -h, --help            show this help message and exit
  --aes_xts, -x         Encrypt data using AES-XTS as used on ESP32-S2, ESP32-C2 and ESP32-C3
  --keyfile KEYFILE, -k KEYFILE
                        File with flash encryption key
  --output OUTPUT, -o OUTPUT
                        Output file for encrypted data.
  --address ADDRESS, -a ADDRESS
                        Address offset in flash where file will be flashed.
  --flash_crypt_conf FLASH_CRYPT_CONF
                        Override FLASH_CRYPT_CONF efuse value (default is 0XF).
```

chipweinberger, 2022-12-11:
The same enhancement is needed for `espefuse.py`

`espefuse.py --port PORT burn_key BLOCK my_flash_encryption_key.bin KEYPURPOSE`

dobairoland, 2022-12-12:
Esptool will have soon PyKCS#11 compliant hardware security module support. It will be probably included in the next minor release. I'll leave a comment when it is ready.

dobairoland, 2023-02-22:
I'm sorry but we have reached the conclusion that we will not support this.

> keyfiles are convenient, but annoying from a security practice. It's extremely cumbersome to delete files in a secure way.

This is true but at least this is clear to the user that it has to be done. There are simple utilities for this which aim to help securely remove files. Or files with key can be put on RAM disk.

Adding keys to command line arguments are even less secure IMO because we are living under the false assumption that the key has not been stored on disk. But as a matter of fact, most shells store commands in some history file. Most of the users are not realizing this. I think this is more dangerous than the situation when you intentionally put a key into a file. 

There are some workarounds possible, as it was show for example in https://github.com/espressif/esptool/pull/841#issuecomment-1436830324. I don't recommend it over using files.

As a more secure way we recommend the PyKCS#11 compliant hardware security module support available since v4.5.

Thank you for understanding that I'm closing this request.

closed 2023-02-22 by dobairoland

chipweinberger, 2023-02-22:
> But as a matter of fact, most shells store commands in some history file.

This is a good point!

Thanks for bringing it up.