espressif/esptool #813: [esptool.py] "This command (0x14) is not supported in Secure Download Mode" happens sporadically

chipweinberger, 2022-12-15:
### Operating System

macOS 12.5.1

### Esptool Version

v4.4.3-215-g9f3f82f54b

### Python Version

3.9.9

### Full Esptool Command Line that Was Run

/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf4.4_py3.9_env/bin/python /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py --port /dev/cu.usbmodem2101 --chip esp32s3 --no-stub get_security_info

### Esptool Output

```shell
Exception has occurred: UnsupportedCommandError
This command (0x14) is not supported in Secure Download Mode
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 505, in command
    raise UnsupportedCommandError(self, op)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 520, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 902, in get_chip_id
    res = self.check_command('get security info', self.ESP_GET_SECURITY_INFO, b'')
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 168, in inner
    return func(*args, **kwargs)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 1343, in check_chip_id
    chip_id = self.get_chip_id()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 719, in connect
    self.check_chip_id()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 116, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 4782, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 5475, in _main
    main()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 5482, in <module>
    _main()
```

### What is the Expected Behaviour?

It should print this:

```
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % esptool.py --chip esp32s3 get_security_info
esptool.py v3.3.2
Found 2 serial ports
Serial port /dev/cu.usbmodem2101
Connecting...
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Enabling default SPI flash mode...
Flags: 0x00000084 (0b10000100)
Flash_Crypt_Cnt: 0x1
Key_Purposes: (0, 2, 3, 7, 0, 0, 12)
Chip_ID: 9
Api_Version: 0
Hard resetting via RTS pin...
```

### More Information

It happens literally about 50% of the time, maybe perfectly alternating between working at not working.

chipweinberger, 2022-12-15:
Does not repro in UART mode

radimkarnis, 2022-12-20:
Hi @chipweinberger,
Thanks for reporting. I tried to reproduce this and can confirm that there is a little bug, I am already fixing it.

Can you please try using the latest esptool from master and confirm that it works when you specify the chip with `-c esp32s3` and doesn't work when autodetection is used (omit the `-c` arg)?

radimkarnis, 2022-12-22:
Update:
I suspect a bug in the ESP32S3 from. 

This happens only when:
- secure download mode is active
- USB-Serial/JTAG is used
- `-c esp32s3` is specified (doesn't happen with autodetection)
- the chip isn't placed into the download mode beforehand

For some reason, the ROM responds with an `INVALID_COMMAND` response on every second try. 

Fortunately, this seems to be only a minor inconvenience that happens only in particular conditions (see above). This can be easily solved on the user side by either running the command again OR not specifying the chip with `-c/--chip` OR switching manually into download mode beforehand.

I am going to try to come up with a solution in esptool. Will leave this open until then.

chipweinberger, 2022-12-23:
Thanks for the update. That's interesting. I added `-chip` because I would hit other "command not supported" issues when I _didn't_ specify. I'll make sure to file another issue if I hit more.

closed 2022-12-23 by espressif-bot

radimkarnis, 2022-12-23:
> I added -chip because I would hit other "command not supported" issues when I didn't specify

@chipweinberger this and the originally reported issue should be now fixed! Could you please try pulling the latest `master` and see if it works? Thanks!

chipweinberger, 2023-01-22:
Sorry for the slow response. I have not returned to this part of my codebase yet!

chipweinberger, 2024-03-10:
So I just hit this again. Had not hit it in awhile. 

ESP-IDF: v5.2.1

```
/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf5.1_py3.9_env/bin/python /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py --port /dev/cu.usbmodem101 --chip esp32s3 --no-stub get_security_info

Traceback (most recent call last):
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 690, in connect
    self.check_chip_id()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 1455, in check_chip_id
    chip_id = self.get_chip_id()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 131, in inner
    return func(*args, **kwargs)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 913, in get_chip_id
    res = self.check_command(
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 436, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 419, in command
    raise UnsupportedCommandError(self, op)
esptool.util.UnsupportedCommandError: This command (0x14) is not supported in Secure Download Mode
```

chipweinberger, 2024-03-10:
has this been fixed in the ROM? If I purchase new chips? Pretty annoying bug!