espressif/esptool #814: [USB Serial JTAG} Can't connect to new fresh out of box - FatalError('Failed to connect to {}: {}'

chipweinberger, 2022-12-15:
### Operating System

macOS 12.5.1

### Esptool Version

v4.4.3-215-g9f3f82f54b

### Python Version

3.9.9

### Chip Description

ESP32-S3 Dev Kit C

### Device Description

Custom PCB

### How is Esptool Run

Mac OS Terminal

### Full Esptool Command Line that Was Run

/firmware/.espressif/python_env/idf4.4_py3.9_env/bin/python /firmware/esp-idf/components/esptool_py/esptool/esptool.py --port /dev/cu.usbmodem1234561 --chip esp32s3 --no-stub get_security_info

### Esptool Output

```shell
Exception has occurred: SystemExit
2
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 5475, in _main
    main()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 4782, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 116, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 696, in connect
    raise FatalError('Failed to connect to {}: {}'

During handling of the above exception, another exception occurred:

  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 5478, in _main
    sys.exit(2)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py", line 5482, in <module>
    _main()
```

### More Information

```
Available: /dev/cu.Bluetooth-Incoming-Port
Available: /dev/cu.usbmodem1234561

Port: /dev/cu.usbmodem1234561
```

What I find interesting is the port number. 1234561 is the highest I've seen. Usually it's like 2101, 1101, etc.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

chipweinberger, 2022-12-15:
I plugged in *another* new device, and same exact issue, 'usbmodem1234561'. 

**This is getting worrying.** I plan on using USB_SERIAL_JTAG to program my device in production.

dobairoland, 2022-12-15:
> v4.4.3-215-g9f3f82f54b

Is the above your ESP-IDF version? I'm judging this from the commands you are running. The esptool version in that case is v3.3.2. It can be seen in the log not shown in your report. You could try the latest esptool version (which is by coincidence v4.4) and see if the issue still occurs or not.

Are you sure you've got the right port? You don't use port autodetection but specify it by yourself? It could help if you capture the full log using additionally the `--trace` argument.

radimkarnis, 2022-12-15:
Judging from the limited log, the failure comes from pySerial. That means a working serial connection cannot be established and esptool cannot continue to work.

These issues are usually caused by hardware. Please check if you are selecting the correct serial port, reset the chip into download mode manually (if your board supports this), and try connecting to it with any serial terminal programs (e.g. `miniterm` provided by pySerial). 

> What I find interesting is the port number. 1234561 is the highest I've seen. Usually it's like 2101, 1101, etc.

This seems quite normal, I see these high numbers on my Mac all the time and it works.

chipweinberger, 2022-12-15:
Thanks for the advice. I will play around with it some more and see if I can get it to work, and report back.

chipweinberger, 2022-12-15:
Ok. I have some important info!

1. It only happens on fresh devices (Esp32-s3 Dev Kit C) over USB Serial JTAG, when the modem is /dev/cu.usbmodem1234561
2. If you enter "UART Download" mode (gpio 0, then reset), it works, and the modem becomes /dev/cu.usbmodem1101
3. If you flash a new application to it, it works, and the modem becomes /dev/cu.usbmodem1101

**So to summarize, it only happens on fresh Esp32-s3 Dev-Kit-C devices while running the preinstalled "Blink" application.** 

**Full Failure Log:**
```
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % esptool.py -p /dev/cu.usbmodem1234561 get_security_info 
esptool.py v3.3.2
Serial port /dev/cu.usbmodem1234561
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0x1B): Possible serial noise or corruption.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % 
```

chipweinberger, 2022-12-15:
To hopefullt give you more information to go on.

**Module Markings:**
ESP32-S3-WROOM-1 w/ 2MB PSRAM, 8MB flash
IC: 21098-ESPS3WROOM1
CMIT ID: 2022DP2892
MON8R2

Fuses:
```
MAC (BLOCK1)  = 7c:df:a1:f3:bc:a4 (OK) R/W 
WAFER_VERSION (BLOCK1)  = 1 R/W (0b001)
PKG_VERSION (BLOCK1)  = ESP32-S3 R/W (0x0)
OPTIONAL_UNIQUE_ID (BLOCK2)  = d0 fd 4b 43 1e 6e 2e a3 86 20 9a 20 57 33 cc 72 R/W 
```

radimkarnis, 2022-12-19:
Hi @chipweinberger,
thanks for the report and investigation. Based on your report, this is not an esptool issue. 

You say that flashing works when you switch to download mode or when you reflash the application. I think it doesn't work "out-of-the-box" because the pre-flashed blink app isn't configured with `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG`. This has to be turned on in order for the console to work, see  [more info here](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32s3/api-guides/usb-serial-jtag-console.html#software-configuration).

Of course, if you reflash with this option turned on, or enter the download mode, the console is initialized and flashing works.

I am going to close this issue, as esptool works normally.

closed 2022-12-19 by radimkarnis

chipweinberger, 2022-12-19:
Before we closed this, we should have opened up an issue to fix the blink application.

Where would that issue be filed?

radimkarnis, 2022-12-19:
So I tried to reproduce and noticed that USB-JTAG/Serial is not even used in this case (otherwise it would respond to the DTR/RTS transitions). The pre-flashed app switches to TinyUSB or some other USB-OTG-based thing.

This can be verified by checking the PID:
```
desc: Espressif Device
hwid: USB VID:PID=303A:4001
```

The USB-JTAG/Serial controller has PID of 0x1001, this is not it.

> Before we closed this, we should have opened up an issue to fix the blink application.
Where would that issue be filed?

There is not a public repository containing the factory apps for our devkits, but I will try to get this to the right people.

chipweinberger, 2023-01-22:
bump. were you able to get this to the right people?

mcrajah, 2024-02-19:
if you hold the boot button and reset the board, the board enumerates with PID 0x1001. The board is stuck in 'boot mode' and does not load any USB stack that changes the PID to 0x4001. At this stage, you should be able to use the esp32s3-builtin.cfg with openocd to start the debug server.