espressif/esptool #817: Yet another MD5 of file does not match data in flash!

osvathrobi, 2022-12-20:
### Operating System

macOS 13

### Esptool Version

v4.4

### Python Version

1.10.6

### Chip Description

ESP8266

### Device Description

Sonoff ZigBee Bridge (zbbridge)

### How is Esptool Run

No

### Full Esptool Command Line that Was Run

sudo esptool.py --port /dev/cu.usbserial-AH02DHFV write_flash -e -fs 2MB -fm dout 0x0 tasmota-zbbridge.bin

### Esptool Output

```shell
esptool.py v4.4
Serial port /dev/cu.usbserial-AH02DHFV
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 48:3f:da:27:50:0d
Stub is already running. No upload is necessary.
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.1s
Flash params set to 0x0330
Compressed 535456 bytes to 373746...
Wrote 535456 bytes (373746 compressed) at 0x00000000 in 33.9 seconds (effective 126.5 kbit/s)...
File  md5: 11a261bbc0c1c8b8da94b712e21ab547
Flash md5: 5686468e2b2267e40a10c76c4296c43c
MD5 of 0xFF is de73e9eae434bf25758c9936bc308f31

A fatal error occurred: MD5 of file does not match data in flash!
```

### More Information

I have tried to flash a Sonoff ZigBee bridge with an FTDI with external 3.3V (from Lab PSU) and common GND. I have tried burning it with a NodeMCU (that has a voltage controller) and I also tried a CH340 based adapter - same result.

I then went out and bought another Sonoff ZbBridge (thinking maybe the chip was faulty) and to my surprise i got the same result. 

I tried erasing the flash before writing. I also tried erasing the flash with erase_region. Always the same result.

When I run `verify_flash -fs 2MB --dif yes 0x0 tasmota-zbbridge.bin` I sometimes get a small diff of only a couple of bytes wrong while other times there can be more or less mismatched bytes. However in the verify process a coupe of memory addresses always appear in the list (like address 0000003a)

ex:
```
Verifying 0x82ba0 (535456) bytes @ 0x00000000 in flash against tasmota-zbbridge.bin...
-- verify FAILED: 10 differences, first @ 0x0000003a
   0000003a f7 ff
   0000fe0e 23 a3
   00012d8d 85 c5
   0002007a 10 12
   00020366 27 2f
   0003c72d a3 e3
   0004fc77 d5 dd
   000503b4 f7 ff
   00066b1e 13 53
   00068f77 21 31
```

example on a different run:
```
Verifying 0x82ba0 (535456) bytes @ 0x00000000 in flash against tasmota-zbbridge.bin...
-- verify FAILED: 22 differences, first @ 0x0000003a
   0000003a f7 ff
   00000070 37 b7
   00000187 a4 b4
   00000245 49 59
   000002ef ea fa
   00000b88 08 28
   00007b96 84 85
   0000e350 23 33
   0000fe0e 23 a3
   0000ff12 e9 ed
   000101ad 2a 2b
   00012d8d 85 c5
   0002007a 10 12
   00020366 27 2f
   0003c72d a3 e3
   0004015e fd ff
   0004025d 01 41
   0004fc77 d5 dd
   0004fcc2 51 d1
   000503b4 f7 ff
   00066b1e 13 53
   00068f77 21 31
```

The really weird thing is that on one occasion it got all the bytes right and got a successful flash (on one of the devices). I have saved the output from console but could not reproduce a successful write on any of these devices ever since. It was just a random successful write.

### Other Steps to Reproduce

I know esptool works correctly because i have flashed other ESP8266 devices on this PC before.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2022-12-20:
Hi @osvathrobi. I'm sorry but I'm not familiar with that hardware. We haven't got any similar issue report so I guess the issue is hardware specific. Have you contacted the manufacturer? What are the instructions how to re-flash those devices? Is it even supported by the manufacturer?

radimkarnis, 2022-12-20:
Hi @osvathrobi,

I am not familiar with this particular piece of HW, but I've flashed tasmota into other devices before. I think the problem here is that the ESP chip and the flash memory chip are embedded into a PCB with other peripherals and even another MCU - Silicon Labs EFR32MG21. 

This poses a problem because you can't be sure nothing is interfering with the ESP while it is flashing. Can you be sure that only the ESP is running and not the other MCU when you connect 3V3 and GND?

> I sometimes get a small diff of only a couple of bytes wrong while other times there can be more or less mismatched bytes ... ...It was just a random successful write.

The fact that this is not reproducible and you get a different result every time supports the idea there is some kind of interference in the system. 

> I know esptool works correctly because i have flashed other ESP8266 devices on this PC before.

I am sorry but we might not be able to solve this issue because it is out of our scope. You admit that esptool works - therefore, this is most likely caused by the hardware and we do not have the capacity to analyze each and every custom ESP-based PCB.

This seems to be a common issue, though. [There is a reddit thread discussing this](https://www.reddit.com/r/tasmota/comments/kej2az/problem_flashing_sonoff_zbbridge_md5_of_file_does/). People are reporting this issue to be caused by their programmer nod giving enough current or being 5V instead of 3V3 - this might be worth looking into.

osvathrobi, 2022-12-20:
Hey guys, thank you for the feedback.  

@dobairoland A lot of people have no issue flashing these exact version of the Sonoff Zigbee boards that's why I thought that maybe it could be something more exotic or maybe issue with some weird write protection that I could not understand as a beginner and maybe someone here knows about.

@radimkarnis I thought i've eliminated the hardware issue by buying 2 boards and also trying different programmers and power supplies but the fact that you feel like it's probably a HW issue means that I will try to follow up and desolder or isolate the 8266 board and see If I can flash it on it's own.

Again thank you for the suggestions - should I close the issue? If I ever end up figuring it out I will add a comment to the closed issue for others to find.

closed 2022-12-20 by osvathrobi

osvathrobi, 2022-12-24:
For whoever has this problem here's what happened to me.. When I've bought 2 brand new Sonoff ZbBridge devices and both had the same MD5 error when flashing I thought about all the usual suspects: maybe 3V3 is wrong, maybe not enough current, maybe something is wrong with my flasher. After excluding all these I thought maybe there is something more advanced wrong so I've added this issue here.

Turns out that in my case the flash chips were faulty in BOTH brand new devices. (unless this is something intentional by the manufacturer, i guess it was just bad luck) 

Solution: I have sourced a new flash chip from another esp8266 module, one that in my case had 4Mb of memory (It needs at least 1Mb I think) I have then swapped out the flash chip from the Sonoff device with a new one that was confirmed working. After swapping out the chip everything worked from the first try. I flashed Tasmota on it successfully. 

So I can confirm that this was a hardware problem!

radimkarnis, 2022-12-24:
@osvathrobi Thanks for the update! It will be useful for future reference. Merry Christmas!