espressif/esptool #818: esptool fails to install

kdschlosser, 2022-12-27:
### Operating System

Windows 11

### Esptool Version

4.4

### Python Version

3.10

### Full Esptool Command Line that Was Run

None

### Esptool Output

```shell
got an error when installing esptool using pip.

Collecting esptool
  Downloading esptool-4.4.tar.gz (223 kB)
     -------------------------------------- 223.6/223.6 kB 6.9 MB/s eta 0:00:00
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting bitstring<4,>=3.1.6
  Downloading bitstring-3.1.9-py3-none-any.whl (38 kB)
Collecting cryptography>=2.1.4
  Downloading cryptography-38.0.4-cp36-abi3-win_amd64.whl (2.4 MB)
     ---------------------------------------- 2.4/2.4 MB 8.6 MB/s eta 0:00:00
Collecting ecdsa>=0.16.0
  Downloading ecdsa-0.18.0-py2.py3-none-any.whl (142 kB)
     -------------------------------------- 142.9/142.9 kB 8.3 MB/s eta 0:00:00
Requirement already satisfied: pyserial>=3.0 in c:\python310\lib\site-packages (from esptool) (3.5)
Collecting reedsolo<=1.5.4,>=1.5.3
  Downloading reedsolo-1.5.4.tar.gz (271 kB)
     -------------------------------------- 271.3/271.3 kB 8.4 MB/s eta 0:00:00
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting cffi>=1.12
  Downloading cffi-1.15.1-cp310-cp310-win_amd64.whl (179 kB)
     ------------------------------------- 179.1/179.1 kB 10.6 MB/s eta 0:00:00
Requirement already satisfied: six>=1.9.0 in c:\python310\lib\site-packages (from ecdsa>=0.16.0->esptool) (1.16.0)
Collecting pycparser
  Using cached pycparser-2.21-py2.py3-none-any.whl (118 kB)
Building wheels for collected packages: esptool, reedsolo
  Building wheel for esptool (setup.py): started
  Building wheel for esptool (setup.py): finished with status 'done'
  Created wheel for esptool: filename=esptool-4.4-py3-none-any.whl size=286028 sha256=31a1580610094d877aab27f1813a23aea3785c3226e52ec84446515ff6e7ecb5
  Stored in directory: c:\users\drsch\appdata\local\pip\cache\wheels\37\44\d0\23d374f0ee6155c68bfe318548d219922b4f9fd2f4a8270e82
  Building wheel for reedsolo (setup.py): started
  Building wheel for reedsolo (setup.py): finished with status 'error'
  Running setup.py clean for reedsolo
Successfully built esptool
Failed to build reedsolo
Installing collected packages: reedsolo, bitstring, pycparser, ecdsa, cffi, cryptography, esptool
  Running setup.py install for reedsolo: started
  Running setup.py install for reedsolo: finished with status 'error'

  error: subprocess-exited-with-error
  
  python setup.py bdist_wheel did not run successfully.
  exit code: 1
  
  [31 lines of output]
  Cython is installed, building creedsolo module
  running bdist_wheel
  running build
  running build_py
[... 61 lines cut ...]
  creedsolo.c(32338): error C2105: '++' needs l-value
  creedsolo.c(32340): error C2105: '--' needs l-value
  creedsolo.c(32588): error C2105: '++' needs l-value
  creedsolo.c(32590): error C2105: '--' needs l-value
  creedsolo.c(33787): warning C4996: 'PyEval_InitThreads': deprecated in 3.9
  creedsolo.c(35153): warning C4996: '_PyUnicode_get_wstr_length': deprecated in 3.3
  creedsolo.c(35169): warning C4996: '_PyUnicode_get_wstr_length': deprecated in 3.3
  creedsolo.c(37955): warning C4996: 'PyUnicode_FromUnicode': deprecated in 3.3
  error: command 'C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.30.30705\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
  [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

Encountered error while trying to install package.

reedsolo
```

I know this is an upstream problem and it's being caused by me having Cython installed
```

### What is the Expected Behaviour?

i for esptool to install

dobairoland, 2022-12-27:
Hi @kdschlosser. I believe this has already been fixed in the master branch and will be available in the next release. Could you try the master branch? For v4.4 [here](https://github.com/espressif/esptool/issues/711#issuecomment-1002257890) is the workaround.

kdschlosser, 2022-12-27:
all I did was uninstall Cython and then it installed without a hitch. Not complicated. It's an upstream problem in reedsolo and it has not been pdated to compile properly on Python 3.10. There was a lot of deprecated C API in Python that has been removed and that is why the compiler is throwing a fit about it.

dobairoland, 2023-01-16:
I believe this has been solved with the latest version of the reedsolo package. Please let us know if this still causing you trouble.

closed 2023-01-16 by dobairoland