espressif/esptool #824: 9260 Failed to execute script esptool due to unhandled exception

AlbertShown, 2023-01-12:
### Operating System

Windows 10

### Esptool Version

esptool.py v4.4

### Python Version

Python 3.10.5

### Chip Description

ESP32-D0WD (revision v1.0)

### Device Description

a complex board.

### Hardware Configuration

MCU connected to many other sensors.

### How is Esptool Run

command line

### Full Esptool Command Line that Was Run

esptool -p COM10 -b 4608000 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader.bin 0x8000 partitions.bin 0x10000 app.bin

### Esptool Output

```shell
esptool.py v4.4
Serial port COM4
Connecting....
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, BLK3 partially reserved, Coding Scheme 3/4
Crystal is 40MHz
MAC: 30:ae:a4:cb:b7:10
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 4608000
Changed.
Configuring flash size...
Traceback (most recent call last):
  File "esptool.py", line 34, in <module>
  File "esptool\__init__.py", line 1026, in _main
  File "esptool\__init__.py", line 808, in main
  File "esptool\cmds.py", line 216, in detect_flash_size
  File "esptool\loader.py", line 868, in flash_id
  File "esptool\loader.py", line 1250, in run_spiflash_command
  File "esptool\loader.py", line 690, in read_reg
  File "esptool\loader.py", line 376, in command
  File "esptool\loader.py", line 308, in read
StopIteration
[9260] Failed to execute script 'esptool' due to unhandled exception!
```

### More Information

It should show an error instead of this python exception.

### Other Steps to Reproduce

You can't reproduce it as it's only in my device.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-01-13:
Hi @AlbertShown,
it seems like the chip stops responding when the SPI flash is interrogated.

Could you please try:
- running the command with the `--no-stub` argument.
- increasing the [`MEM_END_ROM_TIMEOUT`](https://github.com/espressif/esptool/blob/master/esptool/loader.py#L80) timeout variable from 0.05 to e.g. 0.2 or higher. You might need to clone esptool from this repository and install it as [described here](https://docs.espressif.com/projects/esptool/en/latest/esp32/contributing.html#development-setup) to do so.

radostnomne, 2023-01-13:
stuck on the same issue on macos ventura 13. Can't flash the board using arduino ide. My stack trace:
`Sketch uses 264753 bytes (20%) of program storage space. Maximum is 1310720 bytes.
Global variables use 22280 bytes (6%) of dynamic memory, leaving 305400 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.2.1
Serial port /dev/cu.usbserial-1410
Connecting.....Traceback (most recent call last):
  File "esptool.py", line 34, in <module>
  File "esptool/__init__.py", line 1004, in _main
  File "esptool/__init__.py", line 790, in main
  File "esptool/loader.py", line 1108, in flash_set_parameters
  File "esptool/loader.py", line 406, in check_command
  File "esptool/loader.py", line 375, in command
  File "esptool/loader.py", line 307, in read
StopIteration
[68887] Failed to execute script 'esptool' due to unhandled exception!

Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: cc:db:a7:53:33:f8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Failed uploading: uploading error: exit status 1`
Do you have any idea how can I resolve this?

radimkarnis, 2023-01-23:
@AlbertShown @radostnomne could you please try any older esptool version (e.g. v3.0) to see if the issue prevails? You can install it by running `pip install esptool==3.0`. Thank you!

closed 2023-01-26 by espressif-bot

pieterbergmans, 2023-02-05:
I get the same error. I don't quite understand where to run the `pip install esptool==3.0` command suggested by @radimkarnis. I tried it at my command prompt but that didn't work. Besides, it looks like  esp32 versions are packaged with their own version of the esptool. Since esp32 v2.0.6 and v2.0.5 both have esptool v4.2.1, I tried installing v2.0.3 which has esptool v3.3.0. But, esp32 < v2.0.3 cause the following error: `Compilation error: exec: "python": executable file not found in $PATH`.  I did add python to my $PATH but then it prompted my to download a developer tools which I didn't do since it seems like an excessive fix. 

I've been stuck for several hours on this. If anyone can assist, I'd appreciate it. I'm using the M5Stack Fire and running a simple `HelloWorld` script.  The script compiles fine but produces the following error upon upload:

```
Sketch uses 405629 bytes (6%) of program storage space. Maximum is 6553600 bytes.
Global variables use 23888 bytes (0%) of dynamic memory, leaving 4498096 bytes for local variables. Maximum is 4521984 bytes.
esptool.py v4.2.1
Serial port /dev/cu.usbserial-54790373251
Connecting....Traceback (most recent call last):
  File "esptool.py", line 34, in <module>
  File "esptool/__init__.py", line 1004, in _main
  File "esptool/__init__.py", line 790, in main
  File "esptool/loader.py", line 1108, in flash_set_parameters
  File "esptool/loader.py", line 406, in check_command
  File "esptool/loader.py", line 375, in command
  File "esptool/loader.py", line 307, in read
StopIteration
[30870] Failed to execute script 'esptool' due to unhandled exception!

Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:21:84:93:b0:84
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Failed uploading: uploading error: exit status 1
```

pieterbergmans, 2023-02-05:
I ended up using VS Code + PlatformIO. I got it working in just a few minutes.