espressif/esptool #826: Determining amount of SRAM

grymoire, 2023-01-21:
### Is your feature request related to a problem?

I have several ESP32-S3-WROOM modules with differing amounts of FLASH and SRAM. I can use esptool.py  to get the amount of flash memory. Is there a way I can use esptool (or python) to determine the amount of SRAM as well? 

### Describe the solution you'd like

Have chip_id command (or some other command) report the size of SRAM on the chip.

### Describe alternatives you've considered

If there is some location in memory that can tell me this, that would be great. If not, perhaps a JTAG interface?

### Additional context

I'm using a esp-prog board and a flexypin-based adaptor to connect to the castellated modules without soldering them. I want to make sure I solder the right modules onto my board.

radimkarnis, 2023-01-23:
Hi @grymoire,
just to clarify - you probably mean PSRAM (not SRAM), right? 
SRAM is on the chip die and its size doesn't change (e.g. all ESP32-S3s have 512KB of SRAM).

If you do - I am sorry, but PSRAM size autodetection is currently not possible. Please see https://github.com/espressif/esp-idf/issues/9101

radimkarnis, 2023-01-30:
I am closing this feature request because of the reasons explained above.

closed 2023-01-30 by radimkarnis

grymoire, 2023-01-30:
Can ESPTOOL read and write PSRAM? If not, a trivial program could run and
test how much PSRAM is on a WROOM module by writing to a location and
testing if it worked.