espressif/esptool #827: Timeout detecting or configuring flash size

rgov, 2023-01-22:
### Operating System

macOS 13.1

### Esptool Version

esptool.py v4.4, esptool.py v3.0

### Python Version

Python 3.10.9

### Chip Description

ESP8266

### Device Description

[Blinkinlabs EightByEight Blinky](https://shop.blinkinlabs.com/collections/wearables/products/eightbyeight-blinky), [repo](https://github.com/Blinkinlabs/EightByEight)

### Hardware Configuration

unsure

### How is Esptool Run

Arduino IDE 2.0.3 w/ esp8266 3.11; also comand line

### Full Esptool Command Line that Was Run

esptool.py --trace --chip esp8266 --port /dev/cu.usbmodem32123101 --baud 460800 --before default_reset --after hard_reset write_flash --flash_mode qio --flash_freq 40m --flash_size keep --no-compress 0 FlaschenTaschenReceiver.ino.bin

### Esptool Output

```shell
[Full log here](https://gist.github.com/rgov/e77c90210e7cd7e10055283c21222aa5).

Stub running...
Changing baud rate to 460800
TRACE +0.000 command op=0x0f data len=8 wait_response=1 timeout=3.000 data=0008070000c20100
TRACE +0.000 Write 18 bytes: 
    c0000f0800000000 000008070000c201 | ................
    00c0                              | ..
TRACE +0.006 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010f0200000000000000c0
TRACE +0.000 Received full packet: 010f0200000000000000
Changed.
TRACE +0.058 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=1c020060
TRACE +0.000 Write 14 bytes: c0000a0400000000001c020060c0
TRACE +3.003 No serial data received.
TRACE +0.002 Unable to verify flash chip connection (No serial data received.).
TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=1c020060
TRACE +0.000 Write 14 bytes: c0000a0400000000001c020060c0
TRACE +0.001 Unable to perform XMC flash chip startup sequence ().
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00048fff...
TRACE +0.004 command op=0x02 data len=16 wait_response=1 timeout=3.000 data=a0840400130000000040000000000000
TRACE +0.000 Write 26 bytes: 
    c000021000000000 00a0840400130000 | ................
    0000400000000000 00c0             | ..@.......
Traceback (most recent call last):
  File "/opt/homebrew/bin/esptool.py", line 34, in <module>
    esptool._main()
  File "/opt/homebrew/Cellar/esptool/4.4/libexec/lib/python3.11/site-packages/esptool/__init__.py", line 1026, in _main
    main()
  File "/opt/homebrew/Cellar/esptool/4.4/libexec/lib/python3.11/site-packages/esptool/__init__.py", line 826, in main
    operation_func(esp, args)
  File "/opt/homebrew/Cellar/esptool/4.4/libexec/lib/python3.11/site-packages/esptool/cmds.py", line 527, in write_flash
    blocks = esp.flash_begin(uncsize, address, begin_rom_encrypted=encrypted)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/esptool/4.4/libexec/lib/python3.11/site-packages/esptool/loader.py", line 799, in flash_begin
    self.check_command(
  File "/opt/homebrew/Cellar/esptool/4.4/libexec/lib/python3.11/site-packages/esptool/loader.py", line 407, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/esptool/4.4/libexec/lib/python3.11/site-packages/esptool/loader.py", line 376, in command
    p = self.read()
        ^^^^^^^^^^^
  File "/opt/homebrew/Cellar/esptool/4.4/libexec/lib/python3.11/site-packages/esptool/loader.py", line 308, in read
    return next(self._slip_reader)
           ^^^^^^^^^^^^^^^^^^^^^^^
StopIteration
```
```

### More Information

The settings are based on the Getting Started guide for the EightByEight which are given for an older Arduino IDE release. I'm not sure how to translate "2M (1M SPIFFS)" to the `--flash_size` argument.

<img width="322" alt="image" src="https://user-images.githubusercontent.com/108767/213932028-57994487-9966-4832-b734-7a97a6caa8dd.png">

Occasionally fails with `esptool.util.UnsupportedCommandError: Invalid (unsupported) command 0x8` instead.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

rgov, 2023-01-22:
I successfully flashed with Arduino esp8266 2.4.2 with esptool v0.4.13.

radimkarnis, 2023-01-23:
Hi @rgov, could you please try older versions of `esptool.py`? We could try to see if this is a regression introduced in any of the later versions or if this tool. You mention you've tried `esptool.py v4.4 (esptool.py v3.0 through Arduino IDE)` - does this fail the same way in both versions?

> I successfully flashed with Arduino esp8266 2.4.2 with esptool v0.4.13.

This is actually `esptool-ck` from https://github.com/igrr/esptool-ck, it is a different implementation written in C.

rgov, 2023-01-23:
> You mention you've tried `esptool.py v4.4 (esptool.py v3.0 through Arduino IDE)` - does this fail the same way in both versions?

I believe it does fail in the same way. I tried a few other releases (esp8266 2.7.0 maybe) but I didn't record exactly which versions I tried.

radimkarnis, 2023-01-27:
Hi @rgov,
if you try the latest esptool master from this repository, you should now get more information about the issue.
Please try that and we'll know more.

Also, you can try running the command with the `--no-stub` argument.

closed 2023-02-20 by dobairoland