espressif/esptool #828: write_flash complains about --flash_size 8MB even though same parameter encoded inside of binary

quiret, 2023-01-23:
### Operating System

Windows 11 64bit

### Esptool Version

4.4.0

### Python Version

3.10.9

### Chip Description

ESP32-WROOM-32U

### Device Description

Problem occurs when trying to build the framework-arduinoespressif32 PlatformIO package, combined with espressif32@6.0.0 for the MKS TinyBee V1.0, ESP32-WROOM-32U which has 8MB cache. First off, the default toolchain fails completely. Inside of the board building main.py script there are the following tool invocations:

```
        ElfToBin=Builder(
            action=env.VerboseAction(" ".join([
                '"$PYTHONEXE" "$OBJCOPY"',
                "--chip", mcu, "elf2image",
                "--flash_mode", "${__get_board_flash_mode(__env__)}",
                "--flash_freq", "${__get_board_f_flash(__env__)}",
                "--flash_size", board.get("upload.flash_size", "4MB"),
                "-o", "$TARGET", "$SOURCES"
            ]), "Building $TARGET"),
            suffix=".bin"
        ),
(...)
        UPLOADER=join(
            platform.get_package_dir("tool-esptoolpy") or "", "esptool.py"),
        UPLOADERFLAGS=[
            "--chip", mcu,
            "--port", '"$UPLOAD_PORT"',
            "--baud", "$UPLOAD_SPEED",
            "--before", board.get("upload.before_reset", "default_reset"),
            "--after", board.get("upload.after_reset", "hard_reset"),
            "write_flash", "-z",
            "--flash_mode", "${__get_board_flash_mode(__env__)}",
            "--flash_freq", "${__get_board_f_flash(__env__)}",
            "--flash_size", board.get("upload.flash_size", "detect")
        ],
(...)
            UPLOADERFLAGS=[
                "--chip", mcu,
                "--port", '"$UPLOAD_PORT"',
                "--baud", "$UPLOAD_SPEED",
                "--before", "default_reset",
                "--after", "hard_reset",
                "write_flash", "-z",
                "--flash_mode", "$BOARD_FLASH_MODE",
                "--flash_size", board.get("upload.flash_size", "detect"),
                "$FS_START"
            ],
```
In my opinion this is a design error of esptool because all parameters of flash_size are passed the same to each stage of ESP32 image building but the tool still complains like this:

```
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00080fff...
Warning: Image file at 0x1000 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=8MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 17104 bytes to 11191...
Writing at 0x00001000... (100 %)
Wrote 17104 bytes (11191 compressed) at 0x0
```
The big problem here is that the tool - for some strange reason - resets the flash-size to 4MB. At least this is what the specification says (https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flash-modes.html#flash-size-flash-size-fs):

`If detection fails, a warning is printed and a default value of of 4MB (4 megabytes) is used.`

This is my interpretation of the error situation regarding image building failure for ESP32-WROOM-32U. Please investigate and fix this design flaw of esptool.

### How is Esptool Run

PlatformIO

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

peterdragun, 2023-01-23:
Hi @quiret,
it seems like the image was built with a 4MB flash size and then it was able to detect 8MB of flash during uploading. Can you please verify that `upload.flash_size` is set correctly? Looks like this variable was not set so the default value was used -> `4MB` in `ElfToBin` and `detect` in `UPLOADERFLAGS`.

Could you please provide at least the header of the image (first 24 bytes), this will make further investigation easier as it seems that issue could be there.

quiret, 2023-01-23:
First point:

**Log content (RELEVANT ITEMS IN BOLD)**

> Processing mks_tinybee (board: marlin_MKS_TinyBee; platform: espressif32@6.0.0; framework: arduino)
> ---------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> **BUILDERS --flash_size=8MB**
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/marlin_MKS_TinyBee.html
> PLATFORM: Espressif 32 (6.0.0) > MKS TinyBee V1.0
> HARDWARE: ESP32 240MHz, 512KB RAM, 8MB Flash
> DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES: 
>  - framework-arduinoespressif32 @ 0.0.0+sha.f740159 
>  - tool-esptoolpy @ 1.40400.0 (4.4.0) 
>  - tool-mkfatfs @ 2.0.1 
>  - tool-mklittlefs @ 1.203.210628 (2.3)
>  - tool-mkspiffs @ 2.230.0 (2.30)
>  - toolchain-xtensa-esp32 @ 11.2.0+2022r1
>
> Converting Marlin.ino
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 28 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- SPI @ 1.0
> |-- EEPROM @ 1.0.3
> |-- ArduinoOTA @ 1.0
> |   |-- Update @ 1.0
> |   |-- WiFi @ 1.0
> |   |-- ESPmDNS @ 1.0
> |   |   |-- WiFi @ 1.0
> |-- ESPmDNS @ 1.0
> |   |-- WiFi @ 1.0
> |-- WiFi @ 1.0
> |-- FS @ 1.0
> |-- SPIFFS @ 1.0
> |   |-- FS @ 1.0
> |-- Wire @ 1.0.1
> Building in release mode
> **UPLOADER --flash_size=8MB**
> Compiling .pio\build\mks_tinybee\src\src\inc\Warnings.cpp.o
> Compiling .pio\build\mks_tinybee\src\src\module\settings.cpp.o
> Marlin/src/inc/Warnings.cpp:48:4: warning: #warning "Safety Alert! Enable USE_WATCHDOG for the final build!" [-Wcpp]
>    48 |   #warning "Safety Alert! Enable USE_WATCHDOG for the final build!"
>       |    ^~~~~~~
> Linking .pio\build\mks_tinybee\firmware.elf
> Retrieving maximum program size .pio\build\mks_tinybee\firmware.elf
> Checking size .pio\build\mks_tinybee\firmware.elf
> Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
> RAM:   [=         ]  13.0% (used 68016 bytes from 524288 bytes)
> Flash: [=         ]  13.9% (used 465034 bytes from 3342336 bytes)
> Building .pio\build\mks_tinybee\firmware.bin
> esptool.py v4.4
> Creating esp32 image...
> Merged 1 ELF section
> Successfully created esp32 image.
> Configuring upload protocol...
> AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
> CURRENT: upload_protocol = esptool
> Looking for upload port...
> Auto-detected: COM3
> Uploading .pio\build\mks_tinybee\firmware.bin
> esptool.py v4.4
> Serial port COM3
> Connecting.....
> Chip is ESP32-D0WD (revision v1.0)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> Crystal is 40MHz
> MAC: 34:94:54:09:e9:10
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 500000
> Changed.
> Configuring flash size...
> Flash will be erased from 0x00001000 to 0x00005fff...
> Flash will be erased from 0x00008000 to 0x00008fff...
> Flash will be erased from 0x0000e000 to 0x0000ffff...
> Flash will be erased from 0x00010000 to 0x00081fff...
> **Warning: Image file at 0x1000 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=8MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum**
> Compressed 17104 bytes to 11191...
> Writing at 0x00001000... (100 %)
> Wrote 17104 bytes (11191 compressed) at 0x00001000 in 0.4 seconds (effective 330.7 kbit/s)...
> Hash of data verified.
> Compressed 3072 bytes to 129...
> Writing at 0x00008000... (100 %)
> Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 635.8 kbit/s)...
> Hash of data verified.
> Compressed 8192 bytes to 47...
> Writing at 0x0000e000... (100 %)
> Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 928.3 kbit/s)...
> Hash of data verified.
> Compressed 465136 bytes to 230650...
> Writing at 0x00010000... (6 %)
> Writing at 0x0001de29... (13 %)
> Writing at 0x0002c1bc... (20 %)
> Writing at 0x0003c573... (26 %)
> Writing at 0x00042519... (33 %)
> Writing at 0x0004860e... (40 %)
> Writing at 0x0004e468... (46 %)
> Writing at 0x00054148... (53 %)
> Writing at 0x00059caf... (60 %)
> Writing at 0x0005f70d... (66 %)
> Writing at 0x00068123... (73 %)
> Writing at 0x0006eec5... (80 %)
> Writing at 0x00075188... (86 %)
> Writing at 0x0007b0b5... (93 %)
> Writing at 0x00081015... (100 %)
> Wrote 465136 bytes (230650 compressed) at 0x00010000 in 5.2 seconds (effective 714.4 kbit/s)...
> Hash of data verified.
> 

Screen of device is BLACK. NO life sign! Bad image generation with 6.0.0! Fault in toolchain! Older version 5.0.0 works correctly.

Second point:

[image]

quiret, 2023-01-24:
Originally I got this issue when I had defined no upload.flash_size parameter and started the upload. Then after I did set a proper upload.flash_size parameter the error persisted. I cleanly retried with 6.0.0 espressif and then I was unable to reproduce the error. Sorry about this but it seems like a misconfiguration may have caused my error. That is why I have decided to close this. Thank you for your kind help.

closed 2023-01-24 by quiret