espressif/esptool #829: Cannot install esptool via pip

killkrt, 2023-01-24:
### Operating System

Linux

### Esptool Version

v4.4

### Python Version

Python 3.10.9

### Full Esptool Command Line that Was Run

pip install esptool

### Esptool Output

```shell
Defaulting to user installation because normal site-packages is not writeable
Collecting esptool
  Using cached esptool-4.4-py3-none-any.whl
Collecting bitstring<4,>=3.1.6
  Using cached bitstring-3.1.9-py3-none-any.whl (38 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting ecdsa>=0.16.0
  Using cached ecdsa-0.18.0-py2.py3-none-any.whl (142 kB)
Collecting reedsolo<=1.5.4,>=1.5.3
  Using cached reedsolo-1.5.4.tar.gz (271 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: cryptography>=2.1.4 in /usr/lib/python3.10/site-packages (from esptool) (38.0.4)
Requirement already satisfied: cffi>=1.12 in /usr/lib/python3.10/site-packages (from cryptography>=2.1.4->esptool) (1.15.1)
Requirement already satisfied: six>=1.9.0 in /usr/lib/python3.10/site-packages (from ecdsa>=0.16.0->esptool) (1.16.0)
Requirement already satisfied: pycparser in /usr/lib/python3.10/site-packages (from cffi>=1.12->cryptography>=2.1.4->esptool) (2.21)
Building wheels for collected packages: reedsolo
  Building wheel for reedsolo (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [124 lines of output]
      Cython is installed, building creedsolo module
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-cpython-310
      copying reedsolo.py -> build/lib.linux-x86_64-cpython-310
      running build_ext
      building 'creedsolo' extension
      creating build/temp.linux-x86_64-cpython-310
      gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -ffat-lto-objects -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -fPIC -I/usr/include/python3.10 -c creedsolo.c -o build/temp.linux-x86_64-cpython-310/creedsolo.o
      creedsolo.c: In function ‘__pyx_tp_dealloc_array’:
      creedsolo.c:32027:5: error: lvalue required as increment operand
      32027 |     ++Py_REFCNT(o);
            |     ^~
      creedsolo.c:32029:5: error: lvalue required as decrement operand
      32029 |     --Py_REFCNT(o);
            |     ^~
      creedsolo.c: In function ‘__pyx_tp_dealloc_memoryview’:
      creedsolo.c:32338:5: error: lvalue required as increment operand
      32338 |     ++Py_REFCNT(o);
            |     ^~
      creedsolo.c:32340:5: error: lvalue required as decrement operand
      32340 |     --Py_REFCNT(o);
            |     ^~
      creedsolo.c: In function ‘__pyx_tp_dealloc__memoryviewslice’:
      creedsolo.c:32588:5: error: lvalue required as increment operand
      32588 |     ++Py_REFCNT(o);
[... 250 lines cut ...]

× Encountered error while trying to install package.
╰─> reedsolo

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure
```

### What is the Expected Behaviour?

No error

### More Information

This issue is quite similar to https://github.com/espressif/esptool/issues/615

I've already tried to execute the following commands, but it didn't solve the issue:
`pip install --upgrade reedsolo`
`pip install --upgrade reedsolo --install-option="--cythonize" --verbose`
`pip install --only-binary :all: --extra-index-url https://dl.espressif.com/pypi/ reedsolo`

GNU C++ version: 12.2.1 20230111
Clang version: 15.0.7

Arch Linux: 6.1.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 18 Jan 2023 19:54:38 +0000 x86_64 GNU/Linux

dobairoland, 2023-01-25:
Hi @killkrt. The issue has not been fixed in v4.4, only in the master branch. Could you please try the master branch or 4.5.dev2 instead?

(`pip install esptool==4.5.dev2`)

killkrt, 2023-01-25:
Thanks @dobairoland that solved the issue!

closed 2023-01-25 by killkrt