espressif/esptool #830: Can't flash a TinyPico ESP32 board

fingerskier, 2023-01-29:
### Operating System

Windows 10

### Esptool Version

4.4

### Python Version

3.10.8

### Chip Description

ESP32

### Device Description

TinyPico v3

### Hardware Configuration

No

### How is Esptool Run

Command line

### Full Esptool Command Line that Was Run

esptool --chip esp32 --port COM27 --baud 912600 write_flash -z 0x1000 g:\image\tinypico-20220618-v1.19.1.bin

### Esptool Output

```shell
esptool.py v4.4
Serial port COM27
Connecting....
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:75:25:ca:a0:4c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 912600
Changed.
Configuring flash size...
Traceback (most recent call last):
  File "C:\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Python310\Scripts\esptool.exe\__main__.py", line 7, in <module>
  File "C:\Python310\lib\site-packages\esptool\__init__.py", line 1026, in _main
    main()
  File "C:\Python310\lib\site-packages\esptool\__init__.py", line 826, in main
    operation_func(esp, args)
  File "C:\Python310\lib\site-packages\esptool\cmds.py", line 322, in write_flash
    if esp.get_secure_boot_enabled():
  File "C:\Python310\lib\site-packages\esptool\targets\esp32.py", line 168, in get_secure_boot_enabled
    efuses = self.read_reg(self.EFUSE_RD_ABS_DONE_REG)
  File "C:\Python310\lib\site-packages\esptool\loader.py", line 690, in read_reg
    val, data = self.command(
  File "C:\Python310\lib\site-packages\esptool\loader.py", line 376, in command
    p = self.read()
  File "C:\Python310\lib\site-packages\esptool\loader.py", line 308, in read
    return next(self._slip_reader)
StopIteration
```

### More Information

Initially I was able to run a micropython program on the board;  After uploading code (which was developed for this board) it gives the error [above].
I am able to upload a working C++ program from Arduino IDE.

### Other Steps to Reproduce

Prior to flashing I ran this command to erase the memory:

esptool --chip esp32 --port COM27 erase_flash

Result:

esptool.py v4.4
Serial port COM27
Connecting....
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:75:25:ca:a0:4c
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 6.4s
Hard resetting via RTS pin...

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

fingerskier, 2023-01-30:
Here are the results:

esptool --chip esp32 --port COM27 --baud 912600 write_flash --force -z 0x1000 g:\image\tinypico-20220618-v1.19.1.bin

esptool.py v4.4
Serial port COM27
Connecting....
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:75:25:ca:a0:4c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 912600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00173fff...
Compressed 1519248 bytes to 983442...
Traceback (most recent call last):
  File "C:\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Python310\Scripts\esptool.exe\__main__.py", line 7, in <module>
  File "C:\Python310\lib\site-packages\esptool\__init__.py", line 1026, in _main
    main()
  File "C:\Python310\lib\site-packages\esptool\__init__.py", line 826, in main
    operation_func(esp, args)
  File "C:\Python310\lib\site-packages\esptool\cmds.py", line 525, in write_flash
    blocks = esp.flash_defl_begin(uncsize, len(image), address)
  File "C:\Python310\lib\site-packages\esptool\loader.py", line 109, in inner
    return func(*args, **kwargs)
  File "C:\Python310\lib\site-packages\esptool\loader.py", line 975, in flash_defl_begin
    self.check_command(
  File "C:\Python310\lib\site-packages\esptool\loader.py", line 407, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Python310\lib\site-packages\esptool\loader.py", line 376, in command
    p = self.read()
  File "C:\Python310\lib\site-packages\esptool\loader.py", line 308, in read
    return next(self._slip_reader)
StopIteration

esptool --no-stub --chip esp32 --port COM27 --baud 912600 write_flash -z 0x1000 g:\image\tinypico-20220618-v1.19.1.bin

esptool.py v4.4
Serial port COM27
Connecting......
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:75:25:ca:a0:4c
Changing baud rate to 912600
Changed.
Enabling default SPI flash mode...

A fatal error occurred: No serial data received.

radimkarnis, 2023-01-30:
Thanks for trying. That's interesting, your chip connects and communicates with esptool correctly, but the ROM crashes after the flash memory chip gets configured. 

Two more things to try:
1) lower the baudrate to: `--baud 115200`
2) Does this also happen if you try to flash a different binary?

If that doesn't help at all, could you please share the binary? Maybe I will be able to reproduce this.

fingerskier, 2023-01-30:
Huzzah!

Works w/ baud-rate of 115200.  What do you suppose the issue is?  Is this a cable or chip or sumnelse?

radimkarnis, 2023-01-31:
I have tried to reproduce this with high baudrates and my chip fails with `A fatal error occurred: Failed to write compressed data to flash after seq 3 (result was C400: Failed SPI operation)`. This might imply that the integrated flash chip is not keeping up.

I guess the baudrate is just too high for this HW configuration. The combination of the USB-to-UART chip, USB cable and drivers come into play at these high speeds. It's always advised to use slower baudrates when things go wrong.

I will close this issue as it's solved now. Thank you for reporting!

closed 2023-01-31 by radimkarnis