espressif/esptool #831: ESP32 DevKit V4 download more with RTS/DTR

michelepagot, 2023-01-31:
### Operating System

openSUSE TW

### Esptool Version

v4.4

### Python Version

python3.10.9

### Chip Description

ESP32-WROVER-IE

### Device Description

Espressif devKit V4
[image]
[image]

### Hardware Configuration

Only the USB cable between the PC and the devKit

### How is Esptool Run

No IDE

### Full Esptool Command Line that Was Run

esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size 8MB 0x1000 .pio/build/esp32dev/bootloader.bin 0x8000 .pio/build/esp32dev/partitions.bin 0xe000 $TOOLROOT/piotools/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin 0x10000 .pio/build/esp32dev/firmware.bin

### Esptool Output

```shell
esptool.py v4.4
Serial port /dev/ttyUSB0
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

### More Information

Command success in time to time

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-01-31:
Please go through https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#bootloader-won-t-respond from the troubleshooting guide, especially Boot Mode Selection including the Manual Bootloader section and report back to us about the results.

michelepagot, 2023-01-31:
h3. Check you are passing the correct serial port on the command line.

Done and yes. Running the command multiple time with the same value for the `--port` argument could in time to time result in a proper flashing sequence. Another confirmation is that if the same esptools command is performed and the board is manually forced in the BootMode using the buttons on the board, the flash sequence complete properly.

h3. Check you have permissions to access the serial port, and other software (such as modem-manager on Linux) is not trying to interact with it. A common pitfall is leaving a serial terminal accessing this port open in another window and forgetting about it.

I'm able to use `screen` on the same machine, with same user, on the same port. Multiple sequential executions of esptools, on the same port, can eventually result in a PASS. Running esptools and configuring Debug Mode using board buttons ALWAYS result in a proper flashing.

h3. Check the chip is receiving 3.3V from a stable power source (see [Insufficient Power](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#insufficient-power) for more details.)

I'm using Espressif devkit V4 that I suppose provide stable power to the module and the chip. I'm powering all the thing with the USB cable. No USB hub, short good quality USB cable.

h3. Check that all pins are connected as described in [Boot Mode Selection](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#boot-mode). Check the voltages at each pin with a multimeter, “high” pins should be close to 3.3V and “low” pins should be close to 0V.

I'm using Espressif devkit V4 board that I suppose is properly designed about it. I can only comment that in the https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#automatic-bootloader there's a sentence refering to the fact that 

```
If this circuitry is implemented (all Espressif boards have it), adding a capacitor between the EN pin and GND (in the 1uF-10uF range) is necessary 
```

but looking at the Devkit V4 schematic from https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch.pdf  seems that a 0.1 uF is used C9. But please double check that as C14 is also there.

h3. If you have connected other devices to GPIO pins, try removing them and see if esptool starts working.

I have nothing connected on any pin.

h3. Try using a slower baud rate (-b 9600 is a very slow value that you can use to verify it’s not a baud rate problem).

Tested and does not seems to help to enter in Download Mode in a reliable way

radimkarnis, 2023-01-31:
Hi @michelepagot, thanks for reading the troubleshooting guide.

You can try the latest dev version of esptool by running `pip install esptool==4.5.dev2`. There have been some changes in the reset strategy, maybe these will help in your case.

michelepagot, 2023-01-31:
I can confirm that on my side, `4.5.dev2` provide more reliable results. Thanks

radimkarnis, 2023-02-01:
Thanks for confirming!

closed 2023-02-01 by radimkarnis