espressif/esptool #832: ESP32-S3 - Random crashes when reading/writing flash

mrdude2478, 2023-02-04:
### Operating System

Windows 10 64 bit

### Esptool Version

Release v4.4

### Python Version

Python 3.11

### Chip Description

ESP32-S3

### Device Description

Lilygo-T-Dongle S3

### Hardware Configuration

Nothing attached to the board, onboad LED and TFT screen are attached/pre-soldered to GPIO pins.

### How is Esptool Run

Windows tool

### Full Esptool Command Line that Was Run

python esptool.py -b 460800 --port COM3 --chip esp32s3 read_flash 0x0 0x1000000 T-Dongle-S3.bin

### More Information

Flash read at various baud rates, esptool randomly crashes at various percentages during the read/write cycle. Also during writing the flash seems to have more success flashing using the No Stub option.

### Other Steps to Reproduce

Tried reading and writing using these commands.

Backup:
python esptool.py -b 460800 --port COM3 --chip esp32s3 read_flash 0x0 0x1000000 T-Dongle-S3.bin

Flash:
python esptool.py -b 460800 --port COM3 write_flash --flash_freq keep 0x000000 T-Dongle-S3.bin

Using no-stub option, gives more success but takes hours to read and mostly fails trying to read the full 16MB flash.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-02-04:
Hi @mrdude2478,
please try the latest dev version by running `pip install esptool==4.5.dev3`. The `v4.4` version doesn't include the latest improvements for the S3.

mrdude2478, 2023-02-09:
Sorry for the late reply, I had lent out the dongle to a friend and just got it back. I am still getting the same random read errors with 4.5.dev3.

Here's the output:

```
C:\Users\MrDude\AppData\Local\Programs\Python\Python311\Scripts>esptool.py.exe -b 460800 --port COM3 --chip esp32s3 read_flash 0x0 0x1000000 T-Dongle-S3.bin
esptool.py v4.5.dev3
Serial port COM3
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:41:32:c8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
2965504 (17 %)
A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0xffc bytes
```

Regards.

radimkarnis, 2023-02-09:
> Note: If flashing via Arduino IDE, I never get an error flashing, but with esptool, I usually get an error, So com port and PC settings are fine.

But you don't read flash with Arduino, only write to it - so, does flashing directly with esptool also work?
If not, it's a regression between esptool versions (the one you use bundled with Arduino and `v4.5.dev3` you've downloaded manually.)

If writing flash works, we can focus solely on the `read_flash` operation.
Here are some things to try:
1) Try with the `--no-stub` argument
2) Try lowering the baud rate, e.g. `-b 115200`

mrdude2478, 2023-02-10:
Thanks, with --no-stub I get more success (I already said this in the first post), but it takes a very long time - over an hour to read, and sometimes crashes with the same errors. I have also tried reading at various baud rates all the way down to 9600. I am aware you don't read flash with Arduino, only write - but I never get any errors writing with Arduino, whereas with esptool I do. If you look at the output I posted I clearly states that I am using "esptool.py v4.5.dev3" as per your advice. I know you are trying to help, I am just pointing out the issues though with ESP32-S3, I know it's a new-ish chip, maybe the issue is something to do with timing during read cycles???

radimkarnis, 2023-02-10:
> Thanks, with --no-stub I get more success (I already said this in the first post)

Sorry, I overlooked that.

> If you look at the output I posted I clearly states that I am using "esptool.py v4.5.dev3"

But I am asking about the version of esptool in Arduino - it actually also uses esptool.py to flash the chip. So if flashing works in Arduino but doesn't when esptool is downloaded directly, there must be a regression between these two versions. If we can locate the version when this regression got introduced, it will be easy to fix.

mrdude2478, 2023-02-11:
This has been added to boards manager:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

Installed board:
Boards Manager - esp32, Espressif Systems version 2.0.6 Installed

Arduino output says - esptool.py v4.2.1 during flashing.

Does that help?

radimkarnis, 2023-02-13:
> Arduino output says - esptool.py v4.2.1 during flashing.

Yes, this is what I am asking about. So this means there was a change somewhere between `v4.2.1` and `v4.4` that is causing flashing/reading to fail in your environment.

In that case, a quick fix for you is to just use esptool v4.2.1 (`pip install esptool==4.2.1`).

We can try to identify the exact change that introduced the regression. Would you be willing to help us with that?

If yes, you would have to [install esptool in development mode](https://docs.espressif.com/projects/esptool/en/latest/esp32/contributing.html#development-setup). Please, let me know if you can do that.

mrdude2478, 2023-02-14:
Thanks for the replay, I installed the latest build in dev mode - this is the output when trying to dump:

'''
Requirement already satisfied: filelock<4,>=3.4.1 in c:\users\mrdude\appdata\roaming\python\python311\site-packages (from virtualenv>=20.10.0->pre-commit->esptool==4.6.dev0) (3.9.0)
Installing collected packages: esptool
  Attempting uninstall: esptool
    Found existing installation: esptool 4.6.dev0
    Uninstalling esptool-4.6.dev0:
      Successfully uninstalled esptool-4.6.dev0
  Running setup.py develop for esptool
Successfully installed esptool-4.6.dev0

C:\msys64\home\MrDude\PC\esptool>python esptool.py -b 460800 --port COM3 --chip esp32s3 read_flash 0x0 0x1000000 T-Dongle-S3.bin
esptool.py v4.6-dev
Serial port COM3
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:41:32:c8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
81920 (0 %)
A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0xfde bytes

C:\msys64\home\MrDude\PC\esptool>python esptool.py -b 460800 --port COM3 --chip esp32s3 read_flash 0x0 0x1000000 T-Dongle-S3.bin
esptool.py v4.6-dev
Serial port COM3
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:41:32:c8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
761856 (4 %)
A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0xfa5 bytes

C:\msys64\home\MrDude\PC\esptool>
'''

mrdude2478, 2023-02-14:
Update: I have uninstalled python, cleaned the registry, removed all python environmental variables, removed all orphaned python files from my AppData/local folder & roaming folders + temp files, carried out a disc cleanup, rebooted my PC and checked to see if anything python related exists - so I have purged my system of python and anything python related.

Then I ran the latest esptool.exe posted yesterday - and the problem still persists even after everything is cleaned???

```
C:\Users\MrDude\Desktop\lilygo\Flash-dump\tools>esptool -b 460800 --port COM3 --chip esp32s3 read_flash 0x0 0x1000000 T-Dongle-S3.bin
esptool.py v4.5
Serial port COM3
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:41:32:c8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
5173248 (30 %)
A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0xf27 bytes
```

Maybe it's the onboad 16MB flash, it just seems wierd I can always flash successfully with Arduino and the baudrate uses 921600 with that as well, where it fails with esptool.exe.

I will leave python uninstalled for now, and follow instructions from this point.

mrdude2478, 2023-02-14:
Next thing I did was manually replace esptool.exe in Arduino from 4.1 to 4.5 - see pic:

[image]

Then flashed the board from Arduino - 
```
Sketch uses 1496753 bytes (47%) of program storage space. Maximum is 3145728 bytes.
Global variables use 123048 bytes (37%) of dynamic memory, leaving 204632 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5
Serial port COM3
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:41:32:c8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0017dfff...
Compressed 14768 bytes to 10148...
Writing at 0x00000000... (100 %)
Wrote 14768 bytes (10148 compressed) at 0x00000000 in 0.3 seconds (effective 343.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.1 seconds (effective 224.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 381.2 kbit/s)...
Hash of data verified.
Compressed 1497184 bytes to 1000867...
Writing at 0x00010000... (1 %)
--->
Writing at 0x0017cfa3... (100 %)
Wrote 1497184 bytes (1000867 compressed) at 0x00010000 in 10.9 seconds (effective 1098.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

This is weird, perhaps Arduino is using some different command line option to flash as I never get any errors with it ???

mrdude2478, 2023-02-14:
Next using the same command line options that Arduino uses to flash:
```
esptool.exe --chip esp32s3 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 16MB 0x10000 test.bin
```
Output:
```
C:\Users\MrDude\Desktop\lilygo\Flash-dump\tools>esptool.exe --chip esp32s3 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 16MB 0x10000 test.bin
esptool.py v4.5
Serial port COM3
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:41:32:c8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00010000 to 0x0015bfff...
Compressed 1357088 bytes to 910195...
Writing at 0x000b1a44... (51 %)Traceback (most recent call last):
  File "esptool\__init__.py", line 1031, in _main
  File "esptool\__init__.py", line 831, in main
  File "esptool\cmds.py", line 580, in write_flash
  File "esptool\loader.py", line 131, in inner
  File "esptool\loader.py", line 1030, in flash_defl_block
  File "esptool\loader.py", line 435, in check_command
  File "esptool\loader.py", line 404, in command
  File "esptool\loader.py", line 337, in read
StopIteration

A fatal error occurred: The chip stopped responding.
```

mrdude2478, 2023-02-14:
Further testing shows this this commit is what broke the functionality:

Oct - 17 - Flasher stub: Increase CPU frequency to 240 MHz for ESP32-S3
https://github.com/espressif/esptool/commit/0656f77d7980fe29c7969efd9f04b23083f22ec7 - broken here.

Oct - 11 - OK
https://github.com/espressif/esptool/commit/8386dcbfb0f72faf1f08d6e908f5119e2dd4f5e9 - works OK

radimkarnis, 2023-02-14:
@mrdude2478 thanks for your awesome investigation! 

I suspected this commit might be the culprit. There were multiple reports that this caused some issues, but this was later fixed in https://github.com/espressif/esptool/commit/f37f38433786f73d8a8274b146859ac9955ac1e7 - this fix was reported to help in all cases (see https://github.com/espressif/esptool/issues/808 for more info).

For some reason, this didn't help in your case. What's also weird is that in your case it's pretty random (in some cases flashing even succeeds). In other reports, this failed right away.

Unfortunately, I am not able to reproduce this issue. It has proven to be heavily dependent on the chip and flashing environment. So it is quite difficult to debug and fix.

radimkarnis, 2023-02-14:
> I then replaced esptool/targets/stub_flasher/stub_flasher_32s3.json from commit https://github.com/espressif/esptool/commit/0656f77d7980fe29c7969efd9f04b23083f22ec7 with the one in the latest esptool.py v4.6-dev build and this seems to fix the issues I was having flashing the chip, however it wouldn't read the chip if over 0xFFFFF bytes.

Ok so this is getting even weirder. The stub in the current `v4.6-dev` is the same as the one in `v4.5`, this shouldn't make any difference - meaning `v4.5` should therefore work.

mrdude2478, 2023-02-14:
Thanks for replying, I don't know how these stub files are encypted - they look to be some sort of hashed base64, but I couldn't decrypt them, do you have any idea so that I can compare both files and them maybe edit the contents of the one in commit 0656f77d7980fe29c7969efd9f04b23083f22ec7 so that it can read a 16MB flash?

radimkarnis, 2023-02-14:
@mrdude2478 of course, you can find the source in the `flasher_stub` folder (https://github.com/espressif/esptool/tree/master/flasher_stub). There are instructions on how to build and test the stub.

mrdude2478, 2023-02-14:
> @mrdude2478 of course, you can find the source in the `flasher_stub` folder (https://github.com/espressif/esptool/tree/master/flasher_stub). There are instructions on how to build and test the stub.

Thanks I'll have a look at that.

mrdude2478, 2023-02-14:
> > I then replaced esptool/targets/stub_flasher/stub_flasher_32s3.json from commit [0656f77](https://github.com/espressif/esptool/commit/0656f77d7980fe29c7969efd9f04b23083f22ec7) with the one in the latest esptool.py v4.6-dev build and this seems to fix the issues I was having flashing the chip, however it wouldn't read the chip if over 0xFFFFF bytes.
> 
> Ok so this is getting even weirder. The stub in the current `v4.6-dev` is the same as the one in `v4.5`, this shouldn't make any difference - meaning `v4.5` should therefore work.

Maybe you read that wrong - I am using the stub from commit [0656f77] with the latest esptool (v4.6-dev). The new stub file that comes with everything after  commit [0656f77] is what is causing the issues.

Jason2866, 2023-02-14:
Maybe reverting the increased clock for flash and SPI commit? Seems it makes trouble sometimes. Better a bit slower and working always.

mrdude2478, 2023-02-14:
I am not sure what the vendor is, the dongle is here 16MB flash:

https://www.lilygo.cc/products/t-dongle-s3

I used a magnifying glass to see the flash chip and I think it says winbond, but my eyesight is not the best so I could be mistaken.

radimkarnis, 2023-02-14:
@mrdude2478 there was another solution proposed here: https://github.com/radimkarnis/esptool/tree/test/lower_cpu_freq

If you could try this branch and verify if this works in your case, I would by grateful.

mrdude2478, 2023-02-14:
Yes, I'll try it just in about 10 minutes. Also I think the flash is winbond - W25Q32

Jason2866, 2023-02-14:
@mrdude2478 Winbond if genuine known to be a high quality flash.
Modified the original github actions to build esptool.py Apps and compile and push stubs in github repo when changed. Maybe usefull. Just clone and enable Github Actions
https://github.com/Jason2866/esptool/blob/release/v4/.github/workflows/build_esptool.yml

radimkarnis, 2023-02-14:
@mrdude2478 thanks for verifying! as @Jason2866 said, you need to use the `--add-data` option with pyinstaller and point it to the stub files.

mrdude2478, 2023-02-14:
Thanks, I'll test that out later, I need to have a sleep now as I am struggling to keep my eyes open. Thanks for all the help.

Jason2866, 2023-02-14:
@mrdude2478 Compiled versions with Clock 160Mhz built from latest master, just clock changed https://github.com/Jason2866/esptool/releases/tag/v4.5.1

mrdude2478, 2023-02-14:
> @mrdude2478 Compiled versions with Clock 160Mhz built from latest master, just clock changed https://github.com/Jason2866/esptool/releases/tag/v4.5.1

Thank you very much. That's very much appreciated.

radimkarnis, 2023-02-14:
@Jason2866 thanks for making that easier. @mrdude2478 if this version works for you, I will bring this change upstream and add it to esptool.

mrdude2478, 2023-02-14:
> @Jason2866 thanks for making that easier. @mrdude2478 if this version works for you, I will bring this change upstream and add it to esptool.

Flashing is a 100% improvemnet, I don't get any errors with this version, however reading still has random read issues. I checked from that github page again, and that was the same - maybe it was a fluke the first time I used it, but definately this is an improvement for flashing the chip as I haven't had an error and I've tried flashing about 20 times to test.

Jason2866, 2023-02-14:
Just tested out of curiosity serial flash speed 2M, never worked before. I thought the problem is the USB serial chip which stalls. Seems i was wrong, tried a few times. Was always successful with 2M. So the issue was in esptool.py. Did not test since what commit this high speed is working. Nice!

mrdude2478, 2023-02-14:
With Pyinstaller in windows I managed to get esptool to compile into a single exe like this: I navigated into the root esptool folder and then issued this command:

```
pyinstaller --onefile --icon=ci/espressif.ico --add-data="flasher_stub/*.c;./flasher_stub/" --add-data="flasher_stub/*.py;./flasher_stub/" --add-data="flasher_stub/ld/*.ld;./flasher_stub/ld/" --add-data="flasher_stub/include/*.h;./flasher_stub/include/" --add-data="esptool/*.py;./esptool/" --add-data="esptool/targets/*.py;./esptool/targets/" --add-data="esptool/targets/stub_flasher/*.json;./esptool/targets/stub_flasher/" esptool.py
```

This worked fine and put all the files needed into the proper locations, that command line seems a little long compared to the other examples posted for linux, did I do something wrong & there's perhaps a shorter way??? I am only asking for future reference.

@Jason2866 , did you try reading the entire flash contents as well as writing? With 4.2.1 I can succesfully read partitions if I set the offsets and partition size for firmware/bootloader/partitions sections of the chip - but not the entire chip - so I need to read a section at a time and then manually join them. With the exe you posted I can flash really well (thanks), but reading is a different matter - is there something you can adjust in the code to lower the reading clock speed of the chip as I believe this is why there's an issue?

radimkarnis, 2023-02-15:
> I thought the problem is the USB serial chip which stalls.

@Jason2866 it is important to note that the fix we are talking about only applies to [USB-Serial/JTAG mode](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32s3/api-guides/usb-serial-jtag-console.html) on the S3. In this mode, a USB-to-UART bridge ("USB serial chip") is not used.

Jason2866, 2023-02-15:
@mrdude2478  I am far away to have a clue how to do this. Just copied the value for 160Mhz from the provided test branch in the actual code base. Did not test to read the whole flash with this version. Had no issues with the 240Mhz clocked one so i do not expect this version won't work. Will have a look when i find time for.

Jason2866, 2023-02-15:
> @Jason2866 it is important to note that the fix we are talking about only applies to [USB-Serial/JTAG [...]

Tried both modes (JTAG and classic Serial). Both worked fine for flashing. So a change was done since the last time trying higher USB speed.

closed 2023-02-16 by espressif-bot

mrdude2478, 2023-02-25:
Please re-open issue.

After trying the new builds I was still having issues with crashing, even after the cpu clock speed was reduced by half. I ended up installing the sdk and have found that this was causing the issues in reading/writing.

esptool\flasher_stub\include\soc_support.h

Picture below shows mods:
[image] 

Altering the code to what is shown in the picture and then re-compiling the stubs fixed the errors I was having and now gives the same stability as this commit version 8386dcbfb0f72faf1f08d6e908f5119e2dd4f5e9. After this is where the issues began. I looked at that code and am unsure what adding these addresses on to those clock registers actually does, as when the code was altered the chip is able to be read much better. I put the cpu period back to 240 Mhz, and this was also working fine.

designongit, 2025-05-12:
In case it helps anyone else, I hit an issue where I couldn't program my custom ESP32-C3 board with my own FW (not an ESP example project) as it kept crashing.

I added the `--no-stub` and it worked (count of 3 times so far, with no failures).  Before I discovered this tip above from @radimkarnis (thank you!!) below is the crash I would see when programming over USB from my MacOS Sequoia 15.4.1 (esptool.py v4.8.1, 

```esptool.py --chip esp32c3 -p /dev/tty.usbmodem11401 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 /firmware/bootloader.bin 0x10000 /firmware/app.bin 0x8000 /firmware/partition-table.bin```

```esptool.py v4.8.1
Serial port /dev/tty.usbmodem11401
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:cd:b0:a8:d1:1c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x000e5fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash params set to 0x022f
SHA digest in image updated
Compressed 20832 bytes to 13061...
Wrote 20832 bytes (13061 compressed) at 0x00000000 in 1.5 seconds (effective 107.7 kbit/s)...
Hash of data verified.
Compressed 875568 bytes to 499173...
Writing at 0x0003055e... (12 %)Traceback (most recent call last):
  File "/Users/action/.espressif/python_env/idf5.2_py3.13_env/lib/python3.13/site-packages/esptool/__init__.py", line 1314, in _main
    main()
    ~~~~^^
  File "/Users/action/.espressif/python_env/idf5.2_py3.13_env/lib/python3.13/site-packages/esptool/__init__.py", line 1036, in main
    operation_func(esp, args)
    ~~~~~~~~~~~~~~^^^^^^^^^^^
  File "/Users/action/.espressif/python_env/idf5.2_py3.13_env/lib/python3.13/site-packages/esptool/cmds.py", line 646, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/action/.espressif/python_env/idf5.2_py3.13_env/lib/python3.13/site-packages/esptool/loader.py", line 127, in inner
    return func(*args, **kwargs)
  File "/Users/action/.espressif/python_env/idf5.2_py3.13_env/lib/python3.13/site-packages/esptool/loader.py", line 1132, in flash_defl_block
    self.check_command(
    ~~~~~~~~~~~~~~~~~~^
        "write compressed data to flash after seq %d" % seq,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
        timeout=timeout,
        ^^^^^^^^^^^^^^^^
    )
    ^
  File "/Users/action/.espressif/python_env/idf5.2_py3.13_env/lib/python3.13/site-packages/esptool/loader.py", line 510, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
                ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/action/.espressif/python_env/idf5.2_py3.13_env/lib/python3.13/site-packages/esptool/loader.py", line 479, in command
    p = self.read()
  File "/Users/action/.espressif/python_env/idf5.2_py3.13_env/lib/python3.13/site-packages/esptool/loader.py", line 412, in read
    return next(self._slip_reader)
StopIteration

A fatal error occurred: The chip stopped responding.
```

Adding `--no-stub` as shown below led to screams of happiness:
```esptool.py --chip esp32c3 -p /dev/tty.usbmodem11401 -b 460800 --before=default_reset --no-stub --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 /firmware/bootloader.bin 0x10000 /firmware/app.bin 0x8000 /firmware/partition-table.bin```

```esptool.py v4.8.1
Serial port /dev/tty.usbmodem11401
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:cd:b0:a8:d1:1c
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x000e5fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Erasing flash...
Flash params set to 0x022f
SHA digest in image updated
Took 0.09s to erase flash block
Wrote 21504 bytes at 0x00000000 in 1.9 seconds (90.2 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.41s to erase flash block
Wrote 876544 bytes at 0x00010000 in 77.9 seconds (90.1 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.03s to erase flash block
Wrote 3072 bytes at 0x00008000 in 0.3 seconds (90.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...```

I'd love to know the "why" behind this issue.  I tried writing as slow as 9600 baud, but this is the only thing that worked for me.