espressif/esptool #833: esp32 dev kit v1 download project failed

willi4m26, 2023-02-04:
### Operating System

windows 8.1

### Esptool Version

2022.12.8

### Python Version

3.9.6

### Chip Description

esp32 dev kit v1

### Hardware Configuration

GPIO5

### How is Esptool Run

web, home assistant

### Esptool Output

```shell
esphome:
  name: prova1switch.yaml

esp32:
  board: esp32-c3-devkitm-1

  
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "aPEj0LGZJNd2Pd3iBeu5UHat5MVEDVwxB1+b2VYyK*****"

ota:
  password: "ee2bb2080ab1ad4291e421ef35***"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Prova1Switch Fallback Hotspot"
    password: "IzQn5JSbBl9V"

captive_portal:
    
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO5
      mode: INPUT_PULLUP
    name: "prova12Finestra12"
    device_class: window
```

### More Information

hi, I'm trying to install this code on an esp32 dev kit v1 but it always gives me this error when I try to download the firmware to install it via the esphome web tool. I download through the home assistant of the project. I attach the error and the project. I tried to put various boards. I don't know what I'm wrong. I'm at the beginning and I've read the documentation but maybe I didn't understand much.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-02-04:
I'm sorry but we cannot help with this. This is not an esptool issue. According to the error, the file doesn't exist. That means probably your project has not built it.

closed 2023-02-04 by dobairoland

radimkarnis, 2023-02-06:
Hi @willi4m26,
please see the esphome project issue tracker: https://github.com/esphome/issues
If you don't find a similar issue there, open a ticket in that repository.