espressif/esptool #834: combined bin file not working

deus-est-machina, 2023-02-04:
### Operating System

Windows 11

### Esptool Version

v3.9.3

### Python Version

3.10.0

### Chip Description

ESP32-WROOM-32U

### Device Description

None

### Hardware Configuration

None

### How is Esptool Run

Windows GUI

### Full Esptool Command Line that Was Run

click combine button

### Esptool Output

```shell
test offset :  4096 0x1000
case ok
test offset :  32768 0x8000
case ok
test offset :  65536 0x10000
case ok
dl_list:
 [['bootloader.bin', 4096], ['partition-table.bin', 32768], ['micropython.bin', 65536]]
[4096, 32768, 65536]
['bootloader.bin', 'partition-table.bin', 'micropython.bin']
i:  0
i:  1
i:  2
```

### More Information

When I flash 3 separate bin files it works.

When I flash the combined bin files, I get this on my esp32:
```
Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - reset the device and try again;
  - check connection properties;
  - make sure the device has suitable firmware;
  - make sure the device is not in bootloader mode.
```

### Other Steps to Reproduce

Just combine 3 working bin files into one... and it doesn't work.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-02-05:
I'm sorry but esptool doesn't have a v3.9.3 release nor a Windows GUI. Please try to reproduce the issue with the latest master 
branch, or the v4.4 or v4.5-dev2 releases. 

We cannot reproduce and fix any issues which are not produced by the above releases - without any GUI which might be embed it.

Combinations of bins are working so we will need more information to be able to address this. You could:
1. erase the flash
2. flash those individual bins
3. read back the flash content and compare it with the merged bin. Please report back about the differences.

deus-est-machina, 2023-02-05:
I am using this GUI:

[image]

Am I using a wrong tool here? Or am I on the wrong github? ':D

dobairoland, 2023-02-05:
That is a different tool but uses esptool under the hood. I don't think it has a GH issue tracker and I don't know which version of esptool is used inside.

deus-est-machina, 2023-02-05:
Ok ^^

I will try with Unix and the most recent esptool. I guess we can close this one since it is not relevant.

I will open a new issue if needed, and reference this if needed.

Thank you for your time. Apologies for taking it.

dobairoland, 2023-02-05:
> I will try with Unix and the most recent esptool.

You don't need Unix. You can install esptool in command line. Install Python and then `pip install esptool` will install the latest one. 

We can keep this open until you investigate.

deus-est-machina, 2023-02-05:
Ok, thanks.

Just to be sure, I now need to used [merge_bin](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#merge-binaries-for-flashing-merge-bin) ? I have already gone through the link-chain from #254, which led me to the merge_bin...

deus-est-machina, 2023-02-05:
I just used `esptool.py v4.4`:

`esptool.py --chip esp32 make_image test.bin -f bootloader.bin -a 0x1000 -f partition-table.bin -a 0x8000 -f micropython.bin -a 0x10000`

adapted from [here](https://github.com/windowsair/wireless-esp8266-dap/issues/39#issuecomment-1246198183)

burned it using Thonny, but to no avail:

[image]

deus-est-machina, 2023-02-05:
bin files can be found [here](https://drive.google.com/drive/folders/1ToirbTi6l2fVH2uO_suHT1LQnVnf5JCW?usp=sharing)

Jason2866, 2023-02-05:
@deus-est-machina  you are mixing stuff. Most probably the combined binary is correctly generated from esptool.py. That the esp32 does not work as expected is a different thing. The used files combined to a merged bin have to "fit" exactly together. "Just" using some files will not work.

deus-est-machina, 2023-02-06:
@Jason2866

There are 3 bins:
`bootloader.bin`
`partition_table.bin`
`micropython.bin`

When these three are written to the esp seperately, the esp runs.

I then provide 2 other bins:
`merge_bin.bin`
`make_image.bin`

To illustrate that if I combine the aforementioned 3 bins using `esptool.py`, the esp does not run. 

If I use either merge_bin or make_image (or any other combining tool out there worth mentioning), the esp does not boot correctly.

And lastly, I provide `micropython1-17.bin` which does work. It is completely irrelevant other than to state that the esp is not broken, and that there are bins that do actually work. 

And then, if anything else is needed, I would provide it. 

This here thread is to figure out "why" my setup is faulty. I am sure it is user error, but I would still like to know my error.

deus-est-machina, 2023-02-06:
The problem is not with `esptool`.

`esptool.py -t -p COM7 write_flash -fm dio -fs 4MB -ff 40m 0x0 .\merge_flash.bin > .\merge_flash.log`

Works perfect. The problem seems to be on the side of Thonny. I updated the esptool to v4.4 from v3..

Still does not explain the fact that some combined bins works with Thonny, while the combined bins I created do not work... but it's not this threads concern anymore.

closed 2023-02-06 by deus-est-machina

dobairoland, 2023-02-06:
> bin files can be found [here](https://drive.google.com/drive/folders/1ToirbTi6l2fVH2uO_suHT1LQnVnf5JCW?usp=sharing)

I cannot access them. Have you deleted them?

> To illustrate that if I combine the aforementioned 3 bins using esptool.py, the esp does not run.

Have you connected to the serial output to find out what is going on (what is printed)?

deus-est-machina, 2023-02-08:
@dobairoland 

Hi, I update the [link](https://drive.google.com/drive/folders/1_21PCChCbG11n2Ot4Nfs4dy1yqJV7UV-?usp=share_link)

> Have you connected to the serial output to find out what is going on (what is printed)?

The serial output I posted was only from Thonny:

[image]

I had a look at the Thonny packages, and it had V3.3 `esptool` library. I then updated this to v4.4
[image]
In the hopes that this would solve my problem... it still didn't fix the problem.

But, like I mentioned, if I command line (Note: this is not run in the Thonny venv)
`esptool.py -t -p COM7 write_flash -fm dio -fs 4MB -ff 40m 0x0 .\merge_flash.bin > .\merge_flash.log`

It works.

deus-est-machina, 2023-02-08:
Yeah, that's what I thought as well ^^

Thank you for your assistance though. It was a necessary link in the chain.

radimkarnis, 2023-02-08:
> The serial output I posted was only from Thonny:

It seems to me that the only issue is Thonny not being able to open a serial terminal and communicate with the chip. If you connect to the chip with with another serial terminal program (`screen`, `miniterm`, ...) do you see any output?

deus-est-machina, 2023-02-08:
@radimkarnis 
I didn't try another serial program yet, I'll get back to you on this as soon as I have access to the chip again.