espressif/esptool #835: esptool.py install from Pypi

wittend, 2023-02-04:
### Operating System

Linux Mint 21 (Ubuntu)

### Esptool Version

v 4.4 

### Python Version

Python 3.10.6

### Full Esptool Command Line that Was Run

pip3 install --upgrade esptool  <also tried sudo>

### Esptool Output

```shell
NA - See attached file below.
```

### What is the Expected Behaviour?

Install esptool.py using pip.

### More Information

The problem seems to center on the package reedsolo.

[Pip3 esptool install error output.txt](https://github.com/espressif/esptool/files/10609688/Pip3.esptool.install.error.output.txt)

dobairoland, 2023-02-05:
This is a duplicate of https://github.com/espressif/esptool/issues/829. Please see workaround there.

closed 2023-02-05 by dobairoland

wittend, 2023-02-05:
Thanks!  I didn't see that post.

Dave