espressif/esptool #836: "Unable to verify flash chip connection" with strange boot log.

jaimet, 2023-02-12:
### Operating System

Debian Linux 11 (Bullseye)

### Esptool Version

v4.5.dev3-8-g9929978

### Python Version

Python 3.9.2

### Chip Description

ESP8266

### Device Description

[Wemos Lolin D1 mini Pro V2](https://www.wemos.cc/en/latest/d1/d1_mini_pro.html) connected to my linux laptop using a 1-meter "USB Type A to Micro" cable.

I have also connected the board's battery interface to an 18650 Li-ion cell.

### Hardware Configuration

The only 2 things connected to the board are the USB cable and an 18650 (as explained above).

### How is Esptool Run

No IDE. I'm just running it from "mate-terminal".

### Full Esptool Command Line that Was Run

esptool.py -p /dev/ttyUSB0 -b 9600 flash_id

### Esptool Output

```shell
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 34:94:54:61:ac:3b
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Unable to verify flash chip connection (No serial data received.).
```

### More Information

The bit that I find really interesting is that I have succeeded in the ESP8266's boot log by using `pyserial-miniterm /dev/ttyUSB0 74880`.

When I press the RST ("Reset?") button on the board, I only get 1 line in the terminal:

` ets Jan  8 2013,rst cause:2, boot mode:(3,7)`

The [boot log example in the Espressif documentation](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#boot-log) shows this 1 line followed by an 11-line stanza (`load ...` to `csum ...`):

```
ets Jan  8 2014,rst cause 1, boot mode:(3,7)

load 0x40100000, len 24236, room 16
tail 12
chksum 0xb7
ho 0 tail 12 room 4
load 0x3ffe8000, len 3008, room 12
tail 4
chksum 0x2c
load 0x3ffe8bc0, len 4816, room 4
tail 12
chksum 0x46
csum 0x46
```

I don't see any of this stanza at all. Is this board irreparably broken? 

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-02-13:
Hello @jaimet,
we can try debugging, but this looks to me like a faulty HW. `A fatal error occurred: Unable to verify flash chip connection (No serial data received.).` This message is concerning.

Let's first try to verify if the flash memory chip is readable:

Please run `flash_id` with the `--no-stub` argument: `esptool.py --no-stub -p /dev/ttyUSB0 flash_id` and you should see something like:
```
Manufacturer: 68
Device: 4015
Detected flash size: 2MB
```

Also, please disconnect the battery when you are connecting the devkit with a USB cable. Powering the board from two sources might be dangerous if the circuitry is not implemented correctly.

radimkarnis, 2023-02-13:
@jaimet one last thing to try is to see if an older version of esptool shows different behavior. If, for instance, esptool v3.3 works, then this might be a regression in our software. If not, it is most likely a faulty devkit.

jaimet, 2023-02-14:
@radimkarnis I've checked this using esptool v3.3 and unfortunately the results are identical:

```
$ ./esptool version
esptool.py v3.3
3.3

$ ./esptool --no-stub -p /dev/ttyUSB0 -b 9600 flash_id
esptool.py v3.3
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 34:94:54:61:ac:3b
Enabling default SPI flash mode...

A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption.
```

I am now certain that this board is faulty.

I've ordered a new "ESP32 Lite" and I am looking forward to it arriving this Thursday so that I can continue my project. 

Again, thank you for helping me to check this board. :pray: With best wishes, Jaime.

migue27au, 2023-06-03:
I had the same problem and solved it as follows. At first the esp32-s3 worked correctly. However, after uploading one of my sketches it gave me this error and I could not upload another sketch because of this. To fix it I did the following.
1. Create an empty sketch with a debug line in the loop. Something like Serial.println("debug line");. I tried to upload this code and connect via SERIAL to see if it was uploaded correctly.
2. Upload the code directly from esptool.py. As I program it from the arduino ide, the easiest way is to activate the upload debug in the ide. File > preferences> "Show detailed output while" > "Upload".
3. Upload the sketch from the debug line. It will generate the error "A fatal error occurred: Unable to verify flash chip connection (No serial data received.)." But at the top of the arduino IDE console the executed esptool.py command will appear.
4. Copy the command and run it, adding the --no-stub parameter. This will allow to upload the new sketch. With this you can upload sketches from the arduino IDE again.
5. Fix the bug that caused this error. In my case it was due to a "delete buffer" that I did not place correctly.

In addition to this error I got the following errors:
- Core dump flash config is corrupted!
- A fatal error occurred: No serial data received.

osmanerikci, 2023-08-11:
> I had the same problem and solved it as follows. At first the esp32-s3 worked correctly. However, after uploading one of my sketches it gave me this error and I could not upload another sketch because of this. To fix it I did the following.
> 
> 1. Create an empty sketch with a debug line in the loop. Something like Serial.println("debug line");. I tried to upload this code and connect via SERIAL to see if it was uploaded correctly.
> 2. Upload the code directly from esptool.py. As I program it from the arduino ide, the easiest way is to activate the upload debug in the ide. File > preferences> "Show detailed output while" > "Upload".
> 3. Upload the sketch from the debug line. It will generate the error "A fatal error occurred: Unable to verify flash chip connection (No serial data received.)." But at the top of the arduino IDE console the executed esptool.py command will appear. Something like this:
>    `Las variables Globales usan 48428 bytes (14%) de la memoria dinámica, dejando 279252 bytes para las variables locales. El máximo es 327680 bytes. C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1/esptool.exe --chip esp32s3 --port COM3 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 C:\Users\user\AppData\Local\Temp\arduino_build_393730/MySketch.ino.bootloader.bin 0x8000 C:\Users\user\AppData\Local\Temp\arduino_build_393730/MySketch.ino.partitions.bin 0xe000 C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9/tools/partitions/boot_app0.bin 0x10000 C:\Users\user\AppData\Local\Temp\arduino_build_393730/MySketch.ino.bin  esptool.py v4.5.1 Serial port COM3 `
> 4. Copy the command and run it, adding the **--no-stub** parameter at the begining. This will allow to upload the new sketch. With this you can upload sketches from the arduino IDE again. In my case was:
>    `C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1/esptool.exe --no-stub --chip esp32s3 --port COM3 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 C:\Users\user\AppData\Local\Temp\arduino_build_393730/MySketch.ino.bootloader.bin 0x8000 C:\Users\user\AppData\Local\Temp\arduino_build_393730/MySketch.ino.partitions.bin 0xe000 C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9/tools/partitions/boot_app0.bin 0x10000 C:\Users\user\AppData\Local\Temp\arduino_build_393730/MySketch.ino.bin  `
> 5. Fix the bug that caused this error. In my case it was due to a "delete buffer" that I did not place correctly.
> 
> In addition to this error I got the following errors:
> 
> * Core dump flash config is corrupted!
> * A fatal error occurred: No serial data received.

Some Heroes Don't Wear Capes.

eloquentarduino, 2023-08-28:
@migue27au you save me!

jaimet, 2023-08-28:
@migue27au Unfortunately I no longer have the "faulty" device so I cannot do anymore testing with it to see whether your fix/workaround solves the problem that I had with _my_ device. Would you like me to re-open this bug report? Also (and if I have understood your post correctly!), if you have found a _software_ fix to what was previously considered to be a _hardware_ problem, do you think there would be any benefit in "@-mentioning" radimkarnis ?

RogerParkinson, 2023-09-14:
I just tried the --no-stub fix given by @migue27au  above on an ESP32-S3 and it worked like a charm.
I was ready to order a new board (and they take weeks to arrive where I live) but now I can continue.
Great work @migue27au and thanks.

Blackdead1521g, 2023-10-27:
Una consulta disculpen, cómo se ejecuta un comando en arduino ide o debe ser desde la terminal?

RogerParkinson, 2023-10-28:
@Blackdead1521g from the terminal see @migue27au post above (jun 4) step 4.

camp-easy, 2023-11-04:
> 4\. --no-stub

i did it man but i receeivd this:
esptool.py v4.5.1
Serial port COM5
Connecting...
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: d4:f9:8d:04:6f:20
Enabling default SPI flash mode...
Configuring flash size...
Erasing flash (this may take a while)...

A fatal error occurred: ESP32-C3 ROM does not support function erase_flash.

RogerParkinson, 2023-11-07:
@camp-easy did you abbreviate the command? What I do is get the Arduino IDE to log the upload command (setting in preferences) then copy that to a txt editor and edit the --no-stub command into it. And it needs to be an upload, not just an erase_flash because it doesn't accept --no-stub with erase_flash. Do get the IDE to generate the command for flashing your code, then edit that command to include the --no-stub

robert1x2, 2023-12-09:
Following helped for me:

I brought it into download flashing mode. So I pulled down pin 9 and rebooted esp32c3 using the onboard boot button. After reboot it waited for new code to upload. This was printed out in my Arduino IDE.

This docu I used: https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html#manual-bootloader

RogerParkinson, 2023-12-12:
For my esp32c3 I didn't have to do anything to pin 9. I just unplugged the USB, held the boot button down and plugged it back in. That put it into boot mode and it was happy to load new code (after refusing it earlier).

toddkrein, 2024-01-29:
There is still a problem. One data point: I have the same version of the IDE and board package on two different laptops. On one, 100% success rate. On the other, maybe 5% success rate, unless I do the copy/past of the command with the --no-stub option. I'm happy to run experiments to determine the difference if anyone has suggestions.

toddkrein, 2024-01-29:
Any way to re-open this bug?

jaimet, 2024-01-30:
> Any way to re-open this bug?

Done as requested.

dobairoland, 2024-01-30:
This is closed because we don't have information how to reproduce it. As far as I can tell, it is possible to put the chip with Arduino IDE into a state where it cannot be properly flashed anymore. If you could provide any hints on how to make the Arduino project achieve this then we will happily try to find a solution.

Until then we cannot do anything. The discussion is open and search engines can find it even if this is closed.

closed 2024-01-30 by dobairoland

jaimet, 2024-01-30:
> This is closed because we don't have information how to reproduce it. As far as I can tell, it is possible to put the chip with Arduino IDE into a state where it cannot be properly flashed anymore. [...]

@dobairoland 

If "we" are unable to provide information on how to "break" the chip, is there any way to get useful information out of a chip _which is already broken_?

I ask because it seems as though all we can do at the moment is to:
1) observe that the chip is broken, and
2) do something that "repairs" it.

(Neither of these provides information on how to break it in the first place.)

PS: you mentioned the Arduino IDE. I was _not_ using the Arduino IDE - I _was_ using `esptool`.

dobairoland, 2024-01-30:
Yes, I see your problem and I agree. And hence I don't know what we could do about it in esptool.

> PS: you mentioned the Arduino IDE. I was not using the Arduino IDE - I was using esptool.

But how you made the binaries which "broke" the chip? I was assuming that it was made in Arduino IDE. So far I know about only issues which were caused by binaries generated by Arduino IDE.

jaimet, 2024-01-30:
> But how you made the binaries which "broke" the chip? I was assuming that it was made in Arduino IDE. So far I know about only issues which were caused by binaries generated by Arduino IDE.

_If I remember correctly_ (it was a year ago!), I didn't make the binaries - I was flashing _pre-built_ [micropython binaries](https://micropython.org/download/ESP32_GENERIC/).

toddkrein, 2024-01-30:
> This is closed because we don't have information how to reproduce it. As far as I can tell, it is possible to put the chip with Arduino IDE into a state where it cannot be properly flashed anymore. [...]

I can reproduce it 100% of the time, and have both working and failing development machines, and multiple dev boards. I can take a board that works 100% of the time on my old laptop and it fails (apparently) 100% of the time on the new laptop. I can take a board that fails 100% of the time on the new laptop and it works 100% on the old laptop. So it 100% DOES NOT follow the hardware.

I'll take the binaries from each laptop and move it to the other and try burning it. That should tell us if it's a binary-specific problem (which doesn't match the symptoms I see, so I think it's unlikely). I can also install the IDE on my linux machine and see if it fails there. 

IMHO (since I do this for a living), the fact that we don't have a simple way to create a reproducible / transferable failure isn't a valid reason to close the bug. We still clearly have a bug. (And I can find threads all over the net referring to it in difference contexts both Arduino and non-arduino.) I do understand if you somehow want to mark it "not-actively working on" until we get a transferable case.

toddkrein, 2024-02-01:
Did some more experimentation. Using an ESP32-S3, and the same sketch, both laptops were repeatedly successful in programming the device. So it does seem device specific. I then took the binaries from the working laptop and moved them to the failing laptop, and the programming failed repeatedly. Taking the binaries from the failing laptop and moving them to the working laptop worked repeatedly.

So I think it's pretty clear that the problem likely lies in the serial connection between the laptop and the device, and a timing difference between the -C3 and the -S3  is my vote as the most likely the culprit.

toddkrein, 2024-02-01:
Please re-open this bug.

toddkrein, 2024-02-01:
Curiously enough, I just noticed that the original problem was reported with an ESP8266, and I'm having the same problem with an ESP32-C3. The --no-stub fix works for me, so the problem must be in that code. I'm also adding a trace log of the failing run. I've compared it to a trace on the working laptop, and it's essentially identical until the chip just stops responding. 
[esp32-c3-fail-trace.txt](https://github.com/espressif/esptool/files/14119847/esp32-c3-fail-trace.txt)

toddkrein, 2024-02-01:
Arduino work around (thanks to  @migue27au  for the find). Edit the platform.txt file for ESP. In my case, ~/AppData/local/arduino15/packages/esp32/hardware/esp32/2.0.14. Find the line "Upload Sketch" and just below it is the definition for "tools.esptool_py.upload.pattern_args". add "--no-stub" before "--port". save it. Exit the IDE and relaunch it. It now seems to work 100% of the time for me, even if it takes longer to program. 

Next step would be (and I'll try if I can figure out how to recompile the stub) to replace the command implementations in the stub with the original implementaitons in the ROM until it works.

toddkrein, 2024-02-01:
I've cloned the latest version of the repo, and the problem still reproduces.

radimkarnis, 2024-02-01:
@toddkrein thanks for looking into this. 
What if you try flashing WITH compression when using the ROM bootloader (without `--no-stub`)? Compression can be enforced with `--compress` or `-z` flags.

This way we can see if the issue is generally caused by compressing, or just when compressing with the stub.

toddkrein, 2024-02-01:
I uninstalled python from my machine, because (anecdotally) it seems like after I ran the commands to set up for esptool development, the behaviour changed. I haven't had a chance to revisit that.

I'm using Arduino 2.2.1 and Espressif 2.0.14 (which has esptool v4.5.1). I'm modifying the Platform.txt file to change the command parameters, but I'm also doing a copy/paste of the command from the Arduino output and running it in a virgin CMD window, because that avoids the time it takes to recompile every time.

--no_stub  -z           Works
--no_stub                Works
-z                             Fails
                                Fails

So for this particular instance, it looks like compression isn't the issue, the stub is. (But keep in mind that the same stub binary works when programmed by a different laptop.)

dobairoland, 2024-02-02:
@toddkrein Does the issue happen for all projects you build or do you need something special?

Can we build a publicly available example which is failing for you?

toddkrein, 2024-02-02:
> @toddkrein Does the issue happen for all projects you build or do you need something special?
> 
> Can we build a publicly available example which is failing for you?

I haven't tried anything else. Pick something and I'll give it a whirl. Obviously, the bigger the image the more likely the failure. But since the exact same compiled image works on one laptop and not the other, I think that's a dead end.

It often pukes when it's configuring the Flash, which implies the sketch isn't the issue. My gut is telling me it's something orthogonal - timing, or voltage, or something screwing with the serial drivers. We know it isn't the HW or who compiled the binary.

Here's a stretch - is there a way for me to dump either memory or the register set after a failure? What if it's an undefined variable, and some difference in the USB voltage is impacting it? What if the stub code is getting corrupted somehow? What if one of the SPI config writes isn't working as it should? I think we're going to find it's a wacky edge case.

toddkrein, 2024-02-02:
Here's a random idea... Is there a way to load the stub but not start executing it? Or load it and then verify that it loaded correctly? Do we have the source for the ROM bootloader to compare how functions are implemented? Or better yet, can we create a version of the stub that calls back to the original ROM bootloader to execute the commands?

radimkarnis, 2024-02-02:
> Is there a way to load the stub but not start executing it?

It is possible to prevent the stub from running, you can just delete the entrypoint [here](https://github.com/espressif/esptool/blob/master/esptool/loader.py#L1025) ([see the docs](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/serial-protocol.html#:~:text=0x06-,MEM_END,-Finish%20RAM%20Download) for the `MEM_END` serial protocol command)

> Do we have the source for the ROM bootloader to compare how functions are implemented?

The ROM bootloader code is not public.

> Or better yet, can we create a version of the stub that calls back to the original ROM bootloader to execute the commands?

That defeats the purpose of the stub. It is meant to improve or patch the ROM bootloader functions. If the solution is to just call the ROM functions again, we might as well just get rid of the stub.

On the other hand, the stub is often calling implementations from the ROM: You can see all of the exposed functions from the ROM that can be used in the stub [here](https://github.com/espressif/esptool/blob/master/flasher_stub/ld/rom_32c3.ld#L125).

toddkrein, 2024-02-02:
I'm not suggesting that we get rid of the stub long term, but rather to help pinpoint what's failing in the stub.

tapir, 2024-04-14:
I think this issue needs to be re-open. I don't know what I did but a perfectly running C3 started to give the same error and --no-stub fixed it.

toddkrein, 2024-04-16:
.... And it no longer reproduces on my laptop, after a reboot. So I agree that the problem is very real, but I'm at a loss as to where to look for the problem source.

roshanyadav29, 2024-04-17:
I am facing the same issue. Using ESP32 C3 Board and Arduino IDE v2.3.2. I can confirm --no-stub fix works. Just to add, copy and paste that command into your default command window along with no-stub, and run that. 

It used to work just fine for the last 2 months and now when I got a new c3 board and tried to upload the same code, this issue popped up. Now I can't upload the code using Arduino IDE in both the C3 Boards. The problem is, I tried a few sketches, but every time I have to do the --no-stub fix. Is this the only way?

roshanyadav29, 2024-04-17:
Recommend trying this out to everyone that has this problem. [Xinyuan-LilyGO/T-Display-S3/issues/162]
My understanding was this boot button wouldn't be helpful in esp32c3 (specifically in dfrobot beetle, which uses the D+,D- lines instead of the UART ones.) I have no idea why this worked. But after this, the code uploaded just fine.

toddkrein, 2024-04-17:
To the next person who encounters this: I'd appreciate someone replicating or refuting my anecdotal that unplugging and rebooting their machine changes the behaviour.

tapir, 2024-04-17:
It doesn't for me. It started to happen randomly and it never goes away apart from the --no-stub fix. I've tried same computer linux and windows. unplug/plug/reboot doesn't have any effect.

toddkrein, 2024-04-17:
Thanks, and very curious. That means that something else changed on my laptop.

tapir, 2024-04-17:
I don't think so. It literally started to happen to me inbetween sketch
uploads. One minute it was working and the other it stopped working. Pretty
sure esptool has a bug that causes it. Or the stub itself has an obscure
bug.

toddkrein, 2024-04-17:
Same here, IIRC. But the same modules worked just fine on my other laptop that was up and running at the same time, so it's not a persistent error. And then it started working a couple of days later on the same laptop, and I don't think I installed or updated anything.  That brings me back to something between the python program managing the download, the serial port and its driver and the usb port on the C3 module. Again, using exactly the same stub binary, the same sketch binary, and the same module on a different laptop worked 100% of the time. So there are limited number of things that can be the source.

tapir, 2024-04-17:
can somebody re-open this issue?

dobairoland, 2024-04-18:
I'm following the conversation but it looks like people are just coming here to vent their frustration. I'm sorry we are not re-opening this until we receive information on how to reproduce this. Stories like "everything worked fine but then suddenly it stopped working" is not helpful and is not something we can start with.

You are suggesting that this is a fault of esptool. But all the issues were caused by Arduino projects. There were similar cases for Micropython as well, but mostly Arduino. I'm not aware of other cases (ESP-IDF or dependent frameworks, Platformio, ...).

jaimet, 2024-04-18:
> You are suggesting that this is a fault of esptool. But all the issues were caused by Arduino projects.

@dobairoland Roland, I'm the OP and, as I explained to you [here](https://github.com/espressif/esptool/issues/836#issuecomment-1916853795), my problem had _nothing_ to do with an Arduino project. Again, I was _not_ using the Arduino IDE - I _was_ using `esptool` (and _only_ `esptool`).

You say 'it looks like people are just coming here to vent their frustration'. I suspect **everyone** here is frustrated (psychological projection is an insidious trait that we **all** possess.)

You say "we are not re-opening this until we receive information on how to reproduce this". It is clear that up to now, _no-one_ is able to provide this information. Consequently, I _calmly_ ask [again](https://github.com/espressif/esptool/issues/836#issuecomment-1916853795):

> Is there any way to get useful information out of a chip which is already broken?
> 
> I ask because it seems as though all we can do at the moment is to:
> 
>     observe that the chip is broken, and
>     do something that "repairs" it.
> 
> (Neither of these provides information on how to break it in the first place.)

jaimet, 2024-04-18:
> can somebody re-open this issue?

I can, but I will not. Re-opening this issue will achieve less than nothing - it will antagonize espressif's employees/representatives, and it will just be immediately closed again. (I'm not sure where the pressure to close unresolved bugs originates from, but it clearly exists.) HTH.

dobairoland, 2024-04-18:
@jaimet My previous post was aimed at people who hijacked your report and demanding to reopen it.

> Roland, I'm the OP and, as I explained to you https://github.com/espressif/esptool/issues/836#issuecomment-1916853795, my problem had nothing to do with an Arduino project. Again, I was not using the Arduino IDE - I was using esptool (and only esptool).

You flashed micropython and then the chip broke, you are receiving errors from esptool now. I don't say that micropython broke it but after the chip is broken I don't know what esptool can do about it. Or to better phrase, what could esptool developers do about it. Esptool is a simple serial communication tool. If the serial communication is not working then esptool cannot do anything about it. It is out-of-scope for it. Keeping the issue open doesn't change anything only we waste time on looking at thing we cannot anything about it.

> Is there any way to get useful information out of a chip which is already broken?

No, I don't think so.

tapir, 2024-04-18:
I wasn't trying to be offensive or anything. I'm also not frustrated at all because --no-stub works quite good and if anything finding this thread made me happy because it proved I don't have a faulty board. I also don't have any immediate need for a fix. I just felt like we're talking to a wall here and Espressif would like to keep it open for tracking purposes but looks like it's not how it's done there so I'll just stop suggesting what's logical in my point of view.

dobairoland, 2024-04-18:
@tapir You are receiving "Unable to verify flash chip connection" message for your ESP32-C3 and `--no-stub` is a workaround?

tapir, 2024-04-18:
Yes

dobairoland, 2024-04-18:
@tapir Do you have a project we can flash to broke the chip this way? We are flashing in our pipeline tens of different projects every day and if this would happen for us then we would've fixed it already.

If you can give us the steps to reproduce then please open a new separate issue with the steps. Thanks!

tapir, 2024-04-18:
I'm away from my workstation for a while but I'll try to prepare something. I don't think my chips are broken at all btw and there is a chance that message is a false negative.

tapir, 2024-04-18:
Removed --no-stub from my upload line and now it works somehow. I'll use it like this for a while and try to pinpoint when it starts to happen again.

maccampb, 2024-07-13:
I have the same issue... Sparkfun pocket Qwiic, esp32-C6; after first loading to a board, it takes the upload then all subsequent attempts fail.

manually setting --no-stub allows the board to load again. This version of the chip does not allow for erasing of the flash...

```
esptool.py v4.6
Serial port /dev/cu.usbmodem2101
Connecting...
Chip is ESP32-C6FH4 (QFN32) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 40:4c:ca:48:23:cc:ff:fe
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Unable to verify flash chip connection (No serial data received.).
Failed uploading: uploading error: exit status 2
```
IDE:
Version: 2.3.2
Date: 2024-02-20T09:53:59.281Z
CLI Version: 0.35.3
Copyright © 2024 Arduino SA

Arduino Board Info:

BN: ESP32 Family Device
VID: 0x303A
PID: 0x1001
SN: 40:4C:CA:48:23:CC

SantiagoOlmosMed, 2024-07-22:
> I had the same problem and solved it as follows. At first the esp32-s3 worked correctly. However, after uploading one of my sketches it gave me this error and I could not upload another sketch [...]

You are my hero bro, thanks!!!

NorthernMan54, 2024-10-09:
Hopefully this helps someone with the same issue.  

I came across this error this morning when I switched which Mac I was using for coding.   I had been successfully doing serial updates for a long while on one Mac, but when I switched to the other I had this error.  ( This Mac had been used with other ESP boards for a while, but I recently received a 4848S040CIY1 device )

I looked at what was different between the two, and the working Mac was using /dev/cu.wchusbserial.... and the non-working Mac was using /dev/cu.usbserial.....

I think did a quick look at the serial drivers on both and determined that the non-functional Mac was not using the CH34xVCPDriver serial driver.  So I went and removed the existing serial drivers from the non-functional Mac, and installed the latest CH34xVCPDriver package.  Rebooted, and everything is working.

Akil-18, 2025-04-10:
Tools -> Flash Mode -> DIO (Change from QIO to DIO)
Connect GPIO 0 to Gnd.

This worked for me.