espressif/esptool #838: 4.5: pytest is failing in few units

kloczek, 2023-02-13:
### Operating System

Linux/x86_64

### Esptool Version

4.5

### Python Version

3.8.16

### Full Esptool Command Line that Was Run

N/A

### Esptool Output

```shell
N/A
```

### What is the Expected Behaviour?

I'm packaging your module as an rpm package so I'm using the typical PEP517 based build, install and test cycle used on building packages from non-root account.
- `python3 -sBm build -w --no-isolation`
- because I'm calling `build` with `--no-isolation` I'm using during all processes only locally installed modules
- install .whl file in </install/prefix>
- run pytest with $PYTHONPATH pointing to sitearch and sitelib inside </install/prefix>
- build is performed in env which is *`cut off from access to the public network`* (pytest is executed with `-m "not network"`)

Looks like few unit wich are not using /dev/ttyUSB0 are failing.

### More Information

Here is pytest output:
<details>

```console
+ PYTHONPATH=/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.5-2.fc35.x86_64/usr/lib64/python3.8/site-packages:/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.5-2.fc35.x86_64/usr/lib/python3.8/site-packages
+ /usr/bin/pytest -ra -m 'not network' --deselect test/test_esptool.py::TestAutoDetect::test_auto_detect --deselect test/test_esptool.py::TestBootloaderHeaderRewriteCases::test_flash_header_no_magic_no_rewrite --deselect test/test_esptool.py::TestBootloaderHeaderRewriteCases::test_flash_header_rewrite --deselect test/test_esptool.py::TestErase::test_chip_erase --deselect test/test_esptool.py::TestErase::test_large_region_erase --deselect test/test_esptool.py::TestErase::test_region_erase --deselect test/test_esptool.py::TestFlashDetection::test_flash_id --deselect test/test_esptool.py::TestFlashDetection::test_flash_id_expand_args --deselect test/test_esptool.py::TestFlashDetection::test_flash_id_trace --deselect test/test_esptool.py::TestFlashEncryption::test_blank_efuse_encrypt_write_abort --deselect test/test_esptool.py::TestFlashEncryption::test_blank_efuse_encrypt_write_continue1 --deselect test/test_esptool.py::TestFlashEncryption::test_blank_efuse_encrypt_write_continue2 --deselect test/test_esptool.py::TestFlashSizes::test_flash_size_keep --deselect test/test_esptool.py::TestFlashSizes::test_high_offset --deselect test/test_esptool.py::TestFlashSizes::test_high_offset_no_compression --deselect test/test_esptool.py::TestFlashSizes::test_large_image --deselect test/test_esptool.py::TestFlashSizes::test_large_no_compression --deselect test/test_esptool.py::TestFlashSizes::test_write_no_compression_past_end_fails --deselect test/test_esptool.py::TestFlashSizes::test_write_past_end_fails --deselect test/test_esptool.py::TestFlashing::test_adjacent_flash --deselect test/test_esptool.py::TestFlashing::test_adjacent_independent_flash --deselect test/test_esptool.py::TestFlashing::test_compressed_nostub_flash --deselect test/test_esptool.py::TestFlashing::test_compressible_file --deselect test/test_esptool.py::TestFlashing::test_compressible_non_trivial_file --deselect test/test_esptool.py::TestFlashing::test_correct_offset --deselect test/test_esptool.py::TestFlashing::test_erase_before_write --deselect test/test_esptool.py::TestFlashing::test_erase_range_messages --deselect test/test_esptool.py::TestFlashing::test_highspeed_flash --deselect test/test_esptool.py::TestFlashing::test_length_not_aligned_4bytes --deselect test/test_esptool.py::TestFlashing::test_length_not_aligned_4bytes_no_compression --deselect test/test_esptool.py::TestFlashing::test_no_compression_flash --deselect test/test_esptool.py::TestFlashing::test_partition_table_then_bootloader --deselect test/test_esptool.py::TestFlashing::test_partition_table_then_bootloader_no_compression --deselect test/test_esptool.py::TestFlashing::test_partition_table_then_bootloader_nostub --deselect test/test_esptool.py::TestFlashing::test_short_flash --deselect test/test_esptool.py::TestFlashing::test_single_byte --deselect test/test_esptool.py::TestFlashing::test_write_image_for_another_target --deselect test/test_esptool.py::TestFlashing::test_write_no_overlap --deselect test/test_esptool.py::TestFlashing::test_zero_length --deselect test/test_esptool.py::TestKeepImageSettings::test_detect_size_changes_size --deselect test/test_esptool.py::TestKeepImageSettings::test_explicit_set_size_freq_mode --deselect test/test_esptool.py::TestKeepImageSettings::test_keep_does_not_change_settings --deselect test/test_esptool.py::TestLoadRAM::test_load_ram --deselect test/test_esptool.py::TestMemoryOperations::test_memory_dump --deselect test/test_esptool.py::TestMemoryOperations::test_memory_read --deselect test/test_esptool.py::TestMemoryOperations::test_memory_write --deselect test/test_esptool.py::TestReadIdentityValues::test_read_mac --deselect test/test_esptool.py::TestReadWriteMemory::test_read_chip_description --deselect test/test_esptool.py::TestReadWriteMemory::test_read_write_flash_status --deselect test/test_esptool.py::TestReadWriteMemory::test_read_write_memory_rom --deselect test/test_esptool.py::TestReadWriteMemory::test_read_write_memory_stub --deselect test/test_esptool.py::TestSectorBoundaries::test_end_sector --deselect test/test_esptool.py::TestSectorBoundaries::test_end_sector_uncompressed --deselect test/test_esptool.py::TestSectorBoundaries::test_overlap --deselect test/test_esptool.py::TestStubReuse::test_stub_reuse_with_synchronization --deselect test/test_esptool.py::TestVerifyCommand::test_verify_failure --deselect test/test_esptool.py::TestVerifyCommand::test_verify_success --deselect test/test_esptool.py::TestVerifyCommand::test_verify_unaligned_length --deselect test/test_esptool.py::TestVirtualPort::test_auto_detect --deselect test/test_esptool.py::TestVirtualPort::test_auto_detect_virtual_port --deselect test/test_esptool.py::TestVirtualPort::test_highspeed_flash_virtual_port
==================================================================================== test session starts ====================================================================================
platform linux -- Python 3.8.16, pytest-7.2.1, pluggy-1.0.0
rootdir: /home/tkloczko/rpmbuild/BUILD/esptool-4.5
collected 217 items / 61 deselected / 156 selected

test/test_espefuse.py .......................ssssssssssss......sss.sss..ss..ss.....ssssss.ss........ss..                                                                              [ 52%]
test/test_espsecure.py .......F...F.FF.F.........                                                                                                                                     [ 69%]
test/test_espsecure_hsm.py FF                                                                                                                                                         [ 70%]
test/test_esptool.py ss...sss.ssss..F                                                                                                                                                 [ 80%]
test/test_image_info.py .......                                                                                                                                                       [ 85%]
test/test_imagegen.py ..............                                                                                                                                                  [ 94%]
test/test_merge_bin.py ........                                                                                                                                                       [ 99%]
test/test_modules.py .                                                                                                                                                                [100%]

========================================================================================= FAILURES ==========================================================================================
_______________________________________________________________________________ TestSigning.test_sign_v2_data _______________________________________________________________________________

self = <test_espsecure.TestSigning object at 0x7fcebe46ce50>

    def test_sign_v2_data(self):
        signing_keys = [
            "rsa_secure_boot_signing_key.pem",
            "ecdsa192_secure_boot_signing_key.pem",
            "ecdsa_secure_boot_signing_key.pem",
        ]
        for key in signing_keys:
            with tempfile.NamedTemporaryFile() as output_file:
[... cut ...]
Pygments           2.14.0
PyGObject          3.43.1.dev0
pyserial           3.5
pytest             7.2.1
python-dateutil    2.8.2
python-pkcs11      0.7.0
reedsolo           1.6.1
rpm                4.17.0
scour              0.38.2
setuptools         65.6.3
six                1.16.0
smartypants        2.0.1
testtools          2.5.0
toml               0.10.2
tomli              2.0.1
typogrify          2.0.7
wheel              0.38.4
zipp               3.13.0
```
</details>

### Other Steps to Reproduce

N/A

kloczek, 2023-02-13:
Please have look one more time on what I'm reporing.
As I wtote I've --deselected all units which requires actual hardware to test and not all `test/test_espsecure.py` units are failing.The same is with `test/test_esptool.py`

Other thing that in case all units which requires hardware it would be good to add pytest mak like `@pytest.mark.ttyUSB0` mark to allow easily ommit those units which requires as=criual hardware by executing `pytest -m "not ttyUSB0"'
https://docs.pytest.org/en/7.1.x/how-to/mark.html

> The new test_espsecure_hsm.py needs some additional initialization steps to run. See [here](https://github.com/espressif/esptool/blob/01bd2269e11213834a26dbd3b53aa06fc7ff745d/.github/workflows/test_esptool.yml#L22-L29). Either you can do these or also add it to the ignore list.

OK. Thx 👍 
Will try to adapt that to my build&testing procedure 😄

radimkarnis, 2023-02-13:
As I said:
- `TestConfigFile::test_custom_reset_sequence` needs to be deselected too, it depends on real HW
- `test/test_espsecure.py` should work. [Here](https://github.com/espressif/esptool/actions/runs/4164966361/jobs/7207317175#step:5:287) is an example of a successful run - this doesn't require hardware. Please check your installation, it seems like the errors originate from the `cryptography` package.

>  it would be good to add pytest mark like @pytest.mark.ttyUSB0 mark to allow easily ommit those units which requires actual hardware

This is a good idea. Thanks for the suggestion!

closed 2023-02-23 by espressif-bot

radimkarnis, 2023-02-23:
Hi @kloczek,
you should now be able to run the whole test suite simply as `pytest -m host_test`, as described [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/contributing.html#automated-integration-tests). 
This runs all of the tests without the need for real hardware or installation of additional prerequisites.

This will be included in the next release.

kloczek, 2023-03-01:
Just tested 4.5.1and on first lok looks like it is something wrong around use `flaky`
Here is pytest output:
<details>

```console
+ PYTHONPATH=/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.5.1-2.fc35.x86_64/usr/lib64/python3.8/site-packages:/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.5.1-2.fc35.x86_64/usr/lib/python3.8/site-packages
+ /usr/bin/pytest -ra -m 'not network' -m 'not host_test'
============================= test session starts ==============================
platform linux -- Python 3.8.16, pytest-7.2.1, pluggy-1.0.0
rootdir: /home/tkloczko/rpmbuild/BUILD/esptool-4.5.1
plugins: flaky-3.7.0
collected 217 items / 144 deselected / 73 selected

test/test_espsecure_hsm.py FF                                            [  2%]
test/test_esptool.py EEEEEEEssEEEEEEEEEEEEEEEssEsEEEEEEEEEEEsEEEEEEEEEEs [ 72%]
EEEEEEEsEEEEEEEEEEsE                                                     [100%]

==================================== ERRORS ====================================
__ ERROR at setup of TestFlashEncryption.test_blank_efuse_encrypt_write_abort __

self = <flaky.flaky_pytest_plugin.FlakyPlugin object at 0x7f4585639130>
item = <Function test_blank_efuse_encrypt_write_abort>

    def pytest_runtest_setup(self, item):
        """
        Pytest hook to modify the test before it's run.

        :param item:
            The test item.
        """
        if not self._has_flaky_attributes(item):
            if hasattr(item, 'iter_markers'):
                for marker in item.iter_markers(name='flaky'):
>                   self._make_test_flaky(item, *marker.args, **marker.kwargs)
E                   TypeError: _make_test_flaky() got an unexpected keyword argument 'reruns'

/usr/lib/python3.8/site-packages/flaky/flaky_pytest_plugin.py:239: TypeError
_ ERROR at setup of TestFlashEncryption.test_blank_efuse_encrypt_write_continue1 _

self = <flaky.flaky_pytest_plugin.FlakyPlugin object at 0x7f4585639130>
item = <Function test_blank_efuse_encrypt_write_continue1>

    def pytest_runtest_setup(self, item):
        """
        Pytest hook to modify the test before it's run.

        :param item:
            The test item.
        """
        if not self._has_flaky_attributes(item):
            if hasattr(item, 'iter_markers'):
                for marker in item.iter_markers(name='flaky'):
>                   self._make_test_flaky(item, *marker.args, **marker.kwargs)
E                   TypeError: _make_test_flaky() got an unexpected keyword argument 'reruns'

/usr/lib/python3.8/site-packages/flaky/flaky_pytest_plugin.py:239: TypeError
_ ERROR at setup of TestFlashEncryption.test_blank_efuse_encrypt_write_continue2 _

self = <flaky.flaky_pytest_plugin.FlakyPlugin object at 0x7f4585639130>
item = <Function test_blank_efuse_encrypt_write_continue2>

    def pytest_runtest_setup(self, item):
        """
        Pytest hook to modify the test before it's run.

        :param item:
            The test item.
        """
        if not self._has_flaky_attributes(item):
            if hasattr(item, 'iter_markers'):
[... 1229 lines cut ...]
ERROR test/test_esptool.py::TestMemoryOperations::test_memory_dump - TypeErro...
ERROR test/test_esptool.py::TestMemoryOperations::test_memory_write - TypeErr...
ERROR test/test_esptool.py::TestMemoryOperations::test_memory_read - TypeErro...
ERROR test/test_esptool.py::TestKeepImageSettings::test_keep_does_not_change_settings
ERROR test/test_esptool.py::TestKeepImageSettings::test_detect_size_changes_size
ERROR test/test_esptool.py::TestKeepImageSettings::test_explicit_set_size_freq_mode
ERROR test/test_esptool.py::TestLoadRAM::test_load_ram - TypeError: _make_tes...
ERROR test/test_esptool.py::TestBootloaderHeaderRewriteCases::test_flash_header_rewrite
ERROR test/test_esptool.py::TestBootloaderHeaderRewriteCases::test_flash_header_no_magic_no_rewrite
ERROR test/test_esptool.py::TestAutoDetect::test_auto_detect - TypeError: _ma...
ERROR test/test_esptool.py::TestVirtualPort::test_auto_detect - TypeError: _m...
ERROR test/test_esptool.py::TestVirtualPort::test_auto_detect_virtual_port - ...
ERROR test/test_esptool.py::TestVirtualPort::test_highspeed_flash_virtual_port
ERROR test/test_esptool.py::TestReadWriteMemory::test_read_write_memory_rom
ERROR test/test_esptool.py::TestReadWriteMemory::test_read_write_memory_stub
ERROR test/test_esptool.py::TestReadWriteMemory::test_read_write_flash_status
ERROR test/test_esptool.py::TestReadWriteMemory::test_read_chip_description
ERROR test/test_esptool.py::TestConfigFile::test_custom_reset_sequence - Type...
FAILED test/test_espsecure_hsm.py::TestSigning::test_sign_v2_hsm - SystemExit...
FAILED test/test_espsecure_hsm.py::TestSigning::test_sign_v2_hsm_append_signatures_multiple_steps
=========== 2 failed, 9 skipped, 144 deselected, 62 errors in 0.95s ============
```
</details>

I'm using `flaky` 3.7.0.

radimkarnis, 2023-03-01:
Hello, 
1) we don't use `flaky` in the esptool repo at all.
2) I see you run the tests as`-m 'not host_test'` - while the `host_test` marker is supposed to be used as `-m host_test`.

kloczek, 2023-03-02:
OK I've added `pytest-rerunfailures` to build and and pytest is executed with `-m host_test` and still something is wrong ..
<details>

```console
+ PYTHONPATH=/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.5.1-2.fc35.x86_64/usr/lib64/python3.8/site-packages:/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.5.1-2.fc35.x86_64/usr/lib/python3.8/site-packages
+ /usr/bin/pytest -ra -m 'not network' -m host_test
============================= test session starts ==============================
platform linux -- Python 3.8.16, pytest-7.2.1, pluggy-1.0.0
rootdir: /home/tkloczko/rpmbuild/BUILD/esptool-4.5.1
plugins: rerunfailures-11.0
collected 217 items / 73 deselected / 144 selected

test/test_espefuse.py .......................ssssssssssss......sss.sss.. [ 34%]
ss..ss.....ssssss.ss........ss..                                         [ 56%]
test/test_espsecure.py .......F...F.FF.F.........                        [ 75%]
test/test_esptool.py ......                                              [ 79%]
test/test_image_info.py .......                                          [ 84%]
test/test_imagegen.py ..............                                     [ 93%]
test/test_merge_bin.py ........                                          [ 99%]
test/test_modules.py .                                                   [100%]

=================================== FAILURES ===================================
________________________ TestSigning.test_sign_v2_data _________________________

self = <test_espsecure.TestSigning object at 0x7fa44fe3eac0>

    def test_sign_v2_data(self):
        signing_keys = [
            "rsa_secure_boot_signing_key.pem",
            "ecdsa192_secure_boot_signing_key.pem",
            "ecdsa_secure_boot_signing_key.pem",
        ]
        for key in signing_keys:
            with tempfile.NamedTemporaryFile() as output_file:
                args = self.SignArgs(
                    "2",
                    [self._open(key)],
                    output_file.name,
                    False,
                    False,
                    None,
                    None,
                    None,
                    self._open("bootloader_unsigned_v2.bin"),
                )
>               espsecure.sign_data(args)

test/test_espsecure.py:223:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
../../BUILDROOT/esptool-4.5.1-2.fc35.x86_64/usr/lib/python3.8/site-packages/espsecure/__init__.py:331: in sign_data
    return sign_secure_boot_v2(args)
../../BUILDROOT/esptool-4.5.1-2.fc35.x86_64/usr/lib/python3.8/site-packages/espsecure/__init__.py:490: in sign_secure_boot_v2
    signature_block = generate_signature_block_using_private_key(
../../BUILDROOT/esptool-4.5.1-2.fc35.x86_64/usr/lib/python3.8/site-packages/espsecure/__init__.py:604: in generate_signature_block_using_private_key
    private_key = _load_sbv2_signing_key(keyfile.read())
../../BUILDROOT/esptool-4.5.1-2.fc35.x86_64/usr/lib/python3.8/site-packages/espsecure/__init__.py:231: in _load_sbv2_signing_key
    sk = serialization.load_pem_private_key(
/usr/lib64/python3.8/site-packages/cryptography/hazmat/primitives/serialization/base.py:22: in load_pem_private_key
    return ossl.load_pem_private_key(data, password)
/usr/lib64/python3.8/site-packages/cryptography/hazmat/backends/openssl/backend.py:921: in load_pem_private_key
    return self._load_key(
/usr/lib64/python3.8/site-packages/cryptography/hazmat/backends/openssl/backend.py:1189: in _load_key
    self._handle_key_loading_error()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <OpenSSLBackend(version: OpenSSL 3.0.5 5 Jul 2022, FIPS: False)>

    def _handle_key_loading_error(self) -> typing.NoReturn:
        errors = self._consume_errors()

[... 516 lines cut ...]
SKIPPED [1] test/test_espefuse.py:931: 512 bit keys are only supported on ESP32-S2 and S3
SKIPPED [1] test/test_espefuse.py:949: 512 bit keys are only supported on ESP32-S2 and S3
SKIPPED [1] test/test_espefuse.py:989: 512 bit keys are only supported on ESP32-S2 and S3
SKIPPED [1] test/test_espefuse.py:1059: ESP32-C2-only
SKIPPED [1] test/test_espefuse.py:1076: Only chip with 6 keys
SKIPPED [1] test/test_espefuse.py:1187: ESP32-C2-only
SKIPPED [1] test/test_espefuse.py:1210: Only chips with 6 keys
SKIPPED [1] test/test_espefuse.py:1347: ESP32-C2-only, supports 1 key block
SKIPPED [1] test/test_espefuse.py:1359: ESP32-C2-only, supports 1 key block
SKIPPED [1] test/test_espefuse.py:1371: ESP32-C2-only, supports 1 key block
SKIPPED [1] test/test_espefuse.py:1383: ESP32-C2-only, supports 1 key block
SKIPPED [1] test/test_espefuse.py:1411: Supports 6 key blocks
SKIPPED [1] test/test_espefuse.py:1443: Supports 6 key blocks
SKIPPED [1] test/test_espefuse.py:1489: ESP32-C2-only
SKIPPED [1] test/test_espefuse.py:1508: Only chip with 6 keys
SKIPPED [1] test/test_espefuse.py:1729: For this chip, FE and SB keys go into one BLOCK
SKIPPED [1] test/test_espefuse.py:1745: For this chip, FE and SB keys go into one BLOCK
FAILED test/test_espsecure.py::TestSigning::test_sign_v2_data - ValueError: (...
FAILED test/test_espsecure.py::TestSigning::test_sign_v2_with_pre_calculated_signature
FAILED test/test_espsecure.py::TestSigning::test_verify_signature_signing_key
FAILED test/test_espsecure.py::TestSigning::test_verify_signature_public_key
FAILED test/test_espsecure.py::TestSigning::test_generate_and_extract_key_v2
===== 5 failed, 107 passed, 32 skipped, 73 deselected in 94.89s (0:01:34) ======
```
</details>