espressif/esptool #842: Won't upload program fully - stops uploading at random percentages

bheestandLMT, 2023-02-16:
### Operating System

Windows 11 Pro

### Esptool Version

4.4.0

### Python Version

3.9.10

### Chip Description

 ESP32-S3

### Device Description

DevKitC-1 v1.0
Using the USB port (not UART)

### Hardware Configuration

Nothing else connected to the dev board

### How is Esptool Run

PlatformIO Core 6.1.6 Home 3.4.3

### Full Esptool Command Line that Was Run

esptool.py --chip esp32s3 --port "COM16" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 D:\Github_Desktop\smart_config\.pio\build\esp32s3\bootloader.bin 0x8000 D:\Github_Desktop\smart_config\.pio\build\esp32s3\partitions.bin 0x10000 .pio\build\esp32s3\firmware.bin

### Esptool Output

```shell
PS D:\Github_Desktop\smart_config> python.exe "C:\Users\b123\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32s3 --port "COM16" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 D:\Github_Desktop\smart_config\.pio\build\esp32s3\bootloader.bin 0x8000 D:\Github_Desktop\smart_config\.pio\build\esp32s3\partitions.bin 0x10000 .pio\build\esp32s3\firmware.bin
esptool.py v4.4
Serial port COM16
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e2:3e:90
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x000c4fff...
Compressed 20816 bytes to 13276...
Wrote 20816 bytes (13276 compressed) at 0x00000000 in 0.3 seconds (effective 491.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 428.0 kbit/s)...
Hash of data verified.
Compressed 740672 bytes to 473378...
Writing at 0x0002008a... (10 %)Traceback (most recent call last):
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool.py", line 34, in <module>
    esptool._main()
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 1026, in _main
    main()
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 826, in main
    operation_func(esp, args)
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\cmds.py", line 556, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 109, in inner
    return func(*args, **kwargs)
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 991, in flash_defl_block
    self.check_command(
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 407, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 376, in command
    p = self.read()
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 308, in read
    return next(self._slip_reader)
StopIteration
PS D:\Github_Desktop\smart_config> python.exe "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32s3 --port "COM16" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 D:\Github_Desktop\smart_config\.pio\build\esp32s3\bootloader.bin 0x8000 D:\Github_Desktop\smart_config\.pio\build\esp32s3\partitions.bin 0x10000 .pio\build\esp32s3\firmware.bin
esptool.py v4.4
Serial port COM16
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e2:3e:90
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x000c4fff...
Compressed 20816 bytes to 13276...
Wrote 20816 bytes (13276 compressed) at 0x00000000 in 0.3 seconds (effective 506.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 484.6 kbit/s)...
Hash of data verified.
Compressed 740672 bytes to 473378...
Writing at 0x00038254... (20 %)Traceback (most recent call last):
  File "C:\Users\b123\.platformio\packages\tool-esptoolpy\esptool.py", line 34, in <module>
    esptool._main()
  File "C:\Users\b123\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 1026, in _main
    main()
  File "C:\Users\b123\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 826, in main
    operation_func(esp, args)
  File "C:\Users\b123\.platformio\packages\tool-esptoolpy\esptool\cmds.py", line 556, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\b123\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 109, in inner
    return func(*args, **kwargs)
  File "C:\Users\b123\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 991, in flash_defl_block
    self.check_command(
  File "C:\Users\b123\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 407, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\b123\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 376, in command
    p = self.read()
  File "C:\Users\b123\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 308, in read
    return next(self._slip_reader)
StopIteration
```

### More Information

I have tried different baud rates via the command line and in the Windows port settings - no difference.
 I tried a dozen different USB cables with various lengths and manufacturers to no avail.
The command will work about 1 out of 25 tries - it is random. When it works, I never have an issue with the serial monitor on the same port - it always works fine if the program is able to successfully upload. 
The percentage of upload completed is also random - stops at a different percentage almost every time (from 3% to 97%)

### Other Steps to Reproduce

This was not an issue on my windows 10 machine that died - it may not have been same esptool version though.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-02-17:
Could you please check if the issue persists with the latest master branch, i.e. 7b28699f5141eae8a612a0e0b2d6b20afc26e671? You can find information in the documentation how to install it. I think this has been addressed since the v4.4 release.

bheestandLMT, 2023-02-17:
I updated to version 4.5 and am still having the same issues.

My computer is:
HP ZBook Fury 16 G9 Mobile Workstation PC
12th Gen Intel(R) Core(TM) i9-12950HX, 2300 Mhz, 16 Core(s), 24 Logical Processor(s)
128GB RAM

Here is the log:

`<lambda>(["upload"], [".pio\build\esp32s3\firmware.bin"])
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
BeforeUpload(["upload"], [".pio\build\esp32s3\firmware.bin"])
Auto-detected: COM17
"C:\Users\bheestand\.platformio\penv\Scripts\python.exe" "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32s3 --port "COM17" --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 D:\Github_Desktop\smart_config\.pio\build\esp32s3\bootloader.bin 0x8000 D:\Github_Desktop\smart_config\.pio\build\esp32s3\partitions.bin 0x10000 .pio\build\esp32s3\firmware.bin
esptool.py v4.5
Serial port COM17
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:51:57:e0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x000c2fff...
Compressed 20576 bytes to 13099...
Writing at 0x00000000... (100 %)
Wrote 20576 bytes (13099 compressed) at 0x00000000 in 0.5 seconds (effective 349.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 337.8 kbit/s)...
Hash of data verified.
Compressed 730848 bytes to 467400...
Writing at 0x00010000... (3 %)
Writing at 0x0001ae46... (6 %)
Traceback (most recent call last):
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 1031, in _main
    main()
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 831, in main
    operation_func(esp, args)
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\cmds.py", line 580, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 131, in inner
    return func(*args, **kwargs)
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 1030, in flash_defl_block
    self.check_command(
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 435, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 404, in command
    p = self.read()
  File "C:\Users\bheestand\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 337, in read
    return next(self._slip_reader)
StopIteration

A fatal error occurred: The chip stopped responding.
*** [upload] Error 2
============================================== [FAILED] Took 12.78 seconds ==============================================`

bheestandLMT, 2023-02-18:
I was reading through some of the other responses and decided to try esptool version 4.3 and it uploaded fine.

dobairoland, 2023-02-20:
There was a regression what we have already fixed. In order to check this, you cannot use `pip` but need to install from Github. Instructions are here:

https://docs.espressif.com/projects/esptool/en/latest/esp32/contributing.html#development-setup

JunOllyLi, 2023-02-23:
I am still having the same problem, even with the dev revision of esptool:
>git log | more
commit 7b28699f5141eae8a612a0e0b2d6b20afc26e671
Author: radim.karnis <radim.karnis@espressif.com>
Date:   Wed Feb 15 15:35:35 2023 +0100

    fix(ESP32-S3): Lower CPU freq to improve flasher stub stability

    Closes https://github.com/espressif/esptool/issues/832

pip install -e .
Obtaining file:///C:/edgetx/xesp
Requirement already satisfied: bitstring>=3.1.6 in c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from esptool==4.6.dev0) (4.0.1)
Requirement already satisfied: cryptography>=2.1.4 in c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from esptool==4.6.dev0) (36.0.2)
Requirement already satisfied: ecdsa>=0.16.0 in c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from esptool==4.6.dev0) (0.18.0)
Requirement already satisfied: pyserial>=3.0 in c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from esptool==4.6.dev0) (3.5)
Requirement already satisfied: reedsolo<=1.6.0,>=1.5.3 in c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from esptool==4.6.dev0) (1.5.4)
Requirement already satisfied: cffi>=1.12 in c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from cryptography>=2.1.4->esptool==4.6.dev0) (1.15.1)
Requirement already satisfied: pycparser in c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from cffi>=1.12->cryptography>=2.1.4->esptool==4.6.dev0) (2.21)
Requirement already satisfied: six>=1.9.0 in c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from ecdsa>=0.16.0->esptool==4.6.dev0) (1.16.0)
Installing collected packages: esptool
  Attempting uninstall: esptool
    Found existing installation: esptool 4.6.dev0
    Uninstalling esptool-4.6.dev0:
      Successfully uninstalled esptool-4.6.dev0
  Running setup.py develop for esptool
Successfully installed esptool

esptool.py v4.6-dev
Serial port COM15
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:fc:65:bc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x0026dfff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20640 bytes to 13188...
Writing at 0x00000000... (100 %)
Wrote 20640 bytes (13188 compressed) at 0x00000000 in 0.4 seconds (effective 372.9 kbit/s)...
Hash of data verified.
Compressed 2480384 bytes to 1480853...
Writing at 0x00010000... (1 %)
Writing at 0x0001a16c... (2 %)
Writing at 0x00020d23... (3 %)
Writing at 0x0002a0e7... (4 %)
Traceback (most recent call last):
  File "c:\edgetx\xesp\esptool\__init__.py", line 1032, in _main
    main()
  File "c:\edgetx\xesp\esptool\__init__.py", line 832, in main
    operation_func(esp, args)
  File "c:\edgetx\xesp\esptool\cmds.py", line 580, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "c:\edgetx\xesp\esptool\loader.py", line 131, in inner
    return func(*args, **kwargs)
  File "c:\edgetx\xesp\esptool\loader.py", line 1030, in flash_defl_block
    self.check_command(
  File "c:\edgetx\xesp\esptool\loader.py", line 435, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "c:\edgetx\xesp\esptool\loader.py", line 404, in command
    p = self.read()
  File "c:\edgetx\xesp\esptool\loader.py", line 337, in read
    return next(self._slip_reader)
StopIteration

A fatal error occurred: The chip stopped responding.

This is just an ESP32-S3-WROOM module with my soldered wires. I did suspect the wires I soldered but after reading all those same problems people were having, I feel it maybe not my soldering problem. And once the flash succeeded, I never had problem with monitoring

mrdude2478, 2023-02-26:
Here you go, this should fix it for you.
[(modded)esptool.zip](https://github.com/espressif/esptool/files/10833702/modded.esptool.zip)

It has a modded stub file which has a mod for esp32-s3 clock register address.

JunOllyLi, 2023-02-27:
> Here you go, this should fix it for you. [(modded)esptool.zip](https://github.com/espressif/esptool/files/10833702/modded.esptool.zip)
> 
> It has a modded stub file which has a mod for esp32-s3 clock register address.

@mrdude2478 , thanks! This esptool.exe seems working, so far. I tried to use pip to install the latest esptool from the repo, but it did not fix the problem.
So now I have a question, how can I use the esptool.exe in the "idf.py flash"?

Thanks!

radimkarnis, 2023-02-27:
@JunOllyLi please see [this explanation](https://github.com/espressif/esptool/issues/848#issuecomment-1445906039) why that esptool.exe version works and why you probably shouldn't use it. We will release esptool v4.5.1 with a proper fix soon, it will be automatically available in your IDF installation.

radimkarnis, 2023-06-28:
I believe this should be now fully fixed in the latest esptool release. The USB-Serial/JTAG mode on the S3 is now stable.
Feel free to reopen this if issues persist. Thank you!