espressif/esptool #844: ESP32 problem

cooljackgi, 2023-02-18:
### Operating System

Mac OS (old)

### Esptool Version

esptool.py v4.5

### Python Version

2.7.10

### Chip Description

ESP32 i think

### Device Description

NodeMCU with some hardware... real time clock, tft with touch and a sd card. 

### Hardware Configuration

I played a bit with interrupts:
 timer = timerBegin(0, 80, true); // Initialize timer 0
  timerAttachInterrupt(timer, &onTimer, true); // Attach the onTimer function
  timerAlarmWrite(timer, 1000000, true); // Set the alarm to go off every 1 second
  timerAlarmEnable(timer); // Enable the alarm

void IRAM_ATTR onTimer()
{
  portENTER_CRITICAL_ISR(&timerMux);

portEXIT_CRITICAL_ISR(&timerMux);
}

and now i can't connect anymore to the esp32

### How is Esptool Run

console

### Full Esptool Command Line that Was Run

python3 esptool.py chip_id

### Esptool Output

```shell
Nothing only a 

␀␀
```

### More Information

I played a bit with interrupts:
 timer = timerBegin(0, 80, true); // Initialize timer 0
  timerAttachInterrupt(timer, &onTimer, true); // Attach the onTimer function
  timerAlarmWrite(timer, 1000000, true); // Set the alarm to go off every 1 second
  timerAlarmEnable(timer); // Enable the alarm

void IRAM_ATTR onTimer()
{
  portENTER_CRITICAL_ISR(&timerMux);

portEXIT_CRITICAL_ISR(&timerMux);
}

and now i can't connect anymore to the esp32

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-02-18:
> python3 esptool.py chip_id

Are you in the cloned github repository?

If you installed esptool with pip then this is the  way how you run it:

```python
python3 -m esptool chip_id
```

dobairoland, 2023-02-27:
This doesn't seem to be an esptool issue. I'm closing for inactivity.

closed 2023-02-27 by dobairoland