espressif/esptool #847: Erasing flash fails every time

evenastar, 2023-02-24:
### Operating System

macOS 12.3.1 

### Esptool Version

v5.0.1

### Python Version

Python 3.8.9

### Chip Description

ESP32S3R8

### Device Description

This is a custom made PCB. There is just the esp32s3r8 (sample ), flash, crystal and the needed capacitance placed on the pcb (also pull-up on the EN which is stable). Power supply is sufficient and there are no voltage drops on the supply (checked with oscilloscope). I tried different flash ICs: 
WINBOND W25N512GVEIG, MICRON MT29F2G01ABBGDWB-IT:G TR
both for 3.3V SPI Voltage. I tried all possible flash configurations in menuconfig. I tried programming via USB and directly via uart as well. Failure is the same in both cases. I replaced the ESP and flash ic several times and checked the connection to the flash IC on every pin several times. The Esp32s3r8's are engineering samples from digi-key. At the moment i am just trying to program the unmodified hello_world example. 
[image]
[image]
[image]

### Hardware Configuration

no not yet.

### How is Esptool Run

No IDE

### Full Esptool Command Line that Was Run

esptool.py --trace erase_region  4096 4096

### Esptool Output

```shell
esptool.py v4.5
Found 7 serial ports
Serial port /dev/cu.usbserial-144201
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.103 No serial data received.
.TRACE +0.055 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.013 Read 1 bytes: c0
TRACE +0.000 Read 100 bytes: 
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0c0010804000707 | ... ............
    122000000000c0c0 0108040007071220 | . ............. 
    00000000c0c00108 0400070712200000 | ............. ..
    0000c0c001080400 0707122000000000 | ........... ....
    c0c00108                          | ....
TRACE +0.000 Received full packet: 010804000707122000000000
[... 1011 lines cut ...]
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x252c0f, SizeID=0x25), defaulting to 4MB
Flash will be erased from 0x00000000 to 0x00006fff...
Flash will be erased from 0x00010000 to 0x00041fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Erasing flash...
Flash params set to 0x0220

A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption.
CMake Error at run_serial_tool.cmake:55 (message):
  
  /Users/name/.espressif/python_env/idf5.0_py3.8_env/bin/python;;/Users/name/esp/esp-idf-v5.0.1/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed

FAILED: CMakeFiles/flash /Users/name/esp/hello_world/build/CMakeFiles/flash 
cd /Users/name/esp/esp-idf-v5.0.1/components/esptool_py && /usr/local/Cellar/cmake/3.25.2/bin/cmake -D IDF_PATH=/Users/name/esp/esp-idf-v5.0.1 -D "SERIAL_TOOL=/Users/name/.espressif/python_env/idf5.0_py3.8_env/bin/python;;/Users/name/esp/esp-idf-v5.0.1/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;--no-stub;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/name/esp/hello_world/build -P /Users/name/esp/esp-idf-v5.0.1/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/name/esp/hello_world/build/log/idf_py_stderr_output_90809 and /Users/name/esp/hello_world/build/log/idf_py_stdout_output_90809
name@Joels-MBP hello_world % `

the error is consistent every time. No matter what i change. I checked everything several times and can't find the mistake. the Flash chip id is detected but size detection fails every time. 

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-02-24:
Hi @evenastar. Could you please try the latest master branch? There is already a fix there which were not included in 4.5 and might solve your issue.

evenastar, 2023-02-24:
Hi @dobairoland  I tried the latest esptool-4.6.dev0, esp-idf is v5.0.1 
the error is the same: 
```                                               
                                               
Executing action: all (aliases: build)
Running ninja in directory /Users/joelname/esp/hello_world/build
Executing "ninja all"...
[1/4] cd /Users/joelname/esp/hello_world/build/esp-idf/esptool_py && /Users/joelname/.espressif/python_env/idf5.0_py3.8_env/bi...Users/joelname/esp/hello_world/build/partition_table/partition-table.bin /Users/joelname/esp/hello_world/build/hello_world.binhello_world.bin binary size 0x31580 bytes. Smallest app partition is 0x100000 bytes. 0xcea80 bytes (81%) free.
[1/1] cd /Users/joelname/esp/hello_world/build/bootloader/esp-idf/esptool_py && /Users/joelname/.espressif/python_env/idf5.0_p...ents/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/joelname/esp/hello_world/build/bootloader/bootloader.binBootloader binary size 0x6d00 bytes. 0x1300 bytes (15%) free.
[4/4] Completed 'bootloader'ootloader'
Project build complete. To flash, run this command:
/Users/joelname/.espressif/python_env/idf5.0_py3.8_env/bin/python ../esp-idf-v5.0.1/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32s3 --no-stub write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/hello_world.bin
or run 'idf.py -p (PORT) flash'
joelname@Joels-MBP hello_world % idf.py flash                                                    
Executing action: flash
Serial port /dev/cu.usbmodem14301
Connecting...
Detecting chip type... ESP32-S3
Running ninja in directory /Users/joelname/esp/hello_world/build
Executing "ninja flash"...
[1/5] cd /Users/joelname/esp/hello_world/build/esp-idf/esptool_py && /Users/joelname/.espressif/python_env/idf5.0_py3.8_env/bin/python /Users/joelname/esp/esp-idf-v5.0.1/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /Users/joelname/esp/hello_world/build/partition_table/partition-table.bin /Users/joelname/esp/hello_world/build/hello_world.bin
hello_world.bin binary size 0x31580 bytes. Smallest app partition is 0x100000 bytes. 0xcea80 bytes (81%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /Users/joelname/esp/hello_world/build/bootloader/esp-idf/esptool_py && /Users/joelname/.espressif/python_env/idf5.0_py3.8_env/bin/python /Users/joelname/esp/esp-idf-v5.0.1/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/joelname/esp/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x6d00 bytes. 0x1300 bytes (15%) free.
[3/5] No install step for 'bootloader'
[4/5] Completed 'bootloader'
[4/5] cd /Users/joelname/esp/esp-idf-v5.0.1/components/esptool_py && /usr/local/Cellar/cmake/3.25.2/bin/cmake -D IDF_PATH=/Users/joelname/esp/esp-idf-v5.0.1 -D "SERIAL_TOOL=/Users/joelname/.espressif/python_env/idf5.0_py3.8_env/bin/python;;/Users/joelname/esp/esp-idf-v5.0.1/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;--no-stub;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/joelname/esp/hello_world/build -P /Users/joelname/esp/esp-idf-v5.0.1/components/esptool_py/run_serial_tool.cmake
esptool esp32s3 -p /dev/cu.usbmodem14301 -b 460800 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.6-dev
Serial port /dev/cu.usbmodem14301
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:ae:00:bc
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x252c0f, SizeID=0x25), defaulting to 4MB
Flash will be erased from 0x00000000 to 0x00006fff...
Flash will be erased from 0x00010000 to 0x00041fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Erasing flash...
Flash params set to 0x0220

A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption.
CMake Error at run_serial_tool.cmake:55 (message):
  
  /Users/joelname/.espressif/python_env/idf5.0_py3.8_env/bin/python;;/Users/joelname/esp/esp-idf-v5.0.1/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed

FAILED: CMakeFiles/flash /Users/joelname/esp/hello_world/build/CMakeFiles/flash 
cd /Users/joelname/esp/esp-idf-v5.0.1/components/esptool_py && /usr/local/Cellar/cmake/3.25.2/bin/cmake -D IDF_PATH=/Users/joelname/esp/esp-idf-v5.0.1 -D "SERIAL_TOOL=/Users/joelname/.espressif/python_env/idf5.0_py3.8_env/bin/python;;/Users/joelname/esp/esp-idf-v5.0.1/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;--no-stub;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/joelname/esp/hello_world/build -P /Users/joelname/esp/esp-idf-v5.0.1/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/joelname/esp/hello_world/build/log/idf_py_stderr_output_95823 and /Users/joelname/esp/hello_world/build/log/idf_py_stdout_output_95823
```

radimkarnis, 2023-02-24:
Hello @evenastar,

this is most likely a HW issue - esptool fails at the exact moment whenever the ESP starts talking to the SPI flash chip.
The `Warning: Could not auto-detect Flash size (FlashID=0x252c0f, SizeID=0x25), defaulting to 4MB` message is also suspicious. Could you please run the `flash_id` command and post the output here?

Also, looking at the photo, there seem to be blobs of solder and flux all around the board. I suggest you check for any bridges.

evenastar, 2023-02-24:
hello @radimkarnis, 

i checked the pins for bridges and connections and there are no bridges and every Flash connection is fine. I know the PCB doesn't look nice, but that was just a temporary condition right after a component change. 
We also tried on windows w the same result. I too think its an HW issue but I am failing to find the reason for it.

Today I tried yet another Flash-IC 1Gbit Winbond NAND-Flash (W25N01GVZEIG TR)

Here is  flash_id, followed by --trace: 

```
esptool.py v4.6-dev
Serial port /dev/cu.usbmodem14301
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:ae:00:bc
Uploading stub...
Running stub...
Stub running...
Manufacturer: 03
Device: efaa
Detected flash size: Unknown
Flash type set in eFuse: quad (4 data lines)
Hard resetting via RTS pin...
joelname@Joels-MBP hello_world % 
```

esptool.py --trace --port /dev/cu.usbmodem14301 flash_id 
```
esptool.py --trace --port /dev/cu.usbmodem14301 flash_id 
esptool.py v4.6-dev
Serial port /dev/cu.usbmodem14301
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 547 bytes: 
    66660d0a696e7661 6c69642068656164 | ff..invalid head
    65723a2030786666 6666666666660d0a | er: 0xffffffff..
    696e76616c696420 6865616465723a20 | invalid header: 
    3078666666666666 66660d0a696e7661 | 0xffffffff..inva
    6c69642068656164 65723a2030786666 | lid header: 0xff
    6666666666660d0a 696e76616c696420 | ffffff..invalid 
    6865616465723a20 3078666666666666 | header: 0xffffff
    66660d0a696e7661 6c69642068656164 | ff..invalid head
    65723a2030786666 6666666666660d0a | er: 0xffffffff..
    696e76616c696420 6865616465723a20 | invalid header: 
    3078666666666666 66660d0a696e7661 | 0xffffffff..inva
    6c69642068656164 65723a2030786666 | lid header: 0xff
    6666666666660d0a 696e76616c696420 | ffffff..invalid 
    6865616465723a20 3078666666666666 | header: 0xffffff
    66660d0a696e7661 6c69642068656164 | ff..invalid head
    65723a2030786666 6666666666660d0a | er: 0xffffffff..
    696e76616c696420 6865616465723a20 | invalid header: 
    3078666666666666 66660d0a696e7661 | 0xffffffff..inva
    6c69642068656164 65723a2030786666 | lid header: 0xff
    6666666666660d0a 696e76616c696420 | ffffff..invalid 
    6865616465723a20 3078666666666666 | header: 0xffffff
    66660d0a696e7661 6c69642068656164 | ff..invalid head
    65723a2030786666 6666666666660d0a | er: 0xffffffff..
    696e76616c696420 6865616465723a20 | invalid header: 
    3078666666666666 66660d0a4553502d | 0xffffffff..ESP-
    524f4d3a65737033 3273332d32303231 | ROM:esp32s3-2021
    303332370d0a4275 696c643a4d617220 | 0327..Build:Mar 
[... 1107 lines cut ...]
TRACE +0.000 Write 26 bytes: 
    c000091000000000 0020200060000000 | .........  .`...
    70ffffffff000000 00c0             | p.........
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 2 bytes: 0109
TRACE +0.000 Read 1 bytes: 02
TRACE +0.000 Read 4 bytes: 00000000
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 3 bytes: 0000c0
TRACE +0.000 Received full packet: 01090200000000000000
Manufacturer: 03
Device: efaa
Detected flash size: Unknown
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=3c700060
TRACE +0.000 Write 14 bytes: c0000a0400000000003c700060c0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 1 bytes: 01
TRACE +0.000 Read 2 bytes: 0a02
TRACE +0.000 Read 2 bytes: 0000
TRACE +0.000 Read 1 bytes: 01
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 4 bytes: 000000c0
TRACE +0.000 Received full packet: 010a0200000100000000
Flash type set in eFuse: quad (4 data lines)
```

radimkarnis, 2023-02-24:
The reported Device and Manufacturer [are unknown](https://github.com/flashrom/flashrom/blob/master/include/flashchips.h), I am not sure if the flash chip you use is supported.

I am sorry, but I am not sure what the issue is and I do not know what else to recommend to try. I can suggest you to use the free-of-charge reviewing services for circuit diagrams and PCB designs of products powered by Espressif's chips and modules at https://www.espressif.com/en/products/services.

evenastar, 2023-02-24:
ok @radimkarnis, thank you for your help. 
Do you think that unsupported flash could be the only reason for this error?

radimkarnis, 2023-02-24:
> Do you think that unsupported flash could be the only reason for this error?

Certainly not, there might be other reasons. But judging from the suspicious warnings about undetectable flash and from the fact, that only operations involving the SPI flash fail, this is most likely an issue with the flash chip/the devkit design.

One more thing to try is downgrading esptool to v4.2.1. This way we can definitely rule out any regressions on the software side of esptool.

evenastar, 2023-02-24:
I just took a flash ic from an wroom board and with that the flash and flashing is working. So the unsupported flash was the reason for the error. Thank you for your help @radimkarnis !

radimkarnis, 2023-02-27:
Great! I am glad it works!

closed 2023-02-27 by radimkarnis