espressif/esptool #848: CPU commit 7b28699 didn't fix the read/write issues with ESP32-S3

mrdude2478, 2023-02-25:
### Operating System

Windows 10 64Bit

### Esptool Version

Commit: 7b28699f5141eae8a612a0e0b2d6b20afc26e671

### Python Version

3.11

### Full Esptool Command Line that Was Run

esptool -b 921600 --port COM3 --chip esp32s3 read_flash 0xCA2000 0x148000 FAT9.bin

### Esptool Output

```shell
No issues with the output.
```

### What is the Expected Behaviour?

Please revert commit:
https://github.com/espressif/esptool/commit/7b28699f5141eae8a612a0e0b2d6b20afc26e671

After further investigation I was able to fix this issue, please see here at the bottom of the page:
https://github.com/espressif/esptool/issues/832

There was an issue with reading/writing esp32-s3 chip with newer builds of esptool. It was disgused in issue 832. After further inversigation and installing the sdk to build the stub files, I modded the files and found what was making the read/write fails. Please see the bottom of issue 832 (which is closed), for the fix. CPU speed can be put back to 240Mhz.

Thanks

mrdude2478, 2023-02-27:
OK thanks, I do know that since changing this value, I have not experienced a failure yet when reading or writing, so that will so yes you are probably correct, removing this value seems like a viable fix. I guess the first time when the cpu frequency was changed I might have been lucky, however I still got crashes randomly, yesterday I flashed my dongle about 50 times also dumped the entire flash maybe about 20 times to test and didn't get a failure once.

mrdude2478, 2023-02-27:
@radimkarnis 

What about doing something like th-s - in stubflasher.c add these two lines.

```
#if USE_MAX_CPU_FREQ
static bool can_use_max_cpu_freq()
{
  /* Check if any of available USB modes are being used. */
  #if ESP32S3 //added this
  return false; //added this
  #elif WITH_USB_OTG && !WITH_USB_JTAG_SERIAL
  return stub_uses_usb_otg();
  #elif !WITH_USB_OTG && WITH_USB_JTAG_SERIAL
  return stub_uses_usb_jtag_serial();
  #elif WITH_USB_OTG && WITH_USB_JTAG_SERIAL
  return stub_uses_usb_otg() || stub_uses_usb_jtag_serial();
  #else
  return false;
  #endif
}

```

Then if the chip was detected as esp32-s3, it would disable the max cpu frequency. Would that work?

radimkarnis, 2023-02-27:
@mrdude2478 it would work, but if this feature is unstable on the S3, we don't want to risk these issues also on other chips. So it will be reverted for all the chips, until we figure out why it doesn't work for everyone and how to implement it in a safe way.

mrdude2478, 2023-02-27:
OK, thank - you could then just return false from that function then and it would do the same thing for all chips, and would save needing to re-write lots of code until it's figured out why there's an timing issue? Maybe it's down to the xtal in my board? I don't know much about electronics but surely if that deals with the timing it could be something related to that? I ordered another board to check this, I might try replacing that crystal with a different brand to see if it makes any differnce.

JunOllyLi, 2023-02-27:
What I have observed is that the official 4.5 and the latest code (4.6-dev) in the repo behave the same, that sometimes it works most of the time, while sometimes it just does not work at all. And mrdude2478's exe works while the 4.6-dev refuses to work.

I still have not figured out exactly what is the difference between those two scenarios. I am using the same computer (Surface 8 Pro, with nothing else connected, except the charger. I have tried with both charger connected and not connected, did not see any difference), same USB cable, same boards (yes I have different boards, some powered by on-board 5V=>3.V, some powered by external buck converter, and all behave the same). All I can tell is that it normally works while I sit in front of my desk, and not work while I sit in the bed, but I don't see how that would affect the result :(

JunOllyLi, 2023-02-27:
Reading this thread more and here are my thoughts:
So mrdude2478 was writing something intended for SYSTEM_BASE_REG + 0x060 to just SYSTEM_BASE_REG and luckily it worked. Now:
1. What is the value of that?
2. What that value mean to the register at SYSTEM_BASE_REG?
3. Would it still work if we totally just skip the write (neither writing to SYSTEM_BASE_REG + 0x060, nor SYSTEM_BASE_REG)

So there are three possibilities:
a. the value written to SYSTEM_BASE_REG luckily had some bits that affects the behavior and make it more stable
b. the value originally written to SYSTEM_BASE_REG + 0x060 made it fail to work, and we should just ignore it
c. combination of both a and b.

I think it would be a good idea to understand what situation we are in first.

mrdude2478, 2023-02-27:
@JunOllyLi , @radimkarnis allready explained that it's down to the CPU frequencey using the max speed of the chip (see the code above). This is in the stub that is uploaded to the chip before dumping/flashing routines kick in. It's being looked into by people far smarter than me, well at least with far more experience in chip programming and writing software. At least those people are aware of the issues and why there's an issue so I imagine that it will be fixed in the near future as has already been stated. We just meed to wait now and be patient and thank the coders for the hard work they do and for looking into this issue.

JunOllyLi, 2023-02-28:
> @JunOllyLi , @radimkarnis allready explained that it's down to the CPU frequency using the max speed of the chip (see the code above). This is in the stub that is uploaded to the chip before dumping/flashing routines kick in. It's being looked into by people far smarter than me, well at least with far more experience in chip programming and writing software. At least those people are aware of the issues and why there's an issue so I imagine that it will be fixed in the near future as has already been stated. We just meed to wait now and be patient and thank the coders for the hard work they do and for looking into this issue.

But I thought the frequency change had already been in the lates of the repo? And it did not work? At least not working after doing the "pip install"

radimkarnis, 2023-02-28:
Thank you both for your interest and for helping with this issue. 
Good news! We have possibly identified the underlying reason why this keeps happening on the S3. A fix is in the works. If that works, we don't have to revert. Will keep you guys updated.

mrdude2478, 2023-02-28:
@radimkarnis

Thank you for the updated news, that's great that you guys managed to find the issue so quickly. Well done and thank you for looking into this, it's very much appreciated.

radimkarnis, 2023-02-28:
The plan is to temporarily disable this feature and make a stable v4.5.1 release. Then we can properly implement a fix, re-enable it, and test it thoroughly for v4.6.

JunOllyLi, 2023-02-28:
@radimkarnis Thanks for the hard work!

So when can we expect the change (to make it stable in v4.5.1) in the repo and try it out?

radimkarnis, 2023-02-28:
@JunOllyLi it is already in master, you can clone the repo and install esptool ([instructions here](https://docs.espressif.com/projects/esptool/en/latest/contributing.html#development-setup)) to try! It will be released on PyPI tomorrow.

JunOllyLi, 2023-03-01:
it worked pretty well last night. Thanks!

radimkarnis, 2023-03-01:
@JunOllyLi thanks for the confirmation! I am now working on the proper fix, I will ask you to kindly try it when it's ready.

mrdude2478, 2023-03-01:
@radimkarnis

I'll try it out as well once it's done. Thanks.

radimkarnis, 2023-03-21:
@JunOllyLi @mrdude2478 I have created [a development release](https://pypi.org/project/esptool/4.6.dev0/) with a possible fix. It can be installed with `pip install esptool==4.6.dev0`.

Could you please try if that solves the issue for you? Thank you!

JunOllyLi, 2023-03-22:
I tried it several times, so far it works well

mrdude2478, 2023-03-24:
@radimkarnis 

Sorry for the late replay as I just noticed you'd replied. I updated and tried but still get issues.

```
C:\Users\MrDude\Desktop>esptool -b 921600 --port COM1 --chip esp32s3 read_flash 0x0 0x1000000 FAT.bin
esptool.py v4.6.dev0
Serial port COM1
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:44:a5:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
532480 (3 %)
A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0xfb2 bytes

C:\Users\MrDude\Desktop>esptool -b 921600 --port COM1 --chip esp32s3 read_flash 0x0 0x1000000 FAT.bin
esptool.py v4.6.dev0
Serial port COM1
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:44:a5:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
3186688 (18 %)
A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0xf80 bytes
```

Out of two attempts it failed on both.

mrdude2478, 2023-03-28:
@radimkarnis

To be honest, I didn't notice any speed difference with the increase in cpu frequency whenn dumping or flashing. It was pretty much the same speed with the CPU frequency disabled. I am just happy that it's now able to dump and flash properly with 4.5.1  as stability is more important than saving a few seconds. A full 16MB dump takes about 1/2 hour, flashing is much faster though and just takes a couple of minutes.

radimkarnis, 2023-06-27:
Closing this as increasing CPU frequency won't be implemented for the ESP32-S3. If anyone feels like this should be reconsidered, feel free to reopen.

closed 2023-06-27 by radimkarnis