espressif/esptool #849: esptool.exe quarantained by Avira virsus scanner

xinkiknix, 2023-02-27:
### Operating System

Windows 11

### Esptool Version

version 4.5

### Python Version

Python 3.10.10

### Chip Description

ESP32-s3, ESP32

### Device Description

None

### Hardware Configuration

Nope

### How is Esptool Run

Arduino 2.0.3

### Full Esptool Command Line that Was Run

ZA

### Esptool Output

```shell
NA
```

### More Information

Updated Espressif boards using boardmanager

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-03-01:
Hi @xinkiknix,
to sum up - you've installed ESP32 core in Arduino and your antivirus marked the bundled esptool.exe as a virus?

Can you please try downloading the latest windows executable from [here](https://github.com/espressif/esptool/releases/tag/v4.5.1) and see if your antivirus allows it? 

We provide a digital signature for the windows binary, but that might not be enough for some antivirus SW.

xinkiknix, 2023-03-04:
> <html><body>

The analysis you requested is now complete:
--
FileResultesptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/LICENSECleanesptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/README.mdCleanesptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/esp_rfc2217_server.exeCleanesptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/espefuse.exeCleanesptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/espsecure.exeCle  anesptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/esptool.exeClean | File | Result | esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/LICENSE | Clean | esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/README.md | Clean | esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/esp_rfc2217_server.exe | Clean | esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/espefuse.exe | Clean | esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/espsecure.exe | Cle  an | esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/esptool.exe | Clean
File | Result
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/LICENSE | Clean
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/README.md | Clean
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/esp_rfc2217_server.exe | Clean
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/espefuse.exe | Clean
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/espsecure.exe | Cle  an
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/esptool.exe | Clean
For more information regarding our detections please visit                         the Avira Virus Lab page.
Stay safe,                         Your Avira Virus Lab team

</body>
</html>The analysis you requested is now complete:
File	Result
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/LICENSE	Clean
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/README.md	Clean
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/esp_rfc2217_server.exe	Clean
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/espefuse.exe	Clean
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/espsecure.exe	Cle an
esptool-v4.5.1-win64.zip/esptool-v4.5.1-win64/esptool.exe	Clean
For more information regarding our detections please visit the [Avira Virus Lab](http://www.avira.com/en/support-virus-lab) page.
Stay safe,
Your Avira Virus Lab team

radimkarnis, 2023-03-06:
@xinkiknix thank you for the report, the scan shows it is actually clean.

In this case, the only thing you can do is mark `esptool.exe` as safe manually.

I am sorry, but there's nothing else we can do here:
1) The scan is clean.
2) We sign the binary.
3) This false positive doesn't happen with other AV software.

I am closing this ticket, as it's really not an issue with esptool.

closed 2023-03-06 by radimkarnis