espressif/esptool #850: espefuse.py incompatible with bitstring v4.x

chrisPainePodPoint, 2023-03-01:
### Operating System

ubuntu 22.04

### Esptool Version

3.2-dev

### Python Version

3.10.6

### Full Esptool Command Line that Was Run

espefuse.py -h

### Esptool Output

```shell
chris@DESKTOP-H6UHTOK:/mnt/c/Users/ChristopherPaine/Documents/code/arch2.3_wifi_fw$ espefuse.py -h
Traceback (most recent call last):
  File "/home/chris/esp/esp-idf/components/esptool_py/esptool/espefuse.py", line 25, in <module>
    import espressif.efuse.esp32 as esp32_efuse
  File "/home/chris/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/esp32/__init__.py", line 1, in <module>
    from . import operations
  File "/home/chris/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/esp32/operations.py", line 26, in <module>
    from . import fields
  File "/home/chris/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/esp32/fields.py", line 26, in <module>
    from .. import base_fields
  File "/home/chris/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/base_fields.py", line 24, in <module>
    from bitstring import BitArray, BitString
ImportError: cannot import name 'BitString' from 'bitstring' (/home/chris/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/bitstring.py)
chris@DESKTOP-H6UHTOK:/mnt/c/Users/ChristopherPaine/Documents/code/arch2.3_wifi_fw$
```

### What is the Expected Behaviour?

Expected to print the help.

To fix I ran `pip install bitstring==3.1.9` and then `espefuse.py -h` successfully prints the help.

### More Information

Not much verification but tried on two installations. One within a docker that I can reliably recreate and the other in the WSL. Both running ubuntu 22.04.

esptools have been installed using pip. `pip install esptool`.

On windows 10 in the ESP-IDF 4.4 CMD terminal this works fine, but running `pip list` in this context shows that the installer does by default install bistring==3.1.9. Seems to be that under ubuntu pip will install a newer version of bitstring..?

dobairoland, 2023-03-01:
@chrisPainePodPoint What is your version of ESP-IDF? This has been fixed in the latest bugfix release, v4.4.4. 

> esptools have been installed using pip. pip install esptool.

You should not do that. If you are in an ESP-IDF (`<v5.0`) environment then compatible esptool scripts are present. If you install the latest esptool then that one is compatible with ESP-IDF v5.0 and older.

chrisPainePodPoint, 2023-03-01:
So the ESP-IDF environment is v4.4 rather than v4.4.4. I will update to that and then avoid using `pip install esptool`. Thanks for help!

closed 2023-03-01 by chrisPainePodPoint