espressif/esptool #851: ESP32-S3  esptool 4.51 won't upload SPIFFS

wittend, 2023-03-03:
### Operating System

Linux Mint 21 (Ubuntu)

### Esptool Version

v4.5.1

### Python Version

3.10.6

### Chip Description

ESP32-S3 

### Device Description

Adafruit Feather ESP32-S3 2MB PSRAM, others.

### Hardware Configuration

No

### How is Esptool Run

Various.  Arduino IDE 1.8.19,  PlatformIO plugin

### Esptool Output

```shell
SPIFFS] data   : /home/dave/Projects/magnetometer/mag-dev/ESP32-RM3100I2CWeb/data
[SPIFFS] start  : 2686976
[SPIFFS] size   : 1408
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/favicon.png
/index.html
/filesystem.html
/magmon.html
/credentials.h
/config.html
/console-log-html.min.js
[SPIFFS] upload : /tmp/arduino_build_501538/ESP32-RM3100I2CWeb.spiffs.bin
[SPIFFS] address: 2686976
[SPIFFS] port   : /dev/ttyACM0
[SPIFFS] speed  : 921600
[SPIFFS] mode   : dio
[SPIFFS] freq   : 80m

esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting...

A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?
SPIFFS Upload failed!
```

### More Information

This is an ESP32-S3 not ESP32!

### Other Steps to Reproduce

Everything Works well with ESP32 and ESP32-s2

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

Jason2866, 2023-03-03:
This `A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?
SPIFFS Upload failed!` is an indicator that the generated flash command is wrong.
esptool.py is instructed to flash an ESP32 not an ESP32-S3
Most probably not an issue in esptool.py

wittend, 2023-03-03:
Possibly, but who to blame.  I cannot see (or know) the command line generated.  I have NO Idea where the SPIFFS data gets put, and I have no idea where this comes from.

Jason2866, 2023-03-03:
Without infos provided this will lead to nowhere.

wittend, 2023-03-03:
What info[s] are you talking about?

Jason2866, 2023-03-04:
Different question what do you think can be done with your report?
You need to provide an easy example to reproduce the issue. Without nothing can be done.

dobairoland, 2023-03-06:
> A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument

This error message is usual in situations when a wrong chip target is selected in the build system (ESP32 in this case), therefore, the images are build for the wrong chip target and when these passed to esptool with `--chip esp32`, it detects that the connected chip is something different (ESP32-S3 in this case).

This is not an esptool issue.

closed 2023-03-06 by dobairoland