espressif/esptool #855: esptool write_flash: error: the following arguments are required: <address> <filename>

duwardo185, 2023-03-08:
### Operating System

Windows 11

### Esptool Version

4.5.1

### Python Version

3.10

### Chip Description

ESP32-D0WD (revision v1.0)

### Device Description

Not sure about the board or other hardware. I bought my chip on Antons mindstorms hack. The information below is from the chip itself:

Found 1 serial ports
Serial port COM3
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...     
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:11:ae:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4017
Detected flash size: 8MB
Hard resetting via RTS pinAnton's

### Hardware Configuration

Please see above

### How is Esptool Run

VS code on a Windows computer

### Full Esptool Command Line that Was Run

PS C:\Users\duwar\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts> .\esptool.py -b 115200 -–before default_reset –-after hard_reset -–chip esp32 write_flash –flash_mode dio –flash_freq 40m –flash_size detect 0x10000 build/app-template.bin 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin   

### Esptool Output

```shell
trace doesn't provide any output
```

### More Information

I tried searching your TS guide and similar issues.  how do I get the <address> and <filename>?

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-03-09:
Hi @duwardo185.
I am sorry, but I don't understand your issue fully.

`write_flash: error: the following arguments are required: <address> <filename>` error means that the esptool command you run is wrong - there's a typo or an argument at the wrong place.

Let's assume this happens when you run the command you've provided. Taking a closer look, the command has some wrong dashes (e.g. `-` instead of `--`). 

Please try running this corrected cmd: `esptool.py -b 115200 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 build/app-template.bin 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin`

duwardo185, 2023-03-09:
I appreciate your input. Also, I'm not a developer or coder so please bear with me. I tried your command at I got the same output. 

PS C:\Users\duwar> python -m esptool -b 115200 --before default_reset --after hard_reset --chip esp32 write_flash --flash_freq 40m --flash_size detect 0x10000 build/app-template.bin 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]      
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION]
                           [--no-progress] [--verify] [--encrypt]       
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting]    
                           [--force] [--compress | --no-compress]       
                           <address> <filename>
                           [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'build/app-template.bin'

radimkarnis, 2023-03-09:
> I tried your command, I got the same output.

But that is not the same output. Now it's telling you a different thing - that the file `build/app-template.bin` doesn't exist (it probably does, but in a different location than you are pointing to).

That is probably because you are running esptool in your user directory `C:\Users\duwar`. You need to be in your project directory, so the `build/app-template.bin` relative location actually points to that file.

duwardo185, 2023-03-09:
Okay let me try that

duwardo185, 2023-03-09:
Success. Thank you!

closed 2023-03-09 by duwardo185

nasim41, 2024-01-20:
I built a win form c# application that it has command line and I ran command of esptool in this , after that I faced with many issues that I mention them below , please help me , this task has high priority for our team and I struggle with it for 2 weeks. :
1. at the first , after running command of esptool in my win form , it return esptool command to me , it operates like echo . 
2. in output of my win form it show some errors that it says , esptool does not have arguments but in application does not show this, just echo my command. 
3. but when I ran this command in my command line of my pc (manually) , it worked. and it shows everything correctly. 
please see screens. 
thanks. 
[image]
[image]
[image]
[image]
[image]