espressif/esptool #859: v4.5.1 esptool.py won't write flash above 16M on a 32MB ESP32S3 module

chewhs00, 2023-03-14:
### Operating System

Ubuntu 20.04.5 LTS

### Esptool Version

v4.5.1

### Python Version

3.8.10

### Chip Description

ESP32-S3 (revision v0.1), 32MB Octal Flash, 80MHz,  opi_dtr

### Device Description

Module on a custom board.

### How is Esptool Run

Terminal, esptool repo

### Full Esptool Command Line that Was Run

~/esp/esptool/esptool.py --chip esp32s3 --p /dev/ttyUSB0 --no-stub write_flash --flash_size 32MB -z 0x1C10000 dummy.bin

### Esptool Output

```shell
****** Power the unit in Bootloading mode ******

$ ~/esp/esptool/esptool.py -p /dev/ttyUSB0 --chip esp32s3 --no-stub erase_flash
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e4:23:7c
Enabling default SPI flash mode...
Erasing flash (this may take a while)...

A fatal error occurred: ESP32-S3 ROM does not support function erase_flash.

$ ~/esp/esptool/esptool.py -p /dev/ttyUSB0 --chip esp32s3 erase_flash
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e4:23:7c
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 76.2s
Hard resetting via RTS pin...

$ ~/esp/esptool/esptool.py -p /dev/ttyUSB0 --chip esp32s3 --no-stub read_flash 0xc10000 $((256*1024)) read_c10000.bin
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e4:23:7c
Enabling default SPI flash mode...
[... 348 lines cut ...]

$ ~/esp/esptool/esptool.py -p /dev/ttyUSB0 --chip esp32s3 --no-stub read_flash 0x1c10000 $((256*1024)) read5_1c10000.bin
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e4:23:7c
Enabling default SPI flash mode...

A fatal error occurred: Failed to read flash block (result was 01090000: CRC or checksum was invalid)
```

### More Information

My goal is to flash a 256kB binary image at a location above 16MB boundary, in this case at 0x1C10000. To check if it is successfully written to the proper location, I read it back with/without stub, at both locations 0xC10000 and 0x1C10000. With the above sequence of read/write, I still don't know if I wrote to 0x1C10000 correctly or I have accidentally corrupted 0xC10000. This --no-stub option sometimes fails for unknown reason! 

### Other Steps to Reproduce

See above Esptool Output

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-03-15:
Do you have any particular reason to use `--no-stub`? You are right that it is not working but I don't know what we could do about it. The ROM code is invoked and not everything is initialized/implemented there for OPI flash support. 

If you don't use `--no-stub` then the flasher stub of esptool is used where this is implemented and should work.

chewhs00, 2023-03-15:
@dobairoland , I thought stub only works when accessing within 16MB. Is that no longer the case?

dobairoland, 2023-03-15:
Support was added in v4.5.

chewhs00, 2023-03-15:
Any release schedule this get included in idf 4.4 and idf 5.0?

dobairoland, 2023-03-15:
It is already included with ESP-IDF v5.0+. There is no schedule now if or when will be included with v4.4.x.

closed 2023-03-15 by dobairoland