espressif/esptool #860: Max block size for ESP32-c3 with FLASH_DATA command?

bmkuter, 2023-03-16:
### Operating System

Windows 10

### Esptool Version

NA

### Python Version

NA

### Chip Description

ESP32-C3 Mini

### Device Description

ESP32-c3 Mini 

### Hardware Configuration

NXP K70 attached via UART0. 

### How is Esptool Run

NA

### Full Esptool Command Line that Was Run

NA

### Esptool Output

```shell
NA
```

### More Information

I'm manually calling the serial protocol commands described [here ](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/serial-protocol.html#spi-set-parameters) to update my ESP32-c3 coprocessor's firmware from my product's main CPU. Everything works as expected from the ESP32-c3's serial bootloader when writing maximum blocks of 1024 bytes during my FLASH_BEGIN and FLASH_DATA sections. 

However when I up the block size to 2048+, FLASH_DATA reports back an error code (0x5), indicating the “Received message is invalid”. I was wondering if for the ESP32-c3, there is a fixed block size of 1024 bytes per FLASH_DATA call within the ESP32-C3 ROM loader. 

For reference, the command sequence I am sending, before encoding everything into the SLIP packet, is:
(Command Packet 0 - 7) 
0x00 0x03 0x10 0x10 0x8E 0x00 0x00 0x00
(Data Packet, data size = 4096, sequence number = 0)
0x00001000 0x00000000 0x00000000 0x00000000

The same command packet structure works as expected with a data size = 1024, and the data size is mirrored in the leading FLASH_BEGIN command.

### Other Steps to Reproduce

NA

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-03-20:
Hello @bmkuter,

> I was wondering if for the ESP32-c3, there is a fixed block size of 1024 bytes per FLASH_DATA call within the ESP32-C3 ROM loader.

That is right - the ROM bootloader in Espressif chips supports blocks of max 1024 bytes (`FLASH_WRITE_SIZE = 0x400`).
If the stub flasher is used, the block size can be increased up to 16384 bytes (`FLASH_WRITE_SIZE = 0x4000`).

closed 2023-03-20 by dobairoland

bmkuter, 2023-03-20:
Thanks! Can you point me towards any documentation on implementing my own version of the stub flasher? Is there a standard precompiled stub flasher binary one can use to load into IRAM via the MEM_X commands?

bmkuter, 2023-03-20:
Thanks again! For the precompiled JSON, can you provide any clarity on the `data` section versus the `text` section? And do you know if a `MEM_END` command is only expected after writing both sections?

radimkarnis, 2023-03-20:
@bmkuter That's how the toolchain compiles the stub, I didn't really have the chance to study the compiled contents of the `data` and `text` sections. Both of these have to be written before `MEM_END` (`self.mem_finish(stub.entry)`) is called.

bmkuter, 2023-03-20:
Thanks for the insight! 
Know if there are any differences in calculating the check sum between the ROM loader and stub loader? My same function that had worked for calculating the checksum for the ROM loader is returning error 0xC1, ESP_BAD_DATA_CHECKSUM, when using the stub loader and I can't seem to isolate the issue.