espressif/esptool #861: License and sources for binaries in test/

paravoid, 2023-03-18:
### Is your feature request related to a problem?

The "test" directory includes quite a few binaries without their sources: ELF binaries under "elf2image", and images under "images".

There is no LICENSE attached in this directory, and as such, the parent (esptool) license applies, which is the GNU GPL v2.

The GNU GPL mandates that the source code ("preferred form of the work for making modifications to it") should be shipped alongside the binaries. This means that shipping these binaries without sources violates the GPL, and these files are AFAIK (IANAL etc.) undistributable. Even shipping the project/tarballs through GitHub is potentially problematic. (Espressif as the copyright holder has all rights and can ship whatever they want, but redistribution e.g. through a simple fork is when it all breaks down).

### Describe the solution you'd like

Ideally: for upstream to ship the sources to these binaries, so that:
1. The upstream project is distributable by downstreams as-is.
2. The entirety of the project is free and open source.
3. Others can run the test suite.

### Describe alternatives you've considered

If the above is not possible, then an alternative would be to attach a different LICENSE for the binary files, so that the project becomes distributable, even if partially non-free/open source.

### Additional context

My goal is to package (newer versions of) esptool into Debian. As of this week, we have 4.5.1 in Debian experimental. One of the steps that the Debian packaging side has traditionally performed, is to strip the "test" directory.

I would like to:
- Ship the pristine upstream esptool .tar.gz in Debian without modifying it.
- Run the tests (at least the ones not requiring physical hardware) in the Debian CI infrastructure.

Note that in addition to the distributability issue outlined above, Debian has a requirement of everything that is shipped as part of Debian (main) needs to be free and open source, i.e. comply with the Debian Free Software Guidelines (DFSG). So only the solution where the sources are published would work for this purpose.

dobairoland, 2023-03-20:
Thank you for raising this issue. You are right. Unfortunately, providing sources are not easy for us because some of the images were also manually modified, or provided by our customers without sources. We will try to address this soon. But I'm afraid, for now you will have to keep stripping the test directory.

paravoid, 2023-03-20:
Thank you for the quick response and explanation! Understood :)

Some tests would be better than no tests, so there is value in providing at least some of them or doing this incrementally.