espressif/esptool #869: ESP-Prog SPI flash not working on personal custom PCB

DreamStonks, 2023-04-04:
### Operating System

Windows 10 pro 

### Esptool Version

v4.5.1

### Python Version

Python 3.11.1

### Chip Description

ESP32-C3-WROOM-02

### Device Description

The module is connected to the SPI port of an ESP-Prog using the provided flat cables. 

For testing purposes, the chip is only connected to the header pins (U3) with a 1kOhm resistor (R2) between GPIO 8 and VDD. 
[image]
[image]
[image]

### Hardware Configuration

The headers pins used: https://www.digikey.ca/en/products/detail/sullins-connector-solutions/GRPB032VWVN-RC/1786453

### How is Esptool Run

VScode ESP-IDF extension 

### Full Esptool Command Line that Was Run

esptool.py', '-p', 'COM7', '-b', '9600', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32c3', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', '2MB', '0x0', 'bootloader/bootloader.bin', '0x10000', 'hello_world.bin', '0x8000', 'partition_table/partition-table.bin'

### Esptool Output

```shell
*  Executing task: C:\Users\justin\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\justin\esp\esp-idf\components\esptool_py\esptool\esptool.py -p COM7 -b 9600 --before default_reset --after hard_reset --chip esp32c3 write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin 

esptool.py v4.5.1
Serial port COM7
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-C3: Invalid head of packet (0x00): Possible serial noise or corruption.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

 *  The terminal process "C:\Users\justin\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe 'C:\Users\justin\esp\esp-idf\components\esptool_py\esptool\esptool.py', '-p', 'COM7', '-b', '9600', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32c3', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', '2MB', '0x0', 'bootloader/bootloader.bin', '0x10000', 'hello_world.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 2.
```

### More Information

- I tested the voltage of the ESP-Prog to make sure it was 3.3v and it indeed was.
- I tried to wiggle the cable 
- I tried replugging the cable in the other direction 
- I changed the baud rate to 9600
- I tried to unconnect GPIO9 using the header pin junction on the ESP-Prog
- I tried diverse and varied combinations of button pushing (reset and boot)
- I tried using my own breadboard cables
- I tried replacing the chip

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-04-05:
Hello @DreamStonks,

I recommend using our [free-of-charge reviewing services for circuit diagrams and PCB designs](https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design-review), I don't have the capacity to review your design.

> The module is connected to the SPI port of an ESP-Prog using the provided flat cables.

The ESP-Prog is acting as a USB-to-UART bridge, there is no SPI port on it.

I would start by attaching only the 3v3, GND, RX, and TX connections and then see if any data comes out of the ESP (with any serial terminal programs). If not, try switching TX and RX.

When you get a working serial connection, you can focus on the automatic bootloader reset circuitry (attaching GPIO9 and Enable pins)

DreamStonks, 2023-04-05:
Hi @radimkarnis, 

> "I would start by attaching only the 3v3, GND, RX, and TX connections and then see if any data comes out of the ESP (with > any serial terminal programs). If not, try switching TX and RX." 

Using puTTY while putting some pressure on the cables, I can now see "ESP-ROM:esp32c3-api1-20210207" in the terminal which after some digging, I believe to be the first line of the ESP32-C3 startup log...
[image]
https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/Get_Started/Downloading_guide.html

Regarding the free-of-charge reviewing services for circuit diagrams and PCB designs, are they open to hobbyists like myself? I am not affiliated with any company and I do not plan to commercialize the project for the moment.

radimkarnis, 2023-04-14:
@DreamStonks 
yes, it is the first line of the boot log, but there should be more, e.g.:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xf (BROWNOUT_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
```

Maybe something is immediately resetting your chip?
Have you burned any efuses?

radimkarnis, 2023-05-10:
I am closing this issue for inactivity. Feel free to reopen if you manage to debug and get more information.

closed 2023-05-10 by radimkarnis

DreamStonks, 2023-11-11:
> I am closing this issue for inactivity. Feel free to reopen if you manage to debug and get more information.

Forgive me for the very late response... It turned out to be a very easy fix! The cable just needed to be held straight. I hope this can help someone encountering the same issue in the future.