espressif/esptool #870: esptool flashes but does not erase flash

jouellnyc, 2023-04-06:
### Operating System

Linux / uname -a  5.15.0-67-generic #74-Ubuntu SMP Wed Feb 22 14:14:39 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

### Esptool Version

esptool.py v4.6-dev

### Python Version

Python 3.10.6

### Chip Description

ESP32

### Device Description

ESP32-DevKitC-VE Development Board

This one: https://www.amazon.com/gp/product/B087TNPQCV/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

### Hardware Configuration

2.8 Inch ILI93414 screen

### How is Esptool Run

No

### Full Esptool Command Line that Was Run

./esptool.py   --port /dev/ttyUSB0 write_flash -z 0x1000   /tmp/firmware/esp32spiram-20230405-unstable-v1.19.1-1008-gc046b23ea.bin

### Esptool Output

```shell
== w/o --trace:

/esptool.py   --port /dev/ttyUSB0 write_flash -z 0x1000   ~/esp32/firmware/esp32spiram-20230405-unstable-v1.19.1-1008-gc046b23ea.bin
esptool.py v4.6-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 40:22:d8:06:f5:58
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00175fff...
Compressed 1527632 bytes to 990167...
Wrote 1527632 bytes (990167 compressed) at 0x00001000 in 87.3 seconds (effective 140.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin.../esptool.py   --port /dev/ttyUSB0 write_flash -z 0x1000   ~/esp32/firmware/esp32spiram-20230405-unstable-v1.19.1-1008-gc046b23ea.bin
esptool.py v4.6-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
[... 260 lines cut ...]
Hash of data verified.

Leaving...
TRACE +0.000 command op=0x02 data len=16 wait_response=1 timeout=3.000 data=00000000000000000040000000000000
TRACE +0.000 Write 26 bytes: 
    c000021000000000 0000000000000000 | ................
    0000400000000000 00c0             | ..@.......
TRACE +0.006 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 01020200000000000000c0
TRACE +0.000 Received full packet: 01020200000000000000
TRACE +0.000 command op=0x12 data len=4 wait_response=1 timeout=3.000 data=01000000
TRACE +0.000 Write 14 bytes: c0001204000000000001000000c0
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 01120200000000000000c0
TRACE +0.000 Received full packet: 01120200000000000000
Hard resetting via RTS pin...
```

### More Information

It does not seem specific to this board at all.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-04-07:
I'm sorry but I don't get it what is wrong. The output seems normal to me.

jouellnyc, 2023-04-09:
Hi @dobairoland 

I'm terribly sorry. I entered this in error!

radimkarnis, 2023-04-11:
@jouellnyc so can this ticket be closed or will you provide another log?

closed 2023-04-17 by dobairoland